taligentx/dscKeybusInterface #108: panel pc585 pc1565 keypad FAP

luk2009, 2019-07-08:
the F A P buttons of the pc585 and pc1565 keypad do not send any data to the serial port when pressed for 3 seconds

in the pc1832 panel works perfectly

taligentx, 2019-07-10:
Try loading the `KeybusReader` sketch and capture the output while you press each of the F/A/P keys, it'd be useful to see the output.

taligentx, 2019-07-18:
@luk2009 Could you get a log of this?

luk2009, 2019-07-22:
DSC Keybus Interface is online.
    0.13: 00000101 0 10010001 00000001 10010001 11000111 [0x05] Partition 1: Ready Trouble Backlight - Partition ready | Partition 2: disabled
    3.16: 01011101 0 00000000 00000000 00000000 00000000 00000000 01011101 [0x5D] Partition 1 | Status lights flashing: none | Zones 1-32 flashing: none
   12.19: 10100101 0 00010001 00101101 01100000 10110000 00000000 00000000 11110011 [0xA5] 2011.11.11 00:44 | Timestamp
   27.16: 00010001 0 10101010 10101010 10101010 10101010 10101010 [0x11] Keypad slot query
   27.16: 11111111 1 00111111 11111111 11111111 11111111 11111111 [Keypad] Slots active: 1 
   35.51: 01110111 1 11111111 11111111 11111111 11111111 [Keypad] Fire alarm
   35.52: 00011100 0 [0x1C] Verify keypad Fire/Auxiliary/Panic
   35.52: 01110111 1 [Keypad] Fire alarm
   35.59: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
   35.66: 10100101 0 00010001 01101101 01100000 10110000 01001110 00000001 10000010 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Fire alarm
   35.73: 00000101 0 11010101 00010001 10010001 11000111 [0x05] Partition 1: Ready Memory Trouble Fire Backlight - Partition in alarm | Partition 2: disabled
   35.80: 10100101 0 00010001 01101101 01100000 10110000 01010010 00000001 10000110 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Fire alarm restored
   38.09: 11111111 1 00001010 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   38.43: 11111111 1 00100111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   38.67: 11111111 1 00001010 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   39.07: 11111111 1 00100111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   39.13: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
   39.23: 01110101 0 00000000 01110101 [0x75] Partition 1 | Beep pattern: off
   39.28: 00000101 0 10010101 00000001 10010001 11000111 [0x05] Partition 1: Ready Memory Trouble Backlight - Partition ready | Partition 2: disabled
   42.71: 10111011 1 11111111 11111111 11111111 11111111 [Keypad] Auxiliary alarm
   42.72: 00011100 0 [0x1C] Verify keypad Fire/Auxiliary/Panic
   42.72: 10111011 1 [Keypad] Auxiliary alarm
   42.77: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
   42.84: 10100101 0 00010001 01101101 01100000 10110000 01001111 00000001 10000011 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Aux alarm
   42.98: 10100101 0 00010001 01101101 01100000 10110000 01010011 00000001 10000111 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Aux alarm restored
   51.68: 11011101 1 11111111 11111111 11111111 11111111 [Keypad] Panic alarm
   51.70: 00011100 0 [0x1C] Verify keypad Fire/Auxiliary/Panic
   51.70: 11011101 1 [Keypad] Panic alarm
   51.75: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
   51.83: 10100101 0 00010001 01101101 01100000 10110000 01010000 00000001 10000100 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Panic alarm
   51.89: 00000101 0 10010101 00010001 10010001 11000111 [0x05] Partition 1: Ready Memory Trouble Backlight - Partition in alarm | Partition 2: disabled
   51.97: 10100101 0 00010001 01101101 01100000 10110000 01010100 00000001 10001000 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Panic alarm restored
   56.92: 01110111 1 11111111 11111111 11111111 11111111 [Keypad] Fire alarm
   56.94: 00011100 0 [0x1C] Verify keypad Fire/Auxiliary/Panic
   56.94: 01110111 1 [Keypad] Fire alarm
   57.02: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
   57.09: 10100101 0 00010001 01101101 01100000 10110000 01001110 00000001 10000010 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Fire alarm
   57.15: 00000101 0 11010101 00010001 10010001 11000111 [0x05] Partition 1: Ready Memory Trouble Fire Backlight - Partition in alarm | Partition 2: disabled
   57.36: 10100101 0 00010001 01101101 01100000 10110000 01010010 00000001 10000110 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Fire alarm restored
   59.49: 10111011 1 11111111 11111111 11111111 11111111 [Keypad] Auxiliary alarm
   59.51: 00011100 0 [0x1C] Verify keypad Fire/Auxiliary/Panic
   59.51: 10111011 1 [Keypad] Auxiliary alarm
   59.56: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
   59.63: 10100101 0 00010001 01101101 01100000 10110000 01001111 00000001 10000011 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Aux alarm
   59.78: 10100101 0 00010001 01101101 01100000 10110000 01010011 00000001 10000111 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Aux alarm restored
   61.91: 11011101 1 11111111 11111111 11111111 11111111 [Keypad] Panic alarm
   61.93: 00011100 0 [0x1C] Verify keypad Fire/Auxiliary/Panic
   61.93: 11011101 1 [Keypad] Panic alarm
   61.99: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
   62.06: 10100101 0 00010001 01101101 01100000 10110000 01010000 00000001 10000100 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Panic alarm
   62.20: 10100101 0 00010001 01101101 01100000 10110000 01010100 00000001 10001000 [0xA5] 2011.11.11 00:44 | Partition 1 | Keypad Panic alarm restored
   72.22: 00010110 0 00001110 00100011 11110101 00111100 [0x16] PC1555MX | Zone wiring: NC | Interval 4m

luk2009, 2019-07-22:
with dsc status

DSC Keybus Interface is online.
Keybus connected
Partition 1 fire alarm on
Partition 1 fire alarm restored
Trouble status on
Partition 1 exit delay in progress
Partition 1 armed away
Partition 1 disarmed

Fire working
Panic and Auxiliary not working

taligentx, 2019-07-22:
Thanks for the logs, this generation panel seems to send a partition number along with the keypad F/A/P keys (as well as the AC power and battery status messages).  At some point DSC changed the protocol to not send a partition number - which makes sense for these status items that are not tied to a partition.

The library was only setting status if there was no partition number, but there's no conflict if a partition number is sent so I've removed the check.  Try the latest develop branch to see if this is resolved.

luk2009, 2019-07-22:
'class dscKeybusInterface' has no member named 'loop'

this is the error that I get when trying to compile and mark the line 71:

dsc.loop();

luk2009, 2019-07-22:
I'm sorry I didn't have the library updated. everything is working correctly

Keypad aux alarm
Partition 1 exit delay in progress
Partition 1 not ready
Partition 1 armed away
Keypad panic alarm
Partition 1 in alarm
Partition 1 ready
Partition 1 disarmed
Keypad fire alarm
Partition 1 not ready
Partition 1 in alarm
Partition 1 fire alarm on
Partition 1 ready
Partition 1 fire alarm restored

luk2009, 2019-07-22:
After pressing the keys, only the restored one is sent with the F key.
the status of ready and not ready is sent along with the panic, aux or fire keys
the status of not ready is sent, even if the ready light is on.

When you activate the alarm and deactivate it while counting the delay time, send disarmed

luk2009, 2019-07-22:
KeybusReader pc1832
  8.60: 00010001 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0x11] Keypad slot query
    8.60: 11111111 1 00111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Slots active: 1 
   10.81: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   11.21: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   11.53: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   11.90: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   12.08: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
   12.14: 01011101 0 00000000 00000000 00000000 00000000 00000000 01011101 [0x5D] Partition 1 | Status lights flashing: none | Zones 1-32 flashing: none
   12.22: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
   12.26: 10111011 0 00000000 00000000 10111011 [0xBB] Bell: off
   12.35: 11100110 0 00011010 01000000 00000000 00000000 00001001 00000000 00000000 00000000 01001001 [0xE6] 0x1A: Unknown data
   12.38: 01110101 0 00000000 01110101 [0x75] Partition 1 | Tone: off
   12.44: 11001110 0 00000001 00000000 00000000 00000000 00000000 11001111 [0xCE] Unknown data
   12.51: 11001110 0 01000000 11111111 11111111 11111111 11111111 00001010 [0xCE] Unknown data
   12.54: 01110101 0 00010001 10000110 [0x75] Partition 1 | Tone: 1 beep, 1s interval
   12.62: 00000101 0 10000011 00001000 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1: Ready Armed Backlight - Exit delay in progress | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   16.60: 01110101 0 00110001 10100110 [0x75] Partition 1 | Tone: 3 beeps, 1s interval
   17.78: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   18.02: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   18.34: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   18.71: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   19.03: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
   19.18: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
   19.22: 10111011 0 00000000 00000000 10111011 [0xBB] Bell: off
   19.30: 11100110 0 00011010 01000000 00000000 00000000 00001001 00000000 00000000 00000000 01001001 [0xE6] 0x1A: Unknown data
   19.34: 01110101 0 00000000 01110101 [0x75] Partition 1 | Tone: off
   19.40: 11001110 0 00000001 00000000 00000000 00000000 00000000 11001111 [0xCE] Unknown data
   19.46: 11001110 0 01000000 11111111 11111111 11111111 11111111 00001010 [0xCE] Unknown data
   19.54: 00000101 0 10010001 00111110 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1: Ready Trouble Backlight - Partition disarmed | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   21.44: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   21.68: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   21.91: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   22.21: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   22.39: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
   22.53: 11100110 0 00011000 00000001 00000000 00000000 00000000 00000000 00000000 11111111 [0xE6] Partition 1 | Status lights flashing: none | Zones 33-64 flashing: none
   22.57: 10111011 0 00000000 00000000 10111011 [0xBB] Bell: off
   22.66: 11100110 0 00011010 01000000 00000000 00000000 00001001 00000000 00000000 00000000 01001001 [0xE6] 0x1A: Unknown data
   22.69: 01110101 0 00010001 10000110 [0x75] Partition 1 | Tone: 1 beep, 1s interval
   22.75: 11001110 0 00000001 00000000 00000000 00000000 00000000 11001111 [0xCE] Unknown data
   22.82: 11001110 0 01000000 11111111 11111111 11111111 11111111 00001010 [0xCE] Unknown data
   22.90: 00000101 0 10000011 00001000 00010000 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1: Ready Armed Backlight - Exit delay in progress | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
   23.57: 11100110 0 00001101 00000000 11110011 [0xE6] Zones 49-56 open: none
   23.81: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   24.04: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   24.36: 11111111 1 00001010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   24.65: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   24.98: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
[... cut ...]
   48.18: 11101011 0 00000000 00010001 00101101 10100000 00011100 00000000 01010000 00100010 01010111 [0xEB] 2011.11.13 00:07 | Keypad Panic alarm
   48.34: 10100101 0 00010001 00101101 10100000 00011100 01010100 00000000 11110011 [0xA5] 2011.11.13 00:07 | Keypad Panic alarm restored
   48.52: 11101011 0 00000000 00010001 00101101 10100000 00011100 00000000 01010100 00000000 00111001 [0xEB] 2011.11.13 00:07 | Keypad Panic alarm restored
   53.74: 10100101 0 00010001 00101101 10100000 00100000 00000000 00000000 10100011 [0xA5] 2011.11.13 00:08 | Timestamp

luk2009, 2019-07-22:
pc1832  Status

Partition 1 exit delay in progress
Partition 1 disarmed
Timestamp: 2011.11.13 00:04
Keypad panic alarm
Partition 1 not ready
Partition 1 in alarm
Partition 1 ready
Keypad aux alarm
Partition 1 exit delay in progress
Partition 1 disarmed
Keypad fire alarm
Partition 1 not ready
Partition 1 in alarm
Partition 1 fire alarm on
Partition 1 ready
Partition 1 fire alarm restored

luk2009, 2019-07-22:
pc1565 keybusReader
   
   
   39.98: 11111111 1 00001010 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   40.47: 10100101 0 00010001 01101101 01100000 00000000 11101100 11111111 01101110 [0xA5] 2011.11.11 00:00 | Partition 1 | Telephone line trouble
   40.55: 00100111 0 10010001 00000001 01100000 00000000 00000000 00011001 [0x27] Partition 1: Ready Trouble Backlight - Partition ready | Partition 2: Program Fire - Unknown data: 0x00 | Zones 1-8 open: none
   40.55: 11111111 1 00100111 11111111 11111111 11111111 11111111 11111111 [Keypad] 9
   40.70: 11111111 1 00001010 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   41.13: 11111111 1 00100111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   41.22: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
   41.31: 01110101 0 00000000 01110101 [0x75] Partition 1 | Tone: off
   41.36: 00000101 0 10000011 00001000 10010001 11000111 [0x05] Partition 1: Ready Armed Backlight - Exit delay in progress | Partition 2: disabled
   41.47: 00100111 0 10000011 00001000 10010001 11000111 00000000 00001010 [0x27] Partition 1: Ready Armed Backlight - Exit delay in progress | Partition 2: disabled | Zones 1-8 open: none
   41.50: 01110101 0 00010001 10000110 [0x75] Partition 1 | Tone: 1 beep, 1s interval
   42.49: 01110101 0 00110001 10100110 [0x75] Partition 1 | Tone: 3 beeps, 1s interval
   44.30: 11111111 1 00001010 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   44.89: 11111111 1 00100111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   45.14: 11111111 1 00001010 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   45.63: 11111111 1 00100111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   45.75: 00100111 0 10000010 00000101 10010001 11000111 00000000 00000110 [0x27] Partition 1: Armed Backlight - Armed away | Partition 2: disabled | Zones 1-8 open: none
   45.95: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
   46.15: 10100101 0 00010001 01101101 01100000 00000000 10111011 11111111 00111101 [0xA5] 2011.11.11 00:00 | Partition 1 | Armed by master code 40
   46.19: 01110101 0 00000000 01110101 [0x75] Partition 1 | Tone: off
   46.25: 00000101 0 10010001 00111110 10010001 11000111 [0x05] Partition 1: Ready Trouble Backlight - Partition disarmed | Partition 2: disabled
   46.33: 10100101 0 00010001 01101101 01100000 00000010 10011011 00000000 00100000 [0xA5] 2011.11.11 00:00 | Partition 1 | Armed: away
   46.46: 10100101 0 00010001 01101101 01100000 00000000 11100010 11111111 01100100 [0xA5] 2011.11.11 00:00 | Partition 1 | Disarmed by master code 40
   46.53: 00100111 0 10010001 00111110 01100000 00000000 00000000 01010110 [0x27] Partition 1: Ready Trouble Backlight - Partition disarmed | Partition 2: Program Fire - Unknown data: 0x00 | Zones 1-8 open: none
   46.62: 11111111 1 11111111 11111111 11111111 11111011 [Keypad] Zone status notification
   46.71: 11010101 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0xD5] Keypad zone query
   46.71: 11111111 1 00001111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Keypad] Slot 1 zone open
   46.77: 00100111 0 10010001 00111110 11111111 11111111 00000000 11110100 [0x27] Partition 1: Ready Trouble Backlight - Partition disarmed | Zones 1-8 open: none
   47.49: 00100111 0 10010001 00111110 10010001 11000111 00000000 01001110 [0x27] Partition 1: Ready Trouble Backlight - Partition disarmed | Partition 2: disabled | Zones 1-8 open: none
   50.50: 00100111 0 10010001 00000001 10010001 11000111 00000000 00010001 [0x27] Partition 1: Ready Trouble Backlight - Partition ready | Partition 2: disabled | Zones 1-8 open: none
   50.55: 00000101 0 10010001 00000001 10010001 11000111 [0x05] Partition 1: Ready Trouble Backlight - Partition ready | Partition 2: disabled
   52.85: 01110111 1 11111111 11111111 11111111 11111111 [Keypad] Fire alarm
   52.86: 00011100 0 [0x1C] Verify keypad Fire/Auxiliary/Panic
   52.86: 01110111 1 [Keypad] Fire alarm
   52.91: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
   52.99: 10100101 0 00010001 01101101 01100000 00000000 01001110 00000001 11010010 [0xA5] 2011.11.11 00:00 | Partition 1 | Keypad Fire alarm
   53.06: 00000101 0 11010101 00010001 10010001 11000111 [0x05] Partition 1: Ready Memory Trouble Fire Backlight - Partition in alarm | Partition 2: disabled
   53.13: 10100101 0 00010001 01101101 01100000 00000000 01010010 00000001 11010110 [0xA5] 2011.11.11 00:00 | Partition 1 | Keypad Fire alarm restored
   53.55: 00100111 0 11010101 00010001 00000000 00000000 00000000 00001101 [0x27] Partition 1: Ready Memory Trouble Fire Backlight - Partition in alarm | Partition 2: none - Unknown data: 0x00 | Zones 1-8 open: none
   54.51: 00100111 0 11010101 00010001 10010001 11000111 00000000 01100101 [0x27] Partition 1: Ready Memory Trouble Fire Backlight - Partition in alarm | Partition 2: disabled | Zones 1-8 open: none
   56.34: 11111111 1 00001010 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   56.79: 11111111 1 00100111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   57.08: 11111111 1 00001010 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 2
   57.43: 11111111 1 00100111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   57.50: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
[... cut ...]
   77.82: 11111111 1 00100111 11111111 11111111 11111111 [Keypad] Partition 1 | Key: 9
   77.89: 01100100 0 00001100 01110000 [0x64] Partition 1 | Beep: 6 beeps
   77.98: 01110101 0 00000000 01110101 [0x75] Partition 1 | Tone: off
   78.03: 00000101 0 10010101 00000001 10010001 11000111 [0x05] Partition 1: Ready Memory Trouble Backlight - Partition ready | Partition 2: disabled
   78.51: 00100111 0 10010101 00000001 10010001 11000111 00000000 00010101 [0x27] Partition 1: Ready Memory Trouble Backlight - Partition ready | Partition 2: disabled | Zones 1-8 open: none

luk2009, 2019-07-22:
pc1565  status

Keypad panic alarm
Partition 1 not ready
Partition 1 in alarm
Partition 1 ready
Keypad aux alarm
Partition 1 exit delay in progress
Partition 1 disarmed
Keypad fire alarm
Partition 1 not ready
Partition 1 in alarm
Partition 1 fire alarm on
Partition 1 ready
Partition 1 fire alarm restored
Keypad fire alarm
Partition 1 not ready
Partition 1 in alarm
Partition 1 fire alarm on
Partition 1 ready
Partition 1 fire alarm restored

taligentx, 2019-07-27:
> After pressing the keys, only the restored one is sent with the F key.

This is normal, the F/A/P keypad keys are one-time notifications.  For `Fire`, the panel sets a fire status for the partition that will only be cleared after the partition is armed/disarmed - that's the status tracked by `dsc.fire[partition]`.

> the status of ready and not ready is sent along with the panic, aux or fire keys
> the status of not ready is sent, even if the ready light is on.

For the library, `dsc.ready[partition]` tracks if the panel is ready to arm, and may not match the Ready light.  With the F/A/P keys, the panel is not ready to arm so you get the Not Ready status.  If you need to track the Ready light exactly, you can get the lights status with `dsc.lights[partition]`.

> When you activate the alarm and deactivate it while counting the delay time, send disarmed

That's normal, and also how the DSC IT-100 serial interface sends the status.  It's how you can determine if the exit delay was cancelled.

taligentx, 2019-07-31:
Feel free to re-open if the status behavior still has issues.

closed 2019-07-31 by taligentx