taligentx/dscKeybusInterface #109: No issue But have a question.

Therealjosephchrzempiec, 2019-07-19:
Hello i have been looking at your sketch is there a way to web  arm the system. I have a DSC PC1616  I see a MQTT and a Read and well as a status arduino sketch. But i was wondering if there is a Web arm and disarm sketch? I don't need another app out there and i don't have a home automated system either. But i was looking for a way to have a web sketch. My brother has a DSC PC1515 i was wondering if have you looked into that one yet?

Joseph

taligentx, 2019-07-19:
Yep, take a look at the esp8266 `VirtualKeypad-Web` example sketch.

Therealjosephchrzempiec, 2019-07-19:
Hello i was wondering if you ever looked at the PC1550 or any of the neo systems yet.

taligentx, 2019-07-19:
Please refer to the README.md, this is already addressed.

Therealjosephchrzempiec, 2019-07-19:
I'm sorry i must of over looked it or didn't see it. Thank you.

By the way: You are doing a awesome job a lot of people needs this. Hope to see more systems added in the future.

Joseph

taligentx, 2019-07-22:
Thanks for the feedback!  It's great to hear how the library is being used.

closed 2019-07-22 by taligentx