taligentx/dscKeybusInterface #110: Esp8266 Page can not be displayed?

Therealjosephchrzempiec, 2019-07-21:
Hello i have uploaded the esp8266 sketch called VirtualKeypad-Web. Everything look it it took okay, But when i go go to the http://dsc.local or the ip address i get nothing it says Page can not be displayed. Can someone Please help me? I get no Errors.

Joseph

Therealjosephchrzempiec, 2019-07-21:
Hello just a update. I figure out one problem I didn't upload the Sketch data upload. Because i never did anything like that before. So i figure that out and i got that to upload. Now I'm able to go to the ip address and see the key panel.

So second problem trying to go to the http://dsc.local that still doesn't work page can not be displayed. Can someone please help me out with that? I have looked all over and can not figure out what is wrong.

Joseph

taligentx, 2019-07-22:
> So second problem trying to go to the http://dsc.local that still doesn't work page can not be displayed. 

Does it work if you go directly to the IP address of the esp8266 (as seen through the serial output)?

Therealjosephchrzempiec, 2019-07-22:
Hello i can get the local ip address of 192.168.1.50 and that does work. Just not the http://dsc.local that doesn't work.

joseph

taligentx, 2019-07-22:
> Hello i can get the local ip address of 192.168.1.50 and that does work. Just not the http://dsc.local that doesn't work.

Great to hear that the IP address works.  In that case the interface itself seems to be good to go, you'll just need to decide if you want to do anything for DNS resolution or just continue using the IP address.

The `VirtualKeypad-Web` sketch uses mDNS to announce http://dsc.local on the network, your client and network must support mDNS for this to work.  Otherwise you could look into setting static DNS entries for the interface on your router, for example.  It's a bit out of scope for this repository, though.

closed 2019-07-26 by taligentx

Therealjosephchrzempiec, 2019-07-29:
Thank you taligentc. Yes my network doesn't support mDNS. But it is okay. I figure out the IP address pretty quicky with a mobile phone application.

Joseph