taligentx/dscKeybusInterface #111: Add support for PowerSeries Neo

connected201, 2019-07-22:
hi, i received today Dsc PowerSeries Neo (https://www.dsc.com/dsc-product-families/Neo/1), and tested with current firmware, as expected, it's not working...
question, it,s possible to add support for this model?
i can send to firmware developers, dsc panel to reverse new protocol. Best regards
dsc neo screens:
[image]
[image]
[image]
[image]
[image]

taligentx, 2019-07-22:
Hi @connected201 - please refer to the `README.md`, this has already been addressed.  Neo uses a totally different encrypted protocol so unless someone is able to crack the encryption these systems cannot be directly interfaced.

closed 2019-07-23 by connected201

PascalSI, 2019-09-10:
Hi @connected201 , where can i see how you made a printed circuit board for such a case?

connected201, 2019-09-10:
hi @PascalSI , i wil made some changes, and add gsm support , and i will public all sources