taligentx/dscKeybusInterface #113: Another question about arming and disarming in the ESP8266?

Therealjosephchrzempiec, 2019-07-30:
Hello I was wondering  If there is a simple commands in the esp8266 i can arm and disarm the arm that I'm using it is a DSC PC1616 panel. I'm not good at programming but i just wanted to make a simple command i can attach a couple button it the esp to arm and disarm?

Joseph

taligentx, 2019-07-30:
Hi @Therealjosephchrzempiec, you can arm/disarm by writing the access code as something like:
```
dsc.write("1234");
```

Therealjosephchrzempiec, 2019-07-30:
Hello taligentx thank you. Can i still use the web interface for the esp8266 to do this?

taligentx, 2019-07-31:
Sure, it's up to you how you want it to work, the example sketches show different features and you can combine them as needed.

Therealjosephchrzempiec, 2019-08-01:
Thank you once again taligentx.

taligentx, 2019-08-20:
Feel free to re-open if there is an issue with the library.

closed 2019-08-20 by taligentx