taligentx/dscKeybusInterface #114: [PC1555] Random data being written to keybus?

bourgeois, 2019-08-15:
Before asking my question, I wanted to thank you all for this great work!

**Disclaimer**: I don't know if my system is supported, it seems I have a PC1555, [but it is labelled as classic in my manual](https://imgur.com/pW2a7og) ? So I'm not sure which protocol is is using. Also, it's worth nothing that I have a LED5508 next to my entry door, which is part of the power series I believe?

Anyway, with all that being said, I decided to give it a try and I flashed my esp8266 with the `homebridge-mqtt` example. So far, I am able to read data properly (which is awesome!), I can get zone data in homekit, when I arm the system I get a notification as well so that part seems to work fine.

However, when I connect the `dscWritePin`, LEDs starts flashing on the other keyboards in my house and I hear some random "beeps" from the other keyboards as well. Could it be that my ESP is sending some gibberish data on the data line ?

I am using pin D8 as suggested and I am pretty sure I did all the wiring correctly (I used a 2N3904 transistor). My ESP is powered from the dsc board using a 5V regulator feeding the `vin` pin.

I'm currently running everything on a breadboard, I know it's not ideal, but I wanted to make sure everything is working before putting everything together on a soldered PCB.

Could it be that my system is not supported when writing data, while reading works ? That would be weird, I would assume that if I can read the bus, writing should work as well?

I'll try the develop branch tonight to see if that makes a difference.
Let me know if I missed something or if there is something I should look at!

taligentx, 2019-08-16:
> Before asking my question, I wanted to thank you all for this great work!

Thank you! Hope it's handy.

> I don't know if my system is supported, it seems I have a PC1555

The labeling is confusing but the PC1555 does use the PowerSeries Keybus protocol and will work with this library.

> I'm currently running everything on a breadboard

If the wiring to the transistor pinout is correct, it may be noise from the breadboard wiring triggering the transistor and corrupting the Keybus data line.  This has happened previously with breadboards, try directly soldering all components and keep the wiring between the microcontroller and passive components (resistors and transistor) short.  The DSC clock and data lines can run long distances to the microcontroller and passive components.

bourgeois, 2019-08-16:
Thanks a lot, I’ll give that a try!

bourgeois, 2019-08-16:
Ok, so I tried soldering everything together on a real pcb and I am facing the same issue :( 

What's weird is that I get the same behavior by not plugging in the D8 pin (nothing connected to the base of the transistor ?!). If I remove the dscWritePin connection to the data line (collector pin), I am able to read data fine as before. Am I using the right transistor ? It really is a 2N3904...

bourgeois, 2019-08-19:
Turns out pin order was wrong on my 2N3904, it was collector-base-emiter instead of emiter-base-collector!

it’s all working now, thanks a lot for your help!

closed 2019-08-19 by bourgeois

taligentx, 2019-08-20:
Great to hear, thanks for the follow up!