taligentx/dscKeybusInterface #116: Unavailiable status after reboot of PC and HA

Vagelis1971, 2019-08-25:
Hello people.
im getting an unavailiable status if i reboot the pc or HA . Then to make it work properly i need to arm the system manualy and only after that i can see the status in HA . From this point till the next start up everything works fine .. im on an old power 830 . Maybe you guys have the solution for it ?
Thanks in adv

Vagelis1971, 2019-08-27:
Im sorry i didnt got that. What do you mean?

Vagelis1971, 2019-08-27:
Ahhh. found it .. im using the master...

Vagelis1971, 2019-08-29:
I have upgraded in 1.2 version and the same thing. Im still having that problem i mentioned before . im on a nodemcu V1 . Actualy had the same problem with tamota sonoff mqtt software. but that i sovled by  adding an automation that restarts all my nodemcu tasmota software as HA starts with a delay of 20 sec. All the unavailiable entities got back each time win10 auto restarts HA after an electrical cut out . I tried to reset it manually from nodemcu board but same thing.

taligentx, 2019-08-31:
The develop branch should handle this, it's one of the changes since the last release.

Vagelis1971, 2019-08-31:
k ill try it tonight and let you know. thanks

Vagelis1971, 2019-08-31:
we are getting close. now the situation is as folows. When restart my pc its getting in unavailiable status BUT this time when i press the recet button on the nodemcu pbc then becomes availiable the status. So is there any possible way for your program to accept a recet command from HA as i did with the rest of the  sooff tasmota nodemcu ?

taligentx, 2019-08-31:
To troubleshoot this, try subscribing to the `dsc/Status` MQTT topic on a different computer and see which messages show up while you reboot the HA and MQTT system.  Do you have HA and the MQTT broker on the same system?

The develop branch sketch tracks the connection to the MQTT broker and sends a full status update whenever the MQTT broker connection is reconnected, the sketch doesn't check for HA status.  HA should be picking up the status from the MQTT broker because the MQTT messages from the sketch are marked as persistent, even if HA comes online after the status update is sent, HA should receive the current status from the MQTT broker once HA is online.

taligentx, 2020-10-02:
Closing for lack of updates, feel free to re-open if this is still an issue.

closed 2020-10-02 by taligentx