taligentx/dscKeybusInterface #119: Dsc write stepped working

jjcuen, 2019-09-21:
Hi
I have been use your contribution for almost a year with home assistant(great work by the way!)
I tried to do a few modifications (multi WiFi) but something got messed up and now I can’t send any command to the panel. I went back to the original program (.ino) and I can read all statuses from the panel and keypad but I can’t send commands through HA/MQTT. I have debuged  my .ino file and noticed that all MQTT commands are received by nodemcu8266 through the callback function, and it is received by the dsc handle function but nothing gets sent to the panel (not sure how to debug the dsc.write function) A few days ago it was a hit or miss. Sometimes it accepted commands. Now it stopped completely accepting anything. I already changed wires, power supply for nodemcu, tried three different boards and I even tried a new breadboard and connected everything from scratch, using new resistors and transistor. Still no luck. Do you think I may have damaged something at the panel(I have a DSC 1616). Sorry for the long letter...I’m very frustrated m This was such a great gadget to have automated.... any help will be enormously appreciated. Thanks

jjcuen, 2019-09-21:
Update: well, now its working. I don't know exactly what happened. I soldered again a few wires along with the the transistor. That may have done the trick. It's a little frustrating not knowing what caused it. I'll monitor it for a few days to see if it holds

taligentx, 2020-09-30:
Hi @jjcuen - good to hear the re-soldering resolved the issue.

closed 2020-09-30 by taligentx