taligentx/dscKeybusInterface #120: Wemos D1 overheat => destroyed, what went wrong?

mrmaddotta, 2019-09-23:
Hey Folks

Thanks to the contributors of this project, it's really cool :)

Hoping to get some guidance here. Seems my Wemos D1 Mini overheated when hooked up to a DSC 5005 panel. Here's what happened -

At first I just hooked up the reading connections (didn't have a transistor at the time). Used 10k & 3.3k ohm resistors and all was great. PS - Used a LM2596 step down for the main power.

Finally got hold of a pack of 2N3904 transistors and hooked it up (with a 1k ohm resistor) so I could control the panel. Everything worked great, was able to arm/disarm etc.

About 10min (maybe 5) later, I noticed it had disconnected so naturally went to go double check on the Wemos. Upon touching it, burnt my fingers as it was so hot. In short, it's now dead. 

Any ideas on where I could have gone wrong?

Would appreciate some guidance on this one :)

zzhang1, 2019-10-15:
I have a Wemos D1 R2 that has been working fine for almost a year now.  I loosely followed the NodeMCU schematic for my board because the layout was easy to wire and solder. It called for 15k and 10k resistors, so I went for those. https://github.com/taligentx/dscKeybusInterface/blob/master/extras/PCB%20Layouts/NodeMCU/pcb_screenshot.png

I used a 5V linear regulator (LM7805 with capacitors) to step down from the main 12V pins on the DSC board marked "+/- AUX", and connected accordingly to ground and 5V pin.

Did you try reading the voltage on the read leads after the voltage splitter to make sure it's close to 1.5V? (1.5 because the signal ideally flashes between 0V and 3V so most voltmeters should read ~1.5.) Are you sure the step down was set properly and stably to 3.3v or 5v and the transistor had B/C/E set properly? I almost messed that up because the one I used had a non-standard pin order...

mrmaddotta, 2020-01-09:
@zzhang1 thanks for the suggestions :) I have had a new D1 rewired and running just fine since I opened this issue and everything has been great (clean forgot I opened this issue!). I am going to write this off to me doing something stupid.

Out of curiosity, what capacitors are you using alongside the LM7805? I'd like to remove the clunky LM2596 from my setup and have a few 7805s lying around.

Dilbert66, 2020-07-31:
Just a note. Some of those cheap clone D1 units have undersized 3.3 voltage regulators that only supply 150ma.   I've had 2 die because of it.  You might need to get a 12v to 3.3 volt regulator and feed direct to the 3.3 pin instead of feeding at the 5v .

taligentx, 2020-09-30:
@mrmaddotta - good to hear the issue didn't recur with the replacement D1, closing the issue.

closed 2020-09-30 by taligentx