taligentx/dscKeybusInterface #123: Tested, working fine with PC5010 (aka DSC Power 832)

zzhang1, 2019-10-15:
Hi! Almost one year in, just wanted to report belatedly that everything works great on my PC5010 system. It is a single partition with a 24 of 32 zones in use. 

Working fine:
- Arming home/away and disarming
- Window sensors
- Door sensors
- Motion sensors
- Smoke alarms
- System trouble flag

Untested:
- Low temperature sensor

jmlzone, 2020-01-13:
I just started working with PC5010.  Also works fine.  Tried a little to see if I could get the zone tables from the lcd5500Z keypads, but could not make sense of the data when I told a keypad to broadcast the labels.

Additional to the list above, tried output 3 and output 4 from my expansion box, working fine.
my system has single partition, 2 LCD5500Z keypads, A 433 keypad, expansion box with a second power supply and PC5400 printer interface (I never configured it) expansion zone and outputs connected to other things in the house.

zlazar, 2020-01-20:
Hi everyone,
I tested on my old PC1555 and works fine. I have 2 partitions, and unfortunately the web interface has no implemented multi partition. So I installed a local Blynk server, but I'm confusing to get working together with NodeMCU V3 over WiFi. The Blynk server is running fine, I have the token key, added on Arduino ide these lines:
#include <BlynkSimpleEsp8266.h> or #include <Blynk.h>
char auth[]"xxxxxxxxxxxx"
char server[] = "192.168.2.XX";
but I'get errors:
conflicting declaration 'AsyncWebServer server'

Can anyone please help me with the original code used with Blynk? 
Thank you!

taligentx, 2020-09-30:
@zzhang1 @jmlzone Thanks for confirming the interface works with the PC5010!  I've updated the list of verified panels.

closed 2020-09-30 by taligentx

jmlzone, 2020-09-30:
Thanks for the easy plug and play code.  I will need o get back to my home control project that gies through the pc5010.  I have some expansion GPIO that I use to open/close my garage doors and I was also able to use the interface to actuate those!