taligentx/dscKeybusInterface #124: ARM support

otto76, 2019-10-30:
Hi, guys, do we have plans to support ARM boards, aka Arduino Due?
Should be some challenges with interrupt handling. 

But in general, is there anything else, why its put a side for now? 
O.

taligentx, 2020-09-30:
Hi @otto76 - late response but thanks for the suggestion, is there a particular advantage of the Due vs the ATmega-based Arduino boards in this application?

I like supporting the Uno, etc as the boards are commonly available at local electronics shops worldwide, but the $3-$5USD esp8266/esp32 boards with integrated WiFi are more capable than the $40 Due board.

otto76, 2020-09-30:
Hey, I prefer wired solution instead for wireless for security stuff, therefore dont like esp<xx> board. 
Regular Uno, has very limited amount of memory available

As I have several other monitoring activities going on for the same board, then I am running out of memory, plus performance is lagging. I do have DUE on my hands, thus was interested for solution 

I tried to rewrite code for timers and other DUE specific stuff, but w/ very little success 
Therefore dropped question here :)
o.

Dilbert66, 2020-09-30:
esp32 does support ethernet .

taligentx, 2020-10-01:
Certainly understandable if you have the Due hardware on hand - if you or anyone else gets the interrupts and timers adjusted for the Due, I can merge a PR but it's not currently on my roadmap (if nothing else, I don't have a Due to test with).  

The Arduino Uno, esp8266 and esp32 can connect to the ENC28J60, W5100, and W5500 Ethernet modules over SPI so that's one way to go for a wired connection.

As for performance: https://forum.arduino.cc/index.php?topic=337653.0

The Due is great compared to the atmega328p, but the esp8266@160MHz and esp32 exceed the Due's performance, SRAM, and flash memory specs at 1/10th the cost.

I'll leave the issue open if someone is able to adjust the code for the Due.

closed 2020-11-02 by taligentx