taligentx/dscKeybusInterface #125: "Web virtual keypad" doesn't work for me

jgrosso2171, 2019-11-09:
Hi, sorry for my English, I'm using the google translator. I wanted to try "Web virtual keypad", but it doesn't work for me in the browser. I compile the program, but it doesn't work for me, in serial print it shows me
"........
WiFi connected
192.168.1.44
" and nothing more.
I have doubts if the libraries are being loaded correctly since in the IDE it shows me some in black, I enclose sending captures.

[image]
[image]

jgrosso2171, 2019-11-10:
I could solve it, my problem was with SPIFFS, thanks to this guide https://www.luisllamas.es/como-usar-el-spiffs-del-esp8266-con-el-arduino-ide/
[image]

DiegoMisa, 2019-12-04:
Hi, I have similar problem, But virtual keyboard dont send instrucctions to PC1832,   Can you change anything in your alarm panel?  i read abut *8 902  or section 377

regards

DiegoMisa, 2019-12-04:
[image]

BriceDet, 2020-04-13:
Hi @DiegoMisa, did you solve your trouble?

rverzinkevicius, 2020-04-20:
Hi. I have web virtual keyboard working, i can monitor zones, enter codes, but buttons like stay and chime doesn't work. I tried single click, click and hold, but it just doesn't work. Anyone have ideas/solution/hints?

NFrankzy, 2020-11-19:
Hello, I have a problem, the keyboard doesn't appear, the serial monitor show me this:

WiFi connected: 192.168.0.106
Web server started: http://dsc.local
DSC Keybus Interface is online.

I haven't changed the domain name but when I go to the browser show me that the page "http://dsc.local" is not available

Anyone can help me please?

[image]

taligentx, 2020-12-02:
>Hello, I have a problem, the keyboard doesn't appear, the serial monitor show me this:

WiFi connected: 192.168.0.106
Web server started: http://dsc.local
DSC Keybus Interface is online.

I haven't changed the domain name but when I go to the browser show me that the page "http://dsc.local" is not available

Hi @NFrankzy - are you able to access the web interface from http://192.168.0.106?  If this works, there may be an issue with mDNS on your network.

bofruehw, 2021-01-15:
"Web virtual keypad" doesn't work. HW HW-628 with CH340. Compiled, uploaded OK. Connected to WiFi, no ping, no web server. Serial monitor shows mess and sometimes something like:
          WiFi connected: 192.168.x.x (address is mainly wrong, different than assigned by router)
          Web server started: http://dsc.local
          DSC Keybus Interface is online.
I presume that I have to upload the correct firmware. If so, the question is what should I include?
If it is something else please be so kind and suggest.
Note: New HW-628 was out of the box and worked fine, ping ok.
Blink program was uploaded as for nodeMCU 1.0 - worked fine.
I updated everything and tried many options combinations.
On serial, I had different results, but I think that I'm missing something crucial and basic. Till now I worked with Wemos and Arduino but never something complicated.

THX

taligentx, 2021-01-16:
Hi @bofruehw,

There are a few possibilities - are you using the master or develop branch?  Did you upload the SPIFFS data? 

To make sure the interface to the DSC panel is working, try the `KeybusReader` sketch to make sure you're getting good data without errors.

>WiFi connected: 192.168.x.x (address is mainly wrong, different than assigned by router)

This is a strange issue - you can try erasing the entire flash when uploading the sketch (Arduino IDE > Tools > Erase Flash > All Flash Contents.  To check if WiFi works while the interface is active, try one of the MQTT sketches (HomeAssistant-MQTT, etc) and see if you're able to ping the esp8266 while the sketch is running.

bofruehw, 2021-01-18:
THX. I did everything ok, and I didn't find the right solution but around way.
I mostly solved this problem by the manual setup of IP, gateway, DNS, and subnet mask. Now this stage is faster and works ok even with sending the mail.

Unfortunately, I still have a major problem with speed and restarting when I am using android. Seems that Asinchronious WEB Server is quite unstable.

And one minor problem with OTA upload that doesn't function when DSC is connected (when I just power it on USB it is ok).

Q: Would it be faster if we cut the number of zones? I am using just 8.
Q: I have one zone that is always open. Is it possible to mark it open but ignored (may DSC is treating it as disabled)?

THX

kricon, 2021-01-18:
> Q: Would it be faster if we cut the number of zones? I am using just 8.

You can change settings in dscKeybusInterface.h (it doesn't change zones shown in VirtualKeypad-Web example):
```
const byte dscPartitions = 8;   // Maximum number of partitions - requires 19 bytes of memory per partition
const byte dscZones = 8;        // Maximum number of zone groups, 8 zones per group - requires 6 bytes of memory per zone group
const byte dscBufferSize = 50;  // Number of commands to buffer if the sketch is busy - requires dscReadSize + 2 bytes of memory per command
```
I don't know about what effects on speed it might have, I didn't yet tried the VirtualKeypad-Web example. But I have plans to, and hopefuly change number of zones shown in webpage as I don't need/use 64 zones. I didn't look much at the source so I don't know how hard it would be to do and if I will succeed in that so don't count on it.

> Q: I have one zone that is always open. Is it possible to mark it open but ignored (may DSC is treating it as disabled)?

I noted that in #190: `Worth noting: Zone leds will be green/open even if the zone is not associated to any partition (disabled from system), but their zone definition is not set to 000.`
Can you verify that zone definition (installer programming section `001`) for that zone is set to `00`? Disabling the zone in section `202` isn't enough as system still tracks open/close status of that zone, even if it isn't enabled in any partition.

bofruehw, 2021-01-18:
Thank you. I shall try with suggested changes, but I think an optimized web interface would be appreciated for the mobile.

kricon, 2021-01-31:
@bofruehw I've quickly tested `VirtualKeypad-Web` example and shorten zones from 64 to 16, example below:

Change line: `grid-template-rows: repeat(16, auto);`
To number of zones per column: `grid-template-rows: repeat(4, auto);`

Change line: `for (var i = 1; i <= 64; i++) {`
To number of zones you have: `for (var i = 1; i <= 16; i++) {`

You're making all changes in `/data/index.html` file.

I didn't tried, but it will be nice to put 1 row with 4 columns as example for showing PGM status, after zones.

bofruehw, 2021-01-31:
@kricon Thank you. I'll try.
I would like to add also logging. Now I'm trying with SPIFFS txt file, but I think that log file on web interface would be convenient. Is there any possibility of that?
Also, I added one Dallas DS 18B20 for ambient temperature. I can write it on display but there are only two lines.
Thank you in advence.

kricon, 2021-01-31:
I don't know about logging, as I've said, I'm pretty new when it comes to Arduino world, so I can't help you on that, sorry.
Just guessing, but placing something like box showing latest 30 logs in main web interface, just below zones, should be possible.

For displaying a temperature, you can use the "Event buffer info" line on upmost part of LCD, above the first line. It's more practical for showing temperature readings, and can be always updated (except when pauseZones is called as that line can then be used to print event buffer info) using `root["event_info"]`.

Actually, I was lately thinking of implementing temperature sensors on my project as well, as I have free pair of wires going to each sensor/keypad and placing 18B20 thermometer on that wires wouldn't took much place at all, it can fit into sensor or behind a keypad and I can remotely monitor temperatures from each sensor/keypad. I saw some project of connecting multiple 18BS20 sensors to same ESP input pin, which is nice but probably not needed as I plan to use ESP32 which have a lot free pins so no problem if 10 sensors take 10 pins. Or I can get DHT11/DHT22/AM2302 or even BME280 temperature/humidity sensor and hide it behind sensors, do you know if that's better solution? As I've never used ESP8266 before this project, neither temperature sensors with it, can you give me some hints or it's basically just coping tutorial and placing/integrating it into VirtualKeypad sketch?

bofruehw, 2021-01-31:
> I didn't tried, but it will be nice to put 1 row with 4 columns as example for showing PGM status, after zones.

Try this:
 div#zones_list {
     ...
        grid-template-rows: repeat(4, auto); // was16**
        ...
      }

kricon, 2021-01-31:
That's for adjusting how many zones per column, as I stated above and confirmed by editing it's value then checking it.
Then I get idea to implement PGM outputs for example, as a separate "box" below list of zones. It would probably need to place lot of new code, including "div#pgm_list". But thanks.

bofruehw, 2021-02-01:
By implementing cutting to 8 zones web, as described above, a server is not stable anymore. As I made several changes, I'm still testing, but if someone knows the trick please help, I have no experience with JSON, PHP,.., 
Besides changes in HTML I cut not used zones:
if (ws.count()) {
        char outas[512];
        StaticJsonDocument<200> doc;
        JsonObject root = doc.to<JsonObject>();
        root["open_zone_0"] = dsc.openZones[0];
        /*root["open_zone_1"] = dsc.openZones[1];
        root["open_zone_2"] = dsc.openZones[2];
        root["open_zone_3"] = dsc.openZones[3];
        root["open_zone_4"] = dsc.openZones[4];
        root["open_zone_5"] = dsc.openZones[5];
        root["open_zone_6"] = dsc.openZones[6];
        root["open_zone_7"] = dsc.openZones[7];*/
        serializeJson(root, outas);
        ws.textAll(outas);
      }
But it worked ok with 64 zones.
I'm pretty sure that it is a problem with a server...
THX

bofruehw, 2021-02-01:
> I saw some project of connecting multiple 18BS20 sensors to same ESP input pin

I made a project for logging in and out temperatures for flor heating with 16 and it worked fine with Uno. I shall upgrade it to ESP.

krishy2, 2021-06-29:
I have a very similar problem. MQTT example works, but web interface firmware isn't. Ping responds, but no any web content was served.
SPIFFS had some warnings (outdated) during compile.