taligentx/dscKeybusInterface #126: Switching from stay armed to night armed

bourgeois, 2019-11-14:
I'm trying to write a piece of code that would allow me to change between stay armed and night armed. I tried this but I'm getting a "System is unavailable" on my other keypads for few seconds and then it doesn't seem to work.... Am I missing something (any global boolean flag I need to change) ?

    // homebridge-mqttthing NIGHT_ARM from STAY_ARM
    if (payload[payloadIndex] == 'N' && dsc.armedStay[partition] && !dsc.exitDelay[partition] && !dsc.noEntryDelay[partition]) {
        dsc.writePartition = partition + 1;    // Sets writes to the partition number
        dsc.write("*9");  // Keypad arm with no entry delay
        publishState(mqttPartitionTopic, partition, "N", 0);
        exitState = 'N';
        return;
    }

Am I doing this right ? Would it better to handle this change at the library code level and not the sketch ?

duanemck, 2020-01-06:
@bourgeois Did you figure this out? I have a similar problem except I need to enter *1.
When I do, it seems like the panel is rebooted or something as the physical keypad loses its "Armed" LED for about 10 seconds and then comes back on.

There does appear to be a function in the library for night arming, but when I use it my alarm does not behave the same as "Stay" -> "*1"

bourgeois, 2020-01-08:
No, I never managed to make this work 😞

Let me know if you ever find something!

Dilbert66, 2020-07-21:
On my system you also need to send the access code after the *9 to go to night mode as an additional dsc.write (code)

closed 2020-11-02 by taligentx