taligentx/dscKeybusInterface #131: ESP8266 freezes after few days

tomiie26, 2020-02-06:
I made your project (VirtualKeypad-Blynk.ino), it is very useful and powerful. I use your ESP8266 blynk example, I havent modified it. The ESP sends data to my Blynk server which is on my local area network, working on an Orange PI Zero. I have also made your pcb with a 2n2222 transistor. Everything working fine for few days, 2 or 3 days. After the ESP freezes and need to hard reset. After reset it, working again. I have made some investigation, I have added a terminal and do some printing.  It say "Keybus buffer overflow". I increased the dscBufferSize in the library: src/dscKeybusInterface.h. Now I havent got buffer overflow, but it freezes as before. Does anybody have the same problem? Or yours working for months or years? I have 5v swithing regulator like this:
https://www.banggood.com/3A-UBEC-Module-Low-Ripple-Bluesky-Mini-Switch-Mode-DC-BEC-5V-12V-2-6S-Supply-p-993056.html?rmmds=search
, and I am powering the ESP on the Vin pin with 5V. 

Please help me!

tomiie26, 2020-02-06:
More info: I have DSC PC1616, and I used the latest code 1.2. My ESP exactly a Wemos D1 mini pro. But before I have the same issue with Nodemcu as well, thats why I have tried other board.
Thanks in advance.

lubster85, 2020-02-10:
Hi @tomiie26 - I recently also tried out this project using ESP8266 on NodeMCU (thank you @taligentx ). One difference is that I am powering on one of the 3.3v pins (not the VIN pin). It's only been running for 5 days and no issued so far.

I will let you know if I experience any issued as you've described.

Is there not a chance that it's heating up too much with the voltage drop?

javierferwolf, 2020-02-11:
Hi @tomiie26 and @lubster85 I have the same problem, using NodeMCU ESP8266. The problem happens after a day or two days of working correctly, for the ESP8266 to work again it is necessary to turn off and turn on again.   I have connected the ESP8266 through the serial port to the computer to monitor and know what type of error appears when the problem appears in the ESP8266. 
The problem is the following:

Exception (0):
epc1=0x40212d48 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffebc0 end: 3fffffb0 offset: 01a0
3fffed60:  401008dc 000001dc 000003e8 00000022  
3fffed70:  3fffc200 40100894 00000020 401008dc  
3fffed80:  3ffef584 3ffed63c 3fffc230 4000050c  
3fffed90:  40004858 00000030 00000020 ffffffff  
3fffeda0:  40106ec4 00e8a101 3feffe00 60000200  
3fffedb0:  fffdffff 04000001 fffffffe 04000001  
3fffedc0:  04000003 00000001 04000002 00000008  
3fffedd0:  00000000 00000000 60000600 00000030  
3fffede0:  3fffc200 00000000 00000000 40100abd  
3fffedf0:  00000014 3ffeeee0 3ffef450 4010017b  
3fffee00:  4021c9c4 00000030 00000010 ffffffff  
3fffee10:  4021c9b4 3fff2024 00000000 00000020  
3fffee20:  40100c29 00000005 00000000 40100c18  
3fffee30:  00000000 00000000 0000001f 4010085c  
3fffee40:  3fffc200 40100894 3fffc228 40105981  
3fffee50:  4000050c 40100b54 3fffc258 4000050c  
3fffee60:  40000f68 00000030 00000017 ffffffff  
.
.
.
.
3fffff30:  4f424f20 4e4f5a20 31333700 39343436  
3fffff40:  80000032 3fff7600 0054005f 8026002f  
3fffff50:  3fff1da4 0054005f 0022a398 3ffef524  
3fffff60:  00000014 00000000 4bc6a7f0 00000000  
3fffff70:  00000000 00000001 40100a2a 19999999  
3fffff80:  00000014 00000000 00000001 3ffef524  
3fffff90:  3ffeed0c 3ffeed38 3ffef4e4 40201d56  
3fffffa0:  3fffdad0 00000000 3ffef4e4 40210e00  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1392, room 16 
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

I put the error in the ESP exception Decoder and write the following:

Exception 0: Illegal instruction
PC: 0x40212d48: dscKeybusInterface::redundantPanelData(unsigned char*, unsigned char volatile*, unsigned char) at /home/javier/Arduino/libraries/dscKeybusInterface-master/src/dscKeybusInterface.cpp line 390
EXCVADDR: 0x00000000

Decoding stack results
0x401008dc: timer1_isr_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_timer.cpp line 43
0x40100894: timer1_isr_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_timer.cpp line 36
0x401008dc: timer1_isr_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_timer.cpp line 43
0x40100abd: __digitalWrite(uint8_t, uint8_t) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring_digital.cpp line 86
0x4010017b: dscKeybusInterface::dscClockInterrupt() at /home/javier/Arduino/libraries/dscKeybusInterface-master/src/dscKeybusInterface.cpp line 447
0x4021c9c4: tcpip_tcp_timer at core/timeouts.c line 153
0x4021c9b4: tcpip_tcp_timer at core/timeouts.c line 151
0x40100c29: interrupt_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring_digital.cpp line 173
0x40100c18: interrupt_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/interrupts.h line 29
0x4010085c: ets_post(uint8, ETSSignal, ETSParam) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_main.cpp line 160
0x40100894: timer1_isr_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_timer.cpp line 36
0x40100b54: interrupt_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring_digital.cpp line 135
0x40100a2a: millis() at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring.cpp line 188
[... 96 lines cut ...]
0x40100abd: __digitalWrite(uint8_t, uint8_t) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring_digital.cpp line 86
0x4010017b: dscKeybusInterface::dscClockInterrupt() at /home/javier/Arduino/libraries/dscKeybusInterface-master/src/dscKeybusInterface.cpp line 447
0x4010017b: dscKeybusInterface::dscClockInterrupt() at /home/javier/Arduino/libraries/dscKeybusInterface-master/src/dscKeybusInterface.cpp line 447
0x40100c29: interrupt_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring_digital.cpp line 173
0x40100c18: interrupt_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/interrupts.h line 29
0x40100894: timer1_isr_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_timer.cpp line 36
0x401008dc: timer1_isr_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_timer.cpp line 43
0x40100b54: interrupt_handler(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring_digital.cpp line 135
0x40211214: delay_end(void*) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/core_esp8266_wiring.cpp line 41
[... cut ...]
0x4020ff48: String::copy(char const*, unsigned int) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/WString.cpp line 214
0x4021012c: String::operator=(String const&) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/WString.cpp line 262
0x4021015c: String::String(String const&) at /home/javier/.arduino15/packages/esp8266/hardware/esp8266/2.6.3/cores/esp8266/WString.cpp line 41

More info: I have DSC PC585, and I used the latest code 1.2. too.  Core Version: 2.6.3
some opinion of what would be happening?

tomiie26, 2020-02-11:
Hi @lubster85 and @javierferwolf Thank you for your reply. Mine is working now for 5 days wihout any problem, I had only 2 disconnection, but within a minute it is connected again. My config right now is D1 Mini Pro, that I wrote in my first post. It could be the power supply maybe, and the buffer overflow, but I didnt monitor the errors before like @javierferwolf did. I am waiting few more days if the issue comes again I will do a deep troubleshooting as well and inform you about that. One more thing: I monitor the wifi rssi strength through the loop, so maybe it is a source of the issue. But now as I mentioned it is working for 5 days without problem.
@javierferwolf we should start the investigation with the power supply, somehow we should monitor the voltage and the consupmtion, not for the momentary, I mean we should do it for days. Maybe we have momentary power failure that we dont know. Do you have Blynk server too?
You know I have other projects with ESP8266 and DS18 and DHT temperature sensor, and it is working for years now without any problem. I trust the ESP boards, I think there is only a little bug somewhere.

tomiie26, 2020-02-11:
Other thing about ESP boards, they are very sensitive to 220v inductions. One time I had same issue with Nodemcu, it was a single temperature sensor with relay module, and I figured out when the relay switched the 220v thing the ESP immediately freezed. The relay module was next to the ESP, and when I put it at least 1 meter far away, the issue was gone away. So in my first config the ESP was in my alarm system's box, that have GSM module and 220v power supply as well. But now I put far away from the box and it seems now I have no issue.

javierferwolf, 2020-02-15:
Hi @tomiie26 My project is a modification of the Pushbullet.ino example. but instead of using the Pushbullet service I use Viber for communication with the panel. I have tried with different power supplies, even the ESP8266 is located a safe distance from the panel adapter 16VA, but the problem persists! you said that "you had only 2 disconnection, but within a minute it is connected again". In my case after the error and it freezes, the ESP also tries to connect again, but the ESP cannot connect because it has a soft WDT reset that is caused by the line of code "while (WiFi.status() != WL_CONNECTED) delay(100);"  I can know this because in the "Exception decoder" write in which line is the error of soft WDT reset.  I will try to give ESP less time "delay(10)" and try again.  But the main problem is why every two days or three ESP freezes, with the same problem.

Exception 0: Illegal instruction
PC: 0x40212d48: dscKeybusInterface::redundantPanelData(unsigned char*, unsigned char volatile*, unsigned char) at /home/javier/Arduino/libraries/dscKeybusInterface-master/src/dscKeybusInterface.cpp line 390
EXCVADDR: 0x00000000

jimtng, 2020-02-16:
If you suspect the power supply is the issue, try just powering your esp8266 with a separate USB power supply. That's what I am using for my installation, and it has been running smoothly for at least 6 months. The added benefit is I can easily power cycle the esp8266 if necessary, without opening my alarm box.

l33d, 2020-02-22:
Hi, I have this problem  too, but firstly i solder all connection, problem partially disappear. Sometimes when using virtualblynk keyboard it freezes and must restart to use keyboard again, but information from dsc system works.

tjtjdgur2719, 2020-02-23:
I need help with wiring with wemos D1 R1..
I dont know if I am suppose to use arduino or esp8266 ino file(I'm using home assistant)
because it's both I'm guessing I have to combine the two ino file somehow.

also instead of resistors, can I just use dc buck step down converter(+/- input and set it to 3.3v +/-output). I have trouble connecting dsc yellow and green.

fariddergal, 2020-08-15:
Hi,

I am using ESP8266 D1 Mini Pro, with ext antena,  and I am having the same issue, as it freezes after some days, does any body could solve this issue?  do you know how to enable a watchdog to reset the module  when it gets frozen? this could solve the problem.

taligentx, 2020-10-02:
Hi @tomiie26,

>Everything working fine for few days, 2 or 3 days. After the ESP freezes and need to hard reset. After reset it, working again. I have made some investigation, I have added a terminal and do some printing. It say "Keybus buffer overflow".

>So in my first config the ESP was in my alarm system's box, that have GSM module and 220v power supply as well. But now I put far away from the box and it seems now I have no issue.

Good to hear the issue was resolved by moving the esp8266, these kinds of issues can be tricky to track down, congrats!

As far as the buffer overflow - I'm curious if there's a communication issue between the ESP and the Blynk server that's causing timeouts and the Keybus buffer overflow - on the esp8266/esp32, the buffer is set to 50 commands, which is quite large and in normal use I've never seen it overflow unless `dsc.loop()` was not being called for extended periods.

@javierferwolf @fariddergal,

To narrow down the issue, I recommend running the `Status` example sketch and monitoring the serial output to see if there's still a lockup after a few days - this would help determine if it's a code issue, hardware issue (bad flash on the esp8266/esp32, etc), wiring issue (interference, etc), or a network/timeout issue.  I also recommend erasing the flash completely before uploading the `Status` sketch.  In the Arduino IDE: Tool > Erase Flash > All Flash Contents, or if using CLI: ` esptool.py erase_flash`.

If the `Status` sketch runs without issues, you can start adding in the other components to see exactly which component is causing the lockups.

taligentx, 2020-10-05:
I noticed an issue that could be causing the crash on `redundantPanelData()` - this function should be in the esp8266 and esp32 IRAM because it's being called from an interrupt function that's already in IRAM.  I thought I had updated this when I changed the interrupt to use `redundantPanelData()` but clearly I missed it.  Try the latest develop branch version to see if it helps with this specific crash.

taligentx, 2021-01-22:
This fix is now part of the newly released version 2.0, feel free to re-open or post a new issue if crashes are still occurring.

closed 2021-01-22 by taligentx