taligentx/dscKeybusInterface #134: Partition 1 topic occasionally stays on "pending" (exitDelay) after arming

automatikas, 2020-03-02:
It's been over 6 months I interfaced POWER1864 all works like a charm. But occasionally mqttPartitionTopic topic stays "pending" and "armed_home" or "armed_away" is not published when panel is armed. Never faced problem with disarming. 

runs on old arduino leonardo+eth shield.

Anyone experienced the same issue?

taligentx, 2020-09-29:
Hi @automatikas - late response, I haven't seen other reports of this, though I haven't tested much with Arduino and networking.  I much prefer ESP8266/ESP32 as the Arduino becomes strained on resources and that could be a factor.  I'll keep the issue open for any other Arduino with networking users to chime in.

automatikas, 2020-10-08:
Hi @taligentx 
I thought it could be related to an old Leonardo board. as a sketch was optimised to around 99% of memory. I finally got my Wemos R1 D2 board with ESP8266. Uploaded yesterday and all was working great util this morning when i found retained armed_stay status on MQTT but the system was disarmed. In my situation I also need to use entry and exit delay status to involve other logics in my house automation. Entry and exit delay messages comes out ok, but not always disarmed and armed messages. As this is the same issue on two different platforms i feel that something could be related to my code. Mabe messages buffering is not working? 

I haven't seen any other problems. Zones publishes status, control over mqtt works fine as well. Strange...

This is may code:

```
WiFiClient wifiClient;
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <dscKeybusInterface.h>

// Settings
const char* wifiSSID = "myWifiSSID";
const char* wifiPassword = "myWifiPass";
const char* accessCode = "1234";    // An access code is required to disarm/night arm and may be required to arm based on panel configuration.
const char* mqttServer = "192.168.5.45";    // MQTT server domain name or IP address
const int mqttPort = 1883;      // MQTT server port
const char* mqttUsername = "myMQTTuser";  // Optional, leave blank if not required
const char* mqttPassword = "myMQTTpassword";  // Optional, leave blank if not required

// MQTT topics
const char* mqttClientName = "dscKeybusInterfaceV2";
const char* mqttPartitionTopic = "dsc/partition";  // Sends armed and alarm status per partition: dsc/Get/Partition1 ... dsc/Get/Partition8
const char* mqttZoneTopic = "dsc/openZone/zone";            // Sends zone status per zone: dsc/Get/Zone1 ... dsc/Get/Zone64
const char* mqttAlarmZoneTopic = "dsc/alarmZone/zone";            // Sends Alarm status per zone: dsc/Get/alarmZone1 ... dsc/Get/Zone64
const char* mqttFireTopic = "dsc/fire";            // Sends fire status per partition: dsc/Get/Fire1 ... dsc/Get/Fire8
const char* mqttTroubleTopic = "dsc/trouble";      // Sends trouble status
const char* mqttTroubleAcTopic = "dsc/trouble/ac";      // Sends AC power status
const char* mqttTroubleBatteryTopic = "dsc/trouble/battery";      // Sends battery status
const char* mqttStatusTopic = "dsc/status";
const char* mqttKeyBusStatusTopic = "dsc/status/keybus";
const char* mqttKeyBusOverflowStatusTopic = "dsc/status/keybus/overflow";
const char* mqttStatusKey
PubSubClient mqtt(mqttServer, mqttPort, wifiClient);

// Configures the Keybus interface with the specified pins - dscWritePin is optional, leaving it out disables the
// virtual keypad.
#define dscClockPin D1  // esp8266: D1, D2, D8 (GPIO 5, 4, 15)
#define dscReadPin D2   // esp8266: D1, D2, D8 (GPIO 5, 4, 15)
#define dscWritePin D8  // esp8266: D1, D2, D8 (GPIO 5, 4, 15)
dscKeybusInterface dsc(dscClockPin, dscReadPin, dscWritePin);

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  WiFi.mode(WIFI_STA);
  WiFi.begin(wifiSSID, wifiPassword);
  while (WiFi.status() != WL_CONNECTED) delay(500);
  Serial.print("WiFi connected: ");
  Serial.println(WiFi.localIP());

  mqtt.setCallback(mqttCallback);
  if (mqttConnect()) mqttPreviousTime = millis();
  else mqttPreviousTime = 0;

  // Starts the Keybus interface and optionally specifies how to print data.
  // begin() sets Serial by default and can accept a different stream: begin(Serial1), etc.
  dsc.begin();

  Serial.println(F("DSC Keybus Interface is online."));
}

void loop() {
  mqttHandle();

  if (dsc.handlePanel() && dsc.statusChanged) {  // Processes data only when a valid Keybus command has been read
[... 303 lines cut ...]
      mqttPreviousTime = mqttCurrentTime;
      if (mqttConnect()) {
        if (dsc.keybusConnected) mqtt.publish(mqttStatusTopic, mqttBirthMessage, true);
        Serial.println(F("MQTT disconnected, successfully reconnected."));
        mqttPreviousTime = 0;
      }
      else Serial.println(F("MQTT disconnected, failed to reconnect."));
    }
  }
  else mqtt.loop();
}

bool mqttConnect() {
  if (mqtt.connect(mqttClientName, mqttUsername, mqttPassword, mqttStatusTopic, 0, true, mqttLwtMessage)) {
    mqtt.publish(mqttStatusTopic, mqttBirthMessage, true);
    Serial.print(F("MQTT connected: "));
    Serial.println(mqttServer);
  }
  else {
    Serial.print(F("MQTT connection failed: "));
    Serial.println(mqttServer);
  }
  return mqtt.connected();
}
```

taligentx, 2021-01-05:
Great to hear, and thanks for the follow-up!