taligentx/dscKeybusInterface #135: Telegram bot

PascalSI, 2020-03-09:
Hi, no one tried to use this library with the Telegram bot to control the alarm and receive the event?
I want to try, but not knowing where to start ... I am not very good at programming.

taligentx, 2020-09-29:
Hi @PascalSI - Iate response, you can find an example of using ESP8266 with Telegram at:
https://www.instructables.com/id/Telegram-Bot-With-ESP8266/

You can look at the built-in PushBullet example sketch and modify it for Telegram based on the above example.  If you get this working, feel free to add a PR and I'll add it to the examples.

taligentx, 2020-12-22:
@PascalSI I've added a Telegram bot example sketch for esp8266 and esp32 that allows for control (arm/disarm) through the bot.  If there are any issues with the sketch, I'll leave this issue open for feedback.

PascalSI, 2020-12-31:
I got a problem when compiling, it gives this error:
`
C:\Users\User\Documents\Arduino\sketch_dec31a\sketch_dec31a.ino: In function 'void loop()':
sketch_dec31a:134:9: error: 'class dscKeybusInterface' has no member named 'pauseStatus'
     dsc.pauseStatus = false;
         ^
sketch_dec31a:140:9: error: 'class dscKeybusInterface' has no member named 'pauseStatus'
     dsc.pauseStatus = true;
         ^
sketch_dec31a:154:7: error: 'class dscKeybusInterface' has no member named 'loop'
   dsc.loop();
       ^
sketch_dec31a:183:15: error: 'class dscKeybusInterface' has no member named 'disabled'
       if (dsc.disabled[partition]) continue;
               ^
exit status 1
'class dscKeybusInterface' has no member named 'pauseStatus'`

Could you tell me why this might be?

taligentx, 2021-01-01:
Which version of the library are you using?  The Telegram bot will only work with the current `develop` branch, you'll need to download this branch manually and place it in the Arduino libraries directory.

taligentx, 2021-01-22:
The `Telegram` sketch is released in the new library version 2.0, feel free to re-open if there are still issues after updating the library.

closed 2021-01-22 by taligentx