taligentx/dscKeybusInterface #140: Alarm control panel Issue only showing state on Home Assistant

tjtjdgur2719, 2020-05-11:
I managed to setup everything using wemos d1 mini with home assistant.
Everything works perfectly with all the binary sensors showing up and running.
However, alarm_control_panel (Security System Partition 1) on home assistant does show up and is not unavailable state and if i use physical dsc keypad in the house it shows up the changing state but if i try to click arm on home assistant it does nothing .. the code is correct and matching dsc password. 
Any help?? Thanks

TrendMend, 2020-10-21:
Did you ever figure this out? Thank you.

tjtjdgur2719, 2020-10-21:
Sadly no..plus regular sensor state gets frozen after few days have to restart every time

taligentx, 2020-10-21:
Hi @tjtjdgur2719,

> (..) but if i try to click arm on home assistant it does nothing 

Does sending keys work with the `KeybusReader` sketch?  You should be able to type in a key via serial terminal, have the panel recognize the keypress, and see the key output in the keybus data - this verifies that the wiring for writing is functional.

> .plus regular sensor state gets frozen after few days have to restart every time

For esp8266/esp32, try the current develop branch - there was a bug that could cause crashes.  If on the current develop branch, does the esp8266 stop publishing updates to MQTT?  Can you still ping it after it becomes unresponsive in HA?

TrendMend, 2020-10-21:
Okay so funny thing, I was actually able to make it work by sending
commands via mqtt explorer, and swapping the arm disarm codes in home
assistant. By doing so when I pressed disarm it armed and caught the state.
Then I was able to swap them back and it is functioning now.

taligentx, 2020-10-21:
HI @TrendMend - good to hear it was resolved, odd that the states were out of sync but feel free to post an issue if that happens again.

@tjtjdgur2719 - another possibility specific to the Wemos D1 - some of the clones have low current voltage regulators and can cause reboots/crashes:

https://www.letscontrolit.com/forum/viewtopic.php?t=6603

You can try using something like one of the LM2596 voltage regulator modules set to 3.3v and connect directly to the Wemos 3.3v pin to bypass the onboard regulator.  I've updated the develop branch README.md to note this as a common issue.

tjtjdgur2719, 2020-10-21:
Wow Thank you @taligentx @TrendMend for the detail info & support!

I will try out all the suggestions and see if it works. 

Im currently out in town for a while so I will post an update when I get to work on it.

Thanks!

closed 2020-11-02 by taligentx