taligentx/dscKeybusInterface #141: Partial Armed message  Armed with Bypassed zones

luk2009, 2020-05-15:
Is it possible to get a message to be sent when the alarm is armed with bypassed zones?

01110101 0 00000000 01110101 [0x75] Partition 1 | Tone: off
    00000101 0 10000011 00010101 10010001 11000111 [0x05] Partition 1: Ready Armed Backlight - Arming with bypassed zones | Partition 2: disabled
    01110101 0 00010001 10000110 [0x75] Partition 1 | Tone: 1 beep, 1s interval
    00000101 0 10000011 00001000 10010001 11000111 [0x05] Partition 1: Ready Armed Backlight - Exit delay in progress | Partition 2: disabled
    10100101 0 00010001 00101101 01101111 00100000 00000000 00000000 01110010 [0xA5] 2011.11.11 15:08 | Timestamp
    01110101 0 00110001 10100110 [0x75] Partition 1 | Tone: 3 beeps, 1s interval
    00100111 0 10000010 00000101 10010001 11000111 00000000 00000110 [0x27] Partition 1: Armed Backlight - Armed away | Partition 2: disabled | Zones 1-8 open: none
    00010110 0 00001110 00100011 11110001 00111000 [0x16] PC1555MX | Zone wiring: NC | Armed, Exit *8 +15s, Power-on +2m
    10100101 0 00010001 01101101 01101111 00100000 10111110 00000001 01110001 [0xA5] 2011.11.11 15:08 | Partition 1 | Armed partial: Zones bypassed
    01110101 0 00000000 01110101 [0x75] Partition 1 | Tone: off
    00000101 0 10000010 00000101 10010001 11000111 [0x05] Partition 1: Armed Backlight - Armed away | Partition 2: disabled
    10100101 0 00010001 01101101 01101111 00100001 10110011 00000000 01100110 [0xA5] 2011.11.11 15:08 | Partition 1 | Zone bypassed: 4
    10100101 0 00010001 01101101 01101111 00100000 10011011 00000001 01001110 [0xA5] 2011.11.11 15:08 | Partition 1 | Armed by user code 3
    10100101 0 00010001 01101101 01101111 00100010 10011011 00000000 01001111 [0xA5] 2011.11.11 15:08 | Partition 1 | Armed: away
    11010101 0 10101010 10101010 10101010 10101010 10101010 10101010 10101010 10101010 [0xD5] Keypad zone query
    00100111 0 10000010 00000101 11111111 11111111 00000000 10101100 [0x27] Partition 1: Armed Backlight - Armed away | Zones 1-8 open: none
    00010110 0 00001110 00100011 11110101 00111100 [0x16] PC1555MX | Zone wiring: NC | Interval 4m

Dilbert66, 2020-07-22:
[The library code always captures all messages for the 0x05 cmd in the dsc.status[partition] array. You can capture the dsc.status messages and look for 0x15(armed_with_bypass) , 0xa0  (zone_bypass_used) for each partition in whatever code you use.

taligentx, 2020-09-29:
As @Dilbert66 mentioned, checking `dsc.status[partition]` in your sketch would work, the library does not have a dedicated variable that tracks armed with bypassed zones status.  If there is a common use case for this status, let me know and I can look into adding a dedicated status variable for this.

closed 2020-11-02 by taligentx