taligentx/dscKeybusInterface #142: Arduino freeze after long run - known problem?

danergo, 2020-05-17:
Hi,

I'm using an arduino Uno for this, but after a long time (weeks if not months) it seems unresponsive and the only way to recover is to restart it with power recycle.

Is it possible that `dsc.handlePanel();` never returns? I put it into my `loop`, but loop does not seem working any more.

javierferwolf, 2020-05-28:
@danergo hi, I have the same problem, What do you mean dsc.handlePanel(); never returns?  Have you been able to check what type of error appears at the time the arduino freezes?

danergo, 2020-05-29:
@javierferwolf hello, no unfortunately the board is abroad and due to limitations on travel I had no chance to check anything. It just happened two times, but for the first one I just recycled the power and then it was fine again. Then for second time I just only know that the same thing happened but nothing much.

My board is Arduino Uno Wifi Rev2 anyway. Please let me know if you can dig deeper than me. I'm interested. :)

javierferwolf, 2020-06-01:
hello @danergo I have been monitoring the problem and the moment it freezes, I get an error, that through ESP exception Decoder write the following:

Exception 0: Illegal instruction
PC: 0x40212d48: dscKeybusInterface::redundantPanelData(unsigned char*, unsigned char volatile*, unsigned char) at /home/javier/Arduino/libraries/dscKeybusInterface-master/src/dscKeybusInterface.cpp line 390
EXCVADDR: 0x00000000

I don't know what the problem may be!

Dilbert66, 2020-07-31:
If a * command is sent to an invalid or disabled partition , it will hang in a wait loop waiting for the corresponding 9E command.

taligentx, 2020-09-29:
I haven't run across crashes long-term with a generic Arduino Uno board, but I haven't tested the wifi Arduino boards - I prefer ESP8266/ESP32 for applications that require wifi.  The Arduino is tight on resources and wifi networking could be stretching the memory. 

You can also try the develop branch (I know, I know - I need to push this as a release) and see if there is any difference.

danergo, 2020-09-30:
> The Arduino is tight on resources and wifi networking could be stretching the memory.

Arduino Uno Wifi incorporates a Wifi module by U-blox (similar to ESPxy) - with 32bit processor -, and the memory-intensive tasks are handled by this module, arduino really just sees a simple basic usart.

On the side:
Arduino creators should shame on them, they changed the ATMega328 (which was on the standard Arduino Uno) to ATmega4809, which does not have the common AVR registers anymore.

It's a real pain***as* to have a basic timer running, so I needed to modify your code a little bit to have the interval running correctly.

Conclusion:
I'm planning to update to devbranch, for now I did a long-time test without the DSC functionality - keeping only the Wifi up, and it seems working correctly. We can keep this issue open, I'll update it once I switch to devbranch, and hopefully my issue is getting solved just by doing that.

taligentx, 2020-10-02:
> I'm planning to update to devbranch, for now I did a long-time test without the DSC functionality - keeping only the Wifi up, and it seems working correctly. We can keep this issue open, I'll update it once I switch to devbranch, and hopefully my issue is getting solved just by doing that.

To narrow down the issue, I recommend running the `Status` example sketch and monitoring the serial output to see if there's still a lockup - this would help determine if it's a code issue, hardware issue, wiring issue (interference, etc), or a network/timeout issue.  If the `Status` sketch runs without issues, you can start adding in the other components to see exactly which component is causing the lockups. Given the long time between lockups, it may only be practical with a second Arduino running in parallel.

taligentx, 2021-01-22:
Library has been updated to version 2.0, feel free to re-open if the crashes remain on this version.

closed 2021-01-22 by taligentx