taligentx/dscKeybusInterface #145: Gerber and Ascii drill files

Juanko84, 2020-07-23:
Hi, 

I sent the included pcb files in the download to a manufacturer. They came back and said they cant use the files, they need the Gerber and Ascii drill files. Any chance this is available and can be sent to me please?

taligentx, 2020-08-13:
Hi @Juanko84 - the PCB files were contributed by @sj-louw in #21, #22, #23, and #24.  I haven't looked into the manufacturing aspect - is there a way to convert the included files to the format needed?

Juanko84, 2020-09-07:
Hi Nikhil,

I did convert the files to be used by pcb manufacturers. I will forward them.

I have a Question on the DSC Status, I see it can be used as a topic, but its not implemented, how can I use this status as a home assistant property?

I have an envisalink aswell connected to my dsc alarm and I get the below dsc status, can this be achieved with the availability_topic: "dsc/Status" in your setup? Have you got an example for this?

[cid:image001.jpg@01D68500.5A2CFBC0]

[cid:image005.jpg@01D68500.5A2CFBC0]

Kind Regards,

Juan Korb
Technical Business Analyst (int)
IoT.nxt Head Office
[cid:IoT.nxt_Registered-Landscape-Logo_-Grey_150x57px_8da86770-74bf-456c-8d61-db90f90523a3.png]

Address: Bylsbridge Office Park, Building 14, Block C, 1st Floor, Alexandra Road, Highveld Ext 73, Centurion, Gauteng, 0046, South Africa
Phone: +27 12 880 0114
Email: juan.korb@iotnxt.com<mailto:juan.korb@iotnxt.com>
Website: https://www.iotnxt.com

[cid:facebook_7fa50cc5-bd2e-456e-b71e-bcc6975537f7.png]<https://www.facebook.com/iotnxt/>  [cid:linkedin_de143d75-2ff5-460f-b904-2d40ec2e8b51.png] <https://www.linkedin.com/company/iot.nxt>   [cid:twitter_84ab30d9-fec7-4e63-bb5e-8291caf481e8.png] <https://twitter.com/iot_nxt>   [cid:youtube_7ef73685-9dd4-488f-8407-84b0638dac87.png] <https://www.youtube.com/c/IoTnxt>  [cid:Location-Icon_68f5d1d7-c4ad-447e-8890-cfc8f35cf158.png] <https://bit.ly/2vzWxgi>

<https://www.iotnxt.com/contact-us/><https://registrations.corpcam.com/vodacom/registration.aspx?e=&pname=Vodacom03092020>[cid:SIS2040-VBHealthMailbanner_86d134a0-4ec7-4644-baa7-eef383fdd321.png]<https://healthwebcast.vodacombusinessevents.co.za/>

Disclaimer and Confidentiality Notice: IoT.nxt
The information contained in this e-mail communication is confidential. This information is intended solely for the use of the intended recipient named above. If you have received this e-mail communication in error, please notify the sender  juan.korb@iotnxt.com<mailto:juan.korb@iotnxt.com?subject=Disclaimer%20error,%20email%20received%20by%20mistake>  immediately by e-mail if you have received this e-mail by mistake and delete this e-mail from your system. In the event that you are not the intended recipient, you are hereby notified that any disclosure, distribution, use of, printing, copying or other action on the contents of this e-mail communication is strictly prohibited. This e-mail communication and its contents are not intended to be legally binding on IoT.nxt, its affiliates or the sender hereof nor should the contents hereof be construed to amend any contractual arrangement in place, unless explicitly stated herein. IoT.nxt accepts no liability for any loss or damage, whether direct, indirect, consequential or economic, however caused, which may result directly or indirectly from this e-mail communication or from any files attached hereto. Save for bona fide company matters, IoT.nxt does not accept any responsibility for any material, views or opinions contained in this email. This email is further subject to the provisions of the Protection of Personal Information Act 4 of 2013.

taligentx, 2020-09-07:
Hi @Juanko84 - good to hear about the board file!  If they work out and you'd like to contribute, you can add a pull request with the converted files or attach the files here.

Your screenshots aren't showing up, what kind of information are you looking for in the dsc/Status MQTT topic?  Currently the examples send "online" and "offline" when the Keybus is connected/disconnected.

Juanko84, 2020-09-07:
Hi @taligentx 
Can you see the image now? This is the keypad status that the envisalink is reading from the dsc panel, so we know that all those statuses are available to read. This helps to see when the alarm is not arming, that there is an open zone or zone trouble.

[image]

borjaburgos, 2020-11-09:
Hi there! I would also be interested in getting manufacturable formats for the PCB. Something like Gerber would be great. Alternatively, KiCad or EagleCAD would work too. I'm personally interested in the NodeMCU version. 

Thanks!

jvitkauskas, 2020-12-01:
You can generate Gerber files with Eagle for D1 mini shields https://github.com/taligentx/dscKeybusInterface/tree/master/extras/PCB%20Layouts/WemosD1Mini

heytcass, 2021-02-21:
> Hi there! I would also be interested in getting manufacturable formats for the PCB. Something like Gerber would be great. Alternatively, KiCad or EagleCAD would work too. I'm personally interested in the NodeMCU version.
> 
> Thanks!

I am also interested in the NodeMCU version. Was anyone able to convert those to a Gerber file? Looking to send to PCBWay.

borjaburgos, 2021-02-25:
Hey @heytcass I ended up creating my own board, but I went with the D1 mini and an Adafruit MPM360 for power regulation. Let me know if you'd be interested.

[image]
[image]

heytcass, 2021-02-25:
Hey @borjaburgos , that board looks great! And I also have some D1 minis lying around, so I'd be very interested. Thanks!

borjaburgos, 2021-02-25:
@heytcass if you live in the US I'm happy to mail you one PCB – I made some extras that I do not need. My email is my GH username at gmail.com.

neohusky, 2021-03-12:
Hey @borjaburgos 
Any chance you have a spare board to ship down under (AUS).  It would probably be cheaper for me to get it shipped here rather than order a batch.

borjaburgos, 2021-03-16:
@neohusky shoot me an email

Slikkster, 2021-05-19:
Hey @borjaburgos any chance you have a spare board you would be willing to ship to Canada? cheers