taligentx/dscKeybusInterface #146: PC-5016 no alarm trigger?

BriceDet, 2020-07-28:
Really good job is done on this project.
I need your advice: using MQTT example or virtualKeypad on esp8266 and arduino, everything looks wonderful.
But, never the alarm trig status is return (1A, 1S, 1P, but never 1T).
The DSC central is a PC-5016.

If you have no idea, What do you suggest to me to try?

Is it a compatibility trouble with 5016 ?

Thx

BriceDet, 2020-07-31:
Nobody have an idea ? 😥

Dilbert66, 2020-07-31:
Without knowing what the panel sends, it's impossible. use one of the test sketches and print some debug info. Specifically, the paneldata fields.

BriceDet, 2020-07-31:
By 1T I mean (in the MQTT example) the partition1 alarm trigger.

I'll try to trig an alarm with the Wemos plugged with the KeybusReader.
I'll let you know the test results.

BriceDet, 2020-08-09:
What I capture (see attached)
[new 2.txt](https://github.com/taligentx/dscKeybusInterface/files/5048312/new.2.txt)

At the timestamp 9439,35 the alarm is triggered, but nothing catch by dscKeybusProcessData.

What I understant : 
The alarm "zone" status is grep on the command [0xA5], byte 3
_dscKeybusProcessData.cpp_
`void dscKeybusInterface::processPanel_0xA5() {
  (...)
  byte partition = panelData[3] >> 6;
  switch (panelData[5] & 0x03) {
    case 0x00: processPanelStatus0(partition, 6); break;
    case 0x02: processPanelStatus2(partition, 6); break;
  }
}`

With my central (PC-5016), the "final" switch result is 0x01 => not computed.
Did i miss something?

Thank for your advice

taligentx, 2020-08-10:
Hi @BriceDet - your `KeybusReader` sketch output indicates that the alarm triggered data is being decoded successfully:

```
9438.78: 10100101 0 00100000 01100000 01110000 10001000 00001010 11111111 00100110 [0xA5] 2020.08.03 16:34 | Partition 1 | Zone alarm: 2
...
9439.35: 10100101 0 00100000 01100000 01110000 10001000 01001011 11111111 01100111 [0xA5] 2020.08.03 16:34 | Partition 1 | Partition in alarm
```

I'm not seeing anything unusual in the KeybusReader output, you can try capturing the output of the `Status` sketch to check if the alarm triggered status is being set.  If so, the issue is likely on the sketch side rather than the library itself.

As far as how the library interprets the status data - the 0xA5 command contains status for partitions 1 and 2, and the 0xEB command contains status for partitions 1-8 (this seems to have been introduced once DSC had panels supporting more than 2 partitions).  The status data is contained in a single byte, but in different locations - byte 6 for 0xA5 commands and byte 8 for 0xEB commands.  The status data also seems to be split up into different sets as there are more status states than possible to store in a single byte - the set of status data is determined by the first two bits of byte 5 (either 0x00 or 0x02). 

```
Data:

10100101 0 00100000 01100000 01110000 10001000 01001011 11111111 01100111 [0xA5] 2020.08.03 16:34 | Partition 1 | Partition in alarm

dscKeybusProcessData.cpp:
void dscKeybusInterface::processPanel_0xA5() {
 (...) 
  switch (panelData[5] & 0x03) {
    case 0x00: processPanelStatus0(partition, 6); break;
    case 0x02: processPanelStatus2(partition, 6); break;
  }
}
```

This looks at command 0xA5 byte 5, which is `10001000` for the example data.  The first two bits are 00, so case 0x00 is selected and calls `processPanelStatus0` and sends the partition number and the location of the status data (byte 6).  Byte 6 is `01001011` which is 0x4B for "Partition in alarm":

```
void dscKeybusInterface::printPanelStatus0(byte panelByte) {
  bool decoded = true;
  switch (panelData[panelByte]) {
(...)
    // 0x09 - 0x28: Zone alarm, zones 1-32
    // 0x29 - 0x48: Zone alarm restored, zones 1-32
    case 0x49: stream->print(F("Duress alarm")); break;
    case 0x4A: stream->print(F("Disarmed after alarm in memory")); break;
    case 0x4B: stream->print(F("Partition in alarm")); break;
(...)
```

Of course, all of this is guesswork on my part, DSC likely has a different schema so feedback is helpful!

BriceDet, 2020-08-10:
hmm... I read too fast the code. I make a mistake with this 2 lines...
```
byte partition = panelData[3] >> 6;
switch (panelData[5] & 0x03)
```

So my problem seam to not come from here.

What I detect : with the OpenHab-MQTT exemple, the alarm is never detected.
So I decide to check
dsc.alarmChanged[0] // partition 1
dsc.alarm[0] // (partition1)
.. value at each loop.
When I play the same scenario (like in previous file), this two variable stay = 0

I'm a little bit disappointed :)

taligentx, 2020-08-10:
What is the output of the `Status` sketch?

Dilbert66, 2020-08-10:
The 4B subcommand is not processed when the carrier cmd is a 0xA5. The default action is a return if it's not in the inital case statement so it does not fall through to the following logic that checks alarm zones and what not.

Dilbert66, 2020-08-10:
if (panelData[0] == 0xA5) {
    switch (panelData[panelByte]) {

      // Keypad Fire alarm
      case 0x4E: {
        keypadFireAlarm = true;
        if (!pauseStatus) statusChanged = true;
        return;
      }

      // Keypad Aux alarm
      case 0x4F: {
        keypadAuxAlarm = true;
        if (!pauseStatus) statusChanged = true;
        return;
      }

      // Keypad Panic alarm
      case 0x50: {
        keypadPanicAlarm = true;
        if (!pauseStatus) statusChanged =true;
        return;
      }

      // Panel battery trouble
      case 0xE7: {
        batteryTrouble = true;
        batteryChanged = true;
        if (!pauseStatus) statusChanged = true;
        return;
      }

      // Panel AC power failure
      case 0xE8: {
        powerTrouble = true;
        powerChanged = true;
        if (!pauseStatus) statusChanged = true;
        return;
      }

      // Panel battery restored
      case 0xEF: {
        batteryTrouble = false;
        batteryChanged = true;
        if (!pauseStatus) statusChanged = true;
        return;
      }

      // Panel AC power restored
      case 0xF0: {
        powerTrouble = false;
        powerChanged = true;
        if (!pauseStatus) statusChanged = true;
        return;
      }
      default: return;
    }
  }

BriceDet, 2020-08-11:
Thanks @Dilbert66 , with 
`//default: return;`
the `dsc.alarm[]` and `dsc.alarmChanged[]` are updated. 

Do you think that is a mistake or is there a spécific reason for this `default: return` ?
Why it's work with other DSC centrals ?

Dilbert66, 2020-08-11:
as noted above by  taligentx, most systems with support for more than 2 partitions use the EB command for alarm/zone statuses. My PC5010 does.  The A5 is not used in those for this purpose so it works fine.  In your case there is no EB command so it  needs to use the A5.  It's difficult to be able to code for all systems if they are not available to the developer at the time for testing.  I can't say if it was a mistake or if there was another reason at the time but either way, your issue and your help in posting logs makes the code better.

taligentx, 2020-08-11:
Thanks for catching that @Dilbert66 - I'm looking into this, as it looks like a regression in the develop branch, the master branch is not affected.  The `default: return;` actually isn't the issue here, though removing it would skip past the bug in develop.  This section is intended to handle 0xA5 status data that is not tied to a specific partition and set by the panel as partition 0 (keypad-generated alarms, AC power, battery power) - in master:

```
void dscKeybusInterface::processPanelStatus0(byte partition, byte panelByte) {

  // Processes status messages that are not partition-specific
  if (partition == 0 && panelData[0] == 0xA5) {
    switch (panelData[panelByte]) {
      case 0x4E: {       // Keypad Fire alarm
(...)
```

In develop:
```
// Processes status messages that are not partition-specific
  if (panelData[0] == 0xA5) {
    switch (panelData[panelByte]) {
```

I'll need to see if there's a reason why I removed the check to see if the partition is set to 0.

Dilbert66, 2020-08-11:
Ahhh. That explains it. I didnt go back and check into the master branch to compare since I'm working with the develop branch at this point on my personal projects.   DSC has a lot of quirks it seems from panel to panel.   There was an issue with a 5005 panel that I worked with another user on, that would send intermittent bogus 05 messages . I put a pull request on that for your study in case you get bored 👍

taligentx, 2020-08-11:
Found it - I removed the `partition == 0` check because of issue #108 - the PC585 panel was setting the partition to 1 for the keypad F/A/P and power state alarms so the library wasn't catching those alarms.  Pending testing, I'll adjust the default case to check for partition 0:

```
default: if (partition == 0) return;
```

>>>DSC has a lot of quirks it seems from panel to panel.

Very much so, you can see the Keybus protocol get increasingly stuffed with data as panels supported more partitions, zones, and accessories.  I recall coming across a statement or interview from someone at DSC about the Neo series with Corbus and part of the rationale other than encryption was that Keybus was hitting its limits on responsiveness.

>>>There was an issue with a 5005 panel that I worked with another user on, that would send intermittent bogus 05 messages . I put a pull request on that for your study in case you get bored 👍

Thank you!  Knowing that there are bugs in the released (master branch) code is a strong motivator to push a new release, it's overdue but I was hoping for more feedback on the other new features (code unlocker, esp32, etc).

Dilbert66, 2020-08-11:
I'll see if I can test your unlocker code on my pc5010.  I do have some esp32's I can try and test with as well.

BriceDet, 2020-08-11:
The unlocker work verywell with the PC5016  and esp8266 👍

BriceDet, 2020-08-12:
Do you need me to do some tests with my pc5016?

taligentx, 2020-08-13:
> Do you need me to do some tests with my pc5016?

Instead of commenting out `default: return;`, try:

```
default: if (partition == 0) return;
```

The code should look like:
```
void dscKeybusInterface::processPanelStatus0(byte partition, byte panelByte) {

  // Processes status messages that are not partition-specific
  if (panelData[0] == 0xA5) {
    switch (panelData[panelByte]) {
(...)
      // Panel AC power restored
      case 0xF0: {
        powerTrouble = false;
        powerChanged = true;
        if (!pauseStatus) statusChanged = true;
        return;
      }
      default: if (partition == 0) return;
    }
  }
```

This should handle both the PC585 and PC5016 (along with the other panels that don't have the 0xEB command).

BriceDet, 2020-08-18:
It looks good 👍 
Let me modify my pull request