taligentx/dscKeybusInterface #149: Esp32, aumentar buffer

Damian-Mora, 2020-08-09:
He probado está maravillosa librería en el módulo Esp8266 y me ha funcionado muy bien, de vez en cuando me genera problemas en el buffer y me gustaría saber cómo puedo hacer para aumentar el buffer,  también me gustaría saber que tengo q cambiar para q me funcione está librería en un esp32. Muchas gracias a los desarrolladores por compartir sus investigaciones y conocimientos.

taligentx, 2020-08-10:
Hi @Damian-Mora, if Google Translate is correct:

```
I have tried this wonderful library in the Esp8266 module and it has worked very well for me, from time to time it generates problems in the buffer and I would like to know how I can do to increase the buffer, I would also like to know that I have to change so that it works for me this library on an esp32. Many thanks to the developers for sharing their research and knowledge.
```

You can modify the Keybus data buffer by changing `dscBufferSize` in `dscKeybusInterface.h`:

```
(...)
#elif defined(ESP8266) || defined(ESP32)
const byte dscPartitions = 8;
const byte dscZones = 8;
const byte dscBufferSize = 50;
```

I'm curious, what is your sketch doing?  For ESP8266 and ESP32, the library buffers 50 commands and I've never seen a sketch run out of the buffer - is the sketch going a long time without calling `dsc.loop()`?

ESP32 is supported in the develop branch, I still need to merge into a new release but you can use the develop branch without issues.

Damian-Mora, 2020-08-12:
> Hola @ Damian-Mora , si el Traductor de Google es correcto:
> 
> > He probado esta maravillosa librería en el módulo Esp8266 y me ha funcionado muy bien, de vez en cuando genera problemas en el buffer y me gustaría saber como puedo hacer para aumentar el buffer, también me gustaría saber que tengo que cambiar para que me funcione esta biblioteca en un esp32. Muchas gracias a los desarrolladores por compartir sus investigaciones y conocimientos.
> 
> Puede modificar el búfer de datos de Keybus cambiando `dscBufferSize`en `dscKeybusInterface.h`:
> 
> ```
> (...)
> #elif defined(ESP8266) || defined(ESP32)
> const byte dscPartitions = 8;
> const byte dscZones = 8;
> const byte dscBufferSize = 50;
> ```
> 
> Tengo curiosidad, ¿qué está haciendo tu boceto? Para ESP8266 y ESP32, la biblioteca almacena 50 comandos en el búfer y nunca he visto que un boceto se agote del búfer. ¿El boceto pasa mucho tiempo sin llamar `dsc.loop()`?
> 
> ESP32 es compatible con la rama de desarrollo, todavía necesito fusionarme en una nueva versión, pero puedes usar la rama de desarrollo sin problemas.

Muchas gracias por tu pronta respuesta, ya me queda claro lo del buffer de eventos. Te comento estoy utilizando la librería y guardo los datos en database firebase, sucede que al momento de enviar señales si el internet está en óptimas condiciones me funciona de maravilla, pero cuando el internet falla y justo en ese momento está enviando señales a firebase el módulo se queda pegado ya que pierde mucho tiempo al tartar de enviar la señal, y mientras sucede eso se pierde información de la librería  ..  de la única forma que arranca es con un reset de hardware Ya que por software no me funciona. La idea es cambiarlo a un esp32 para utilizar los dos núcleos , uno para la librería dsc y el otro núcleo q se encargue de la nube. Estoy a la espera de q me llegue el módulo esp32 para hacer pruebas

taligentx, 2020-08-13:
>Thank you very much for your prompt response, it is clear to me about the event buffer. I tell you that I am using the library and I save the data in the firebase database, it happens that at the time of sending signals if the internet is in optimal conditions it works wonderfully, but when the internet fails and right at that moment the module is sending signals to firebase It stays stuck because it loses a lot of time trying to send the signal, and while that happens, information from the library is lost .. the only way it starts is with a hardware reset, since it doesn't work for me because of the software. 

In this case, increasing the buffer would only help if the internet connection is down for short periods.  Depending on the Firebase library, you may be able to initiate the connection and then run a loop until the connection is available - for example, in the Email sketch:

```
  while (!smtpClient.available()) {
    dsc.loop();  // Processes Keybus data while waiting on the SMTP response
(...)
    yield();
  }
```

This runs `dsc.loop()` until the SMTP connection is established and the buffer will not be an issue.

>The idea is to change it to an esp32 to use the two cores, one for the dsc library and the other core that takes care of the cloud. I am waiting for the esp32 module to arrive for testing

If you are using the esp32 Arduino library, the Arduino sketch runs in a single task on a single core by default, but this would work if you are able to split off the Firebase code into a separate RTOS task.

Damian-Mora, 2020-08-14:
> In this case, increasing the buffer would only help if the internet connection is down for short periods. Depending on the Firebase library, you may be able to initiate the connection and then run a [...]

> > Muchas gracias por su pronta respuesta, para mí está claro sobre el búfer de eventos. Te digo que estoy usando la librería y guardo los datos en la base de datos de firebase, sucede que a la hora de enviar señales si internet está en óptimas condiciones funciona de maravilla, pero cuando internet falla y justo en ese momento el el módulo está enviando señales a la base de fuego. Se queda atascado porque pierde mucho tiempo tratando de enviar la señal, y mientras eso sucede, la información de la biblioteca se pierde ... la única forma en que comienza es con un reinicio de hardware, ya que no No funciona para mí debido al software.
> 
> En este caso, aumentar el búfer solo ayudaría si la conexión a Internet no funciona durante períodos cortos. Dependiendo de la biblioteca de Firebase, es posible que pueda iniciar la conexión y luego ejecutar un bucle hasta que la conexión esté disponible, por ejemplo, en el boceto del correo electrónico:
> 
> ```
>   while (!smtpClient.available()) {
>     dsc.loop();  // Processes Keybus data while waiting on the SMTP response
> (...)
>     yield();
>   }
> ```
> 
> Esto se ejecuta `dsc.loop()`hasta que se establece la conexión SMTP y el búfer no será un problema.
> 
> > La idea es cambiarlo a un esp32 para usar los dos núcleos, uno para la biblioteca dsc y el otro núcleo que se encarga de la nube. Estoy esperando que llegue el módulo esp32 para probar
> 
> Si está utilizando la biblioteca de Arduino esp32, el boceto de Arduino se ejecuta en una sola tarea en un solo núcleo de forma predeterminada, pero esto funcionaría si pudiera dividir el código de Firebase en una tarea RTOS separada.

Muchas gracias por tus respuestas.

closed 2020-08-14 by Damian-Mora