taligentx/dscKeybusInterface #152: DSC GPRS Alarms

PipeDeveloper, 2020-08-28:
Hello.

Short story
Im an customer from ADT Chile, actually im planning to cut the service because one day some workers enter to my home (while they work at the yard) and ADT didnt notify when the alarm starts yelling. Nothing happened but that disgust me in the way that they aren't giving a good service.

Im wandering if this can help me to integrate some "modern" alarms from the DSC brand as an IoT device. Actually i have a wireless system with GPRS monitor (the system got a phone chip, central motinors it via SMS or 3G i dont know exactly).
My idea is to know the status of my sensors, zones, batery status of sensors (all wireless), and push notifications from my Home Assistant server.

The model of the "box"
DSC GS3055-I

And i found the manual
https://www.manualslib.com/manual/824606/Dsc-Gs3055-Ig.html

My issue here is that i dont know how to do the wiring. On this case terminals are not labeled with "red, yellow, green, black" and there is no documentation about comunicating from the serial port. And i dont want to mess or burn something.

Thank you.

taligentx, 2020-09-07:
Hi @PipeDeveloper - the model you provided is for the GPRS module, which DSC control panel do you have?

PipeDeveloper, 2020-09-07:
It was hard to find, but finally found it behind a wall (I was not present on the instalation)
DSC PC1832

[image]

My idea is keep using the same pannel. Just add a NodeMCU or ESP8266 module to monitor the system from my Home Assistant server.

taligentx, 2020-09-07:
Good to hear you were able to find the panel - and the good news is that the PC1832 is supported and works well with this library.  The microcontroller interface will connect directly the control panel wiring red/black/yellow/green terminals.

PipeDeveloper, 2020-09-07:
Yeah thats nice!

Before closing this issue i have just a few question about the wiring and the firmware. I have some experience with electronics and basic software skills but i didn't find more documentation about this proyect.

First all, about the wiring:
[image]

And second about compiling the firmware to nodeMCU and basic configuration. Like the step by step.

For example for flashing ESP8266 usually there is a .bin file to easily flash it but here it uses Arduino IDE, I know is a famous software and i used for the past for arduino but never for ESP8266, and I know it need extras libraries for networking but i dont know wich one this software uses.

And second about the software itself for initial configuration. I dont know if this create a web page where i can modify basic parameters like my WIFI settings, MQTT server/user/pass. Well all rest i can imagine it will detect automatically all zones and extras, I read part of the code and it looks good.

I really apreciate to answer those questions.
Thanks in advance.

taligentx, 2020-09-29:
> For example for flashing ESP8266 usually there is a .bin file to easily flash it but here it uses Arduino IDE, I know is a famous software and i used for the past for arduino but never for ESP8266, and I know it need extras libraries for networking but i dont know wich one this software uses.

https://github.com/esp8266/Arduino

> And second about the software itself for initial configuration. I dont know if this create a web page where i can modify basic parameters like my WIFI settings, MQTT server/user/pass. Well all rest [...]

You can refer to the built-in example sketches for different implementations - WiFi, MQTT, etc settings are configured in the sketch and then compiled and uploaded to the ESP8266.  

There are a few forums that would be good resources for general info about the ESP8266, Arduino, etc:
https://reddit.com/r/esp8266/
https://reddit.com/r/arduino/
https://www.esp8266.com

PipeDeveloper, 2020-10-16:
Finally got the solution.

https://community.home-assistant.io/t/esp8266-into-existing-alarm-dsc-system/225224

Uses same library and even more.
Thank you very much, this helped me a lot.

closed 2020-10-16 by PipeDeveloper