taligentx/dscKeybusInterface #153: MQTT SSL (Question)

DaniilMir, 2020-09-18:
I've searched for existing issues and didn't found anything related to MQTT SSL connection.
As I can see this project is using SSL for Pushbullet, Email and SMS.
Are there options to enable SSL connection to the MQTT Server?
In my environment port 1883 is disabled for security reasons.

taligentx, 2020-09-29:
Hi @DaniilMir - I haven't tested secure MQTT, it does appear possible for the ESP8266/ESP32 given the discussion at https://github.com/knolleary/pubsubclient/issues/462 .  PRs welcome for secure MQTT versions of the built-in example sketches!

taligentx, 2020-12-24:
Hi @DaniilMir - after testing, secure MQTT works fine and you can use the Pushbullet sketch as a example.  For example, to switch to secure MQTT for the HomeAssistant-MQTT sketch:

1. Include `WiFiClientSecure.h`:
    ```
    #include <ESP8266WiFi.h>
    #include <WiFiClientSecure.h>
    #include <PubSubClient.h>
    #include <dscKeybusInterface.h>
    ```
2. Set mqttPort to 8883.
3. Replace `WiFiClient wifiClient;` with `WiFiClientSecure wifiClient;`
4. Set the encryption authenticate method in `setup()` - for testing without authenticated the MQTT server's certificate:
    ```
    wifiClient.setInsecure();
    mqtt.setCallback(mqttCallback);
    ```

closed 2021-02-04 by taligentx