taligentx/dscKeybusInterface #154: Not Working at all don't know where to start Dsc alarm beep

thib5, 2020-09-20:
Hi, I think my issue is the wirering ... I don't know ... every time I fire the disc with the node plugged it beep like fore ever .... and nothing appear on the disc panel 

I follow this wire guide : 

https://community-assets.home-assistant.io/original/3X/f/1/f11f4a0c5695f5b40bc9b7d9b1ce096dca910955.png

Thanks

thib5, 2020-09-20:
I do not have plug the power like that ... just the ground yellow and green... i used usb for power

Could it be because i dont have the right « Master Code »

Dilbert66, 2020-09-21:
Usb for power is fine.  Master code is irrelvant.  I would suggest that you check your yellow and green line for shorts to ground.  That is what it looks like the issue is to me. One of them is shorted.

thib5, 2020-09-21:
Hey @Dilbert66 I saw that you added a wirering schematic on your github project ! so I cleaned up my bread board and i still have the same issue... This is what it look now :

[image]

Dilbert66, 2020-09-22:
Try to connect with only the yellow one first and see if it affects your alarm.  Then try it with just the green one connected.  This will determine which circuit is causing issues. The only way that the circuit would impact the existing keypad is if you are adding a short on one of those lines.   Is the chip getting hot?

BriceDet, 2020-09-22:
Did tout flash first the code into your ESP?

taligentx, 2020-09-22:
> Could it be becaus I do not have plug the power like that ? ... just the ground yellow and green... i used usb for power

You may run into issues with the separate grounds, try connecting at least the black ground wire from the DSC panel to the microcontroller.

Dilbert66, 2020-09-22:
Yes, a bad ground could most likely be the culprit here as taligentx noted.  A shorted pin  would not sink enough current through the 15k resistor to cause issues.

thib5, 2020-09-23:
Ok so I tested each wire a couple of time and don't know what could be the issue..

Green Only : Nothing
Yellow Only :  Nothing
Black Only : Nothing 

Anything else ( Ex: Green and yellow, Green and Black Green and Yellow .. etc.. )  is doing  Longs beeps ( not usual 2 beep ) and the screen is like "unavailable" ) 

Also quick question .. ? should I plug the AC - of the panel to the ground of the nodemcu ou should I plug the AUX - ?

Thanks

mritalian, 2020-09-25:
Long beep is probably a keybus fault.  I had those issues when using 15k/10k resistive divider.  Either that, or I wired something wrong.

Listen to the 1st reply, and use 33k/10k. this works on my 8266 (looks similar to yours).

Also, listen to the instructions on the main readme, and solder... as much as possible.  I have the resistive divider soldered on splice on the alarm wire, shrink wrap, so I end up with a kind of adapter cable.  Before plugging anything in, measure the yellow to ground and green to ground on both ends of the wire.  Measuring on one end should be 33kohm, on the other it should measure 10k.  Between yellow+green should be 66k on one end and 20k on the other.

Once all is double triple checked, then feed it to a breadboard, if you must, ESP8266 pins, to avoid any kind of EMI.

I do this, but temporary, until I'm happy with it and I'll solder it and put it in a project box.

thib5, 2020-09-26:
Ok so I tried to bypass the bread board just to be sure ... so I plug everything directly on my nodmcu and it still does the same thing ....

I found a way on the panel to stop the beeping but the panel are unusable and I did not receive any update from the nodemcu

taligentx, 2020-09-29:
Which DSC panel model do you have?  Assuming it's a supported panel, try powering off the DSC panel (remove the AC power and battery power), load the `KeybusReader` sketch to the ESP8266, and capture the serial output as you power the DSC panel up.

Regarding the ESP8266 GPIO voltage tolerance: https://ba0sh1.com/2016/08/03/is-esp8266-io-really-5v-tolerant/

As far as personal experience, I've run ESP8266's on 5v GPIO signals for years of continuous uptime with no issues.  It's handy when working with a mix of microcontrollers, though I am planning on updating the README.md to combine the ESP8266 and ESP32 to use 33k/10k resistors as that part of the wiring diagram will eventually include the ESP32-S2 once I work out support for it.

thib5, 2020-09-29:
Ok thanks I will double check everything soon and try to have some Logs of what happen... also I will try to unplug the panel and see what happen

My dsc is PC1616 

Like I asc show I plug the ground to ax - ou ac - ?

taligentx, 2020-09-29:
> Like I ask show I plug the ground to ax - ou ac - ?

As shown in the README.md, ground should be connected to AUX (-)

mritalian, 2020-09-29:
FYI @thib5  I'm on a PC1616 also.

thib5, 2020-09-30:
> FYI @thib5 I'm on a PC1616 also.

Ok cool thanks I will have the time to double check  ( for the third time )  everything tomorrow

thib5, 2020-10-02:
Ok Thank you all again but after double checking everything for the third time ... it does the same thing ... 

If I recap : 

I have a NodeMcu plugged in a pc1616. as soon as I plug more than one wire (no mater which one) my panel does long beep and the panel lcd is empty. 

I tried : 

- Plug every wire one by one ( then I notice that I can't plug more than 1 wire ) 
- With and without bread board 
- Plug the dsc first
- Plug nodeMCU
- Put the logger into debug ... (no new log when I powered the dsc ) 

More info : 
I do not have any code ( master, programer etc ) 
I bought the house like 6 month ago so I don't have any idea how and when. it was installed

This is a pic of ho I plug the panel ( I already send a pic of my nodeMCU )

[image]

mritalian, 2020-10-02:
Well, there is something worth pointing out.  The pins should be in input
mode.  If they are in output mode, it's possible they pull the signal down
or up enough to disrupt the keybus signals.  Try programming the status
example with the appropriate pins.

thib5, 2020-10-02:
> Well, there is something worth pointing out. The pins should be in input mode. If they are in output mode, it's possible they pull the signal down or up enough to disrupt the keybus signals. Try programming the status example with the appropriate pins.
> […](#)

What you mean by put the pin in input mode ?

mritalian, 2020-10-03:
Pins on the ESP, or an Arduino are either in input or output mode.  When
the output is low, there is an internal pull down resistor bringing that
data line to 0, similarly if it's high there's a resistor pulling to vcc.
When in input mode, there is no such internal resistor pulling up or down.
I'm probably not 100% correct as I haven't read the exact specs for the
chip, but  the point is, you shouldn't connect a pin in output mode to the
DSC clk or data, you'll get undefined behaviour.

Also you should read up on the basics of esp / Arduino.  It will save you
time.

thib5, 2020-10-03:
Yes I know what you are talking about sorry I thought you were talking about putting the dsc in input mode ! so yeah eh I understand what you mean but it's not my code ... I'm prety sure it's in input mode because I saw this : 

void dscKeybusInterface::begin(Stream &_stream) {
  pinMode(dscClockPin, INPUT);
  pinMode(dscReadPin, INPUT);

Thanks again for your help

Dilbert66, 2020-10-03:
Honestly to me that points more and more to a grounding issue somewhere.

thib5, 2020-10-03:
> Honestly to me that points more and more to a grounding issue somewhere.

Is there something I can check ? because like I said I triple check everything a couple of time

Dilbert66, 2020-10-03:
At this  point, I'm out of ideas. Welcome to the fun of DIY projects! Im sure you will find your answer.

thib5, 2020-10-03:
Is there somewhere where I can found what loong beep mean ?  I search de web and I found nothing

mritalian, 2020-10-03:
Long beep from the panel for me meant keybus fault.  That panel had a text
display and that's what it would read out with the long beep.

taligentx, 2020-10-04:
Have you measured your resistors with a DMM?  It may just be an artifact of the dark picture, but after increasing the exposure, your resistors that should be 10k ohm (10,000 ohm) to ground look like 10 ohm resistors:
[image]

At this point I would use a DMM to measure the resistors and then go step by step to see what's bringing down the data lines:

1) Disconnect all of your wiring from the DSC panel and verify everything is working correctly.
2) Measure the voltage on the DSC panel between the GRN and BLK terminals with a DMM.
3) Measure the voltage on the DSC panel between the YEL and BLK terminals with a DMM.
4) Remove the NodeMCU from the breadboard.
5) Connect the green terminal to your wiring and measure voltages again.
6) If the panel and keypads are working and voltages are the same, repeat with the yellow line.
7) Attach the NodeMCU and repeat the voltage measurements.

thib5, 2020-10-06:
IT WORKED !! Omg Thanks I feel so dumb ... I didn't know that there was a difference between 10k omh and 10omh ... I asked my local electronics store for 10 omh because I was sure it was the same... like everything was in k .... So I bought new resistor and it work ! anyway sorry for all that I hope that some one will be as stupid as me and will find the answer here !

Last thing : right now my node is plugged in usb, do you have any advice to step down the ac+ from the panel et power my nodeMCU directly from the panel ? Like an eBay / amazon link ahah !

Dilbert66, 2020-10-06:
> Have you measured your resistors with a DMM? It may just be an artifact of the dark picture, but after increasing the exposure, your resistors that should be 10k ohm (10,000 ohm) to ground look like [...]

Nice catch there! I would not even have thought of that.

taligentx, 2020-10-06:
> IT WORKED !!

Great to hear!  

> (...)do you have any advice to step down the ac+ from the panel et power my nodeMCU directly from the panel ? 

Sure, there are a few different options - you'll need a voltage regulator to drop the DSC panel AUX +/- (runs at around 13.75v DC) to either 5v to the `Vin` pin on the NodeMCU or 3.3v to the 3.3v pin.  No part of this interface connects to the AC power terminals.

Here are a few hard to find notes because they're only in the `develop` branch README.md, check the wiring diagram for the details:

Power:
- Arduino boards can be powered directly from the DSC panel
- esp8266/esp32 development boards should use a 5v voltage regulator:
LM2596-based step-down buck converter modules are reasonably efficient and commonly available for under $1USD shipped (eBay, Aliexpress, etc).
- MP2307-based step-down buck converter modules (aka Mini360) are also available but some versions run hot with an efficiency nearly as poor as linear regulators.
- Linear voltage regulators (LM7805, etc) will work but are inefficient and run hot - these may need a heatsink.

Personally I use the LM2596 modules.

Dilbert66, 2020-10-06:
Just to add a note about some of the esp8266, I would be hesitant on powering up some of those clone wemos d1 via 5volts.  They have a very weak 5 volt regulator that cannot provide long term current needed.  I would instead use the lm2596 step-down convert as suggested to power the chip at the 3.3 volt input instead, bypassing the 5volt regulator.  I speak from experience here. I have seen 2 d1's short self destruct after the 5volt regulator dies and takes the chip with it.

taligentx, 2020-10-06:
Also to be clear, the LM2596 modules and Mini360 modules are adjustable voltage regulators - you'll need a DMM to measure the output voltage from the regulator as you turn the adjustment potentiometer to get the voltage you want (5v or 3.3v).

On the LM2596 it's the bronze colored screw on the blue component:

[image]

> Just to add a note about some of the esp8266, I would be hesitant on powering up some of those clone wemos d1 via 5volts. They have a very weak 5 volt regulator that cannot provide long term current [...]

Good call, I'll add a note to the README.md about powering the 3.3v rail directly - the NodeMCU clones all seem to use AMS1117 which seems to work fine, but quite a few of the Wemos D1 clones (not the original LOLIN boards) used the terrible 150mA current regulators:

https://www.letscontrolit.com/forum/viewtopic.php?t=6603

closed 2020-10-20 by taligentx