taligentx/dscKeybusInterface #157: Arduino exemple keybusReader si to big for UNO

BriceDet, 2020-09-30:
Hi,
The last develop version of arduino keyBusReader exemple are too big for the arduino UNO.
Could it add in comment of this exemple? (I can do this but I want to be sure that's the good way).

Thanks

taligentx, 2020-10-01:
Strange, I'm not seeing this issue, testing with Arduino 1.8.13:

```
Sketch uses 31946 bytes (99%) of program storage space. Maximum is 32256 bytes.
Global variables use 950 bytes (46%) of dynamic memory, leaving 1098 bytes for local variables. Maximum is 2048 bytes.
```

Is there a difference in compiler/environment/etc?  It'd be good to track down.

BriceDet, 2020-10-01:
Under 83%, software flash crashed...
Only for me?
(It's an official arduino uno)

taligentx, 2020-10-01:
>Under 83%, software flash crashed...

So you get a crash if flashing a file over ~26kB?  I haven't run across issues filling the flash memory on the atmega328p - even the times when I've wiped out the Arduino bootloader to use the entire 32768 bytes of flash instead of the normal 32256 bytes with the Arduino bootloader.

BriceDet, 2020-10-01:
In fact i didn't identify the root cause:
If I flash the current version, the flash process stop (not responding).
If I comment some line in order to try to have lighter build, the flash process completed.
May you have an idea? 
If you said me that possible to flash this size of builds, I'll look for another reason 😅

taligentx, 2020-10-04:
It'll be a while before I can test this, what is the final compiled size of `KeybusReader` for you?

BriceDet, 2020-10-04:
after my last git push (event buffer decoded) :
```
Arduino: 1.8.12 (Windows 10), Board: "Arduino Uno"

Sketch uses 34056 bytes (105%) of program storage space. Maximum is 32256 bytes.text section exceeds available space in board

Global variables use 972 bytes (47%) of dynamic memory, leaving 1076 bytes for local variables. Maximum is 2048 bytes.
Sketch too big; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing it.
Error compiling for board Arduino Uno.
```
With the lase develop version :
```
Sketch uses 31930 bytes (98%) of program storage space. Maximum is 32256 bytes.
Global variables use 950 bytes (46%) of dynamic memory, leaving 1098 bytes for local variables. Maximum is 2048 bytes.
```
compile process is ok, but upload not working : the IDE freezed at this step
[image]
 
and the upload button stay pressed
[image]

taligentx, 2020-11-21:
Hi @BriceDet - I've consolidated some code in `dscKeybusPrintData.cpp` in the `develop` branch to reduce flash usage.  This includes zone expander decoding but your event buffer decoding (thank you!) is still on the TODO list.  Current `KeybusReader` compile size for the Uno:
```
Sketch uses 28514 bytes (88%) of program storage space. Maximum is 32256 bytes.
Global variables use 950 bytes (46%) of dynamic memory, leaving 1098 bytes for local variables. Maximum is 2048 bytes.
```

This is experimental as I won't have access to the test hardware for another few weeks so feedback would be great if you get a chance to try `develop`.  This is still above the 83% you mentioned as causing crashing, but I haven't seen the same issue  my Uno or Nano originals and clones - are you able to test with a different Arduino module?

kricon, 2020-11-27:
Flashed current develop version on genuine Arduino Uno, everything works, took all logs today thru it:
```
Sketch uses 29872 bytes (92%) of program storage space. Maximum is 32256 bytes.
Global variables use 957 bytes (46%) of dynamic memory, leaving 1091 bytes for local variables. Maximum is 2048 bytes.
```

Then I've tested commit 640d71c from Oct 30 just in case and it also compiled, flashed and run nicely:
```
Sketch uses 31946 bytes (99%) of program storage space. Maximum is 32256 bytes.
Global variables use 950 bytes (46%) of dynamic memory, leaving 1098 bytes for local variables. Maximum is 2048 bytes.
```

I also have non official Arduino Nano clone, but I don't have Mini USB cable handy right now so I cant verify it at the moment.

BriceDet, 2020-11-27:
```
Sketch uses 29926 bytes (92%) of program storage space. Maximum is 32256 bytes.
Global variables use 957 bytes (46%) of dynamic memory, leaving 1091 bytes for local variables. Maximum is 2048 bytes.
C:\Users\MarMotte\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude -CC:\Users\MarMotte\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf -v -patmega328p -carduino -PCOM6 -b115200 -D -Uflash:w:C:\Users\MarMotte\AppData\Local\Temp\arduino_build_350886/KeybusReader.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\MarMotte\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM6
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 3.3
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\MarMotte\AppData\Local\Temp\arduino_build_350886/KeybusReader.ino.hex"
avrdude: writing flash (29926 bytes):

Writing | ###############################################avrdude: stk500_recv(): programmer is not responding
#avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding

```

Very strange  comportment.
I'll buy another arduino next months...

BriceDet, 2020-11-27:
let me closed this issue. It may due to my hardware.

closed 2020-11-27 by BriceDet