taligentx/dscKeybusInterface #158: PC5005 Keybus data capture - feedback requested

taligentx, 2020-10-02:
Per https://github.com/Dilbert66/esphome-dsckeybus/issues/2, there may be issues with the Keybus data coming from PC5005 panels - this issue is open to collect data to see if it's an isolated issue or common to all PC5005 panels.

If you have this panel, load the `KeybusReader` example sketch and capture the serial output while performing various activities: opening zones, arming the system, disarming the system, triggering an alarm, etc.  Thanks!

Dilbert66, 2020-10-02:
Hi Nikhil, i have pull request that corrects this issue.  The problem was that his panel sends spurious packets at odd times. The fix was to add a sort of debounce to the capture routine that selectively ignores the first few commands before accepting as valid.
Here's the link
https://github.com/taligentx/dscKeybusInterface/pull/150

taligentx, 2020-10-02:
Hi, yes I've been taking a look at the PR - in this case I'd like to see the data from `KeybusReader` from at least two or three different PC5005 panels as I'd like to see what's going on with all commands, especially commands with checksums to see if there are any CRC errors, etc popping up.  Command '0x05' is a fundamental part of Keybus and it'd be interesting if all PC5005 panels shipped with firmware that screwed this up.  

At this point I'd rather not change the command processing code based on a single report without much more data.

Dilbert66, 2020-10-02:
Understood.  That was sure an oddball one with that panel.  This might also be revision dependent so that he had a buggy version before DSC fixed it.