taligentx/dscKeybusInterface #164: Firing ALARM status when battery low

otto76, 2020-10-15:
Hey

it seems to DSC is sending ALARM message constantly whenever battery has died 
code for alarm handling: 
`      // Publishes alarm triggered status
      if (dsc.alarmChanged[partition]) {
        dsc.alarmChanged[partition] = false;  // Resets the partition alarm status flag
        if (dsc.alarm[partition]) {
          publishState(mqttPartitionTopic, partition, "T");
        }
      }`

I got a separate message regarding battery status and then comes ALARM msg for partition on regular basis

kricon, 2020-10-16:
I have just tested this with PC585 v2.3 I currently have on desk for testing.
Removed panel battery and keybus reader didnt show any partition alarm status, even if the system was armed, just show normal panel battery trouble.

otto76, 2020-10-19:
Replaced battery and issue gone 
Closing case for now

closed 2020-10-19 by otto76