taligentx/dscKeybusInterface #174: keypad regression ?

BriceDet, 2020-11-30:
With Keybus Reader (develop - esp8266), the 4 digit code master code "4040" sent by an LCD keypad are not decoded (regression since my first commit on eventbuffer) (code ask after *6 request)

```
11:31:55.843 ->     7.32: 11111111 1 00010001 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Unknown data
11:31:56.330 ->     7.79: 11111111 1 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Unknown data
11:31:56.759 ->     8.22: 11111111 1 00010001 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Unknown data
11:31:57.267 ->     8.72: 11111111 1 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Unknown data
```

BriceDet, 2020-11-30:
```
12:37:49.275 ->    29.13: 11111111 1 00010001 11111111 11111111 11111111 [Module/0x05] Partition 1 Key: 4 
12:37:49.710 ->    29.56: 11111111 1 00000000 11111111 11111111 11111111 [Module/0x05] Partition 1 Key: 0 
12:37:50.178 ->    30.01: 11111111 1 00010001 11111111 11111111 11111111 [Module/0x05] Partition 1 Key: 4 
12:37:50.767 ->    30.59: 11111111 1 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Unknown data
```

Finally.... don't know why, but sometime it's send with more or less bytes

taligentx, 2020-11-30:
Thanks for finding this, I had reworked the keypad/module data decoding and missed the keys sent during the 0x0A command, should be fixed in `develop`.

>don't know why, but sometime it's send with more or less bytes

The keypads send keys at the same time as certain panel commands, so the total length will match whatever panel command the keypad happened to use.  Right now, KeybusReader decodes keys on panel commands 0x05, 0x0A, and 0x1B.

taligentx, 2020-12-23:
I fixed a bug where the module data could be listed with the wrong command, previously the module decoding just used the current panel command, but sometimes the current panel command would not match the panel command that the module data was sent on.

The panel command that the module data is sent on is now tracked separately so the `KeybusReader` output should be more accurate.  Let me know if you still see this issue with the current `develop` branch.

taligentx, 2021-01-22:
Feel free to re-open if this issue persists in the new 2.0 release.

closed 2021-01-22 by taligentx