taligentx/dscKeybusInterface #175: Event buffer decoding - panel commands 0xAA, 0xEC

taligentx, 2020-11-30:
Continuing discussion from PR #155 for event buffer decoding thanks to the work by @BriceDet.  Panel commands 0xAA and 0xEC display event buffer messages, and seem to correspond to 0xA5 and 0xEB with an additional byte for an event index number.

Current 0xAA decoding - byte 3 bit 6-7 may be partition number but needs confirmation:
```
*  0xAA: Event buffer
 *  CRC: yes
 *  Structure decoding: *incomplete
 *  Content decoding: *incomplete
 *
 *  Byte 2 bit 0-3: Year digit 2
 *  Byte 2 bit 4-7: Year digit 1
 *  Byte 3 bit 0-1: Day digit part 1
 *  Byte 3 bit 2-5: Month
 *  Byte 3 bit 6-7: Unknown (partition?)
 *  Byte 4 bit 0-4: Hour
 *  Byte 4 bit 5-7: Day digit part 2
 *  Byte 5 bit 0-1: Selects set of status commands
 *  Byte 5 bit 2-7: Minute
 *  Byte 6: Status, printPanelStatus0...printPanelStatus3
 *  Byte 7: Event number
 *  Byte 8: CRC
 *
 *             YYY1YYY2   MMMMDD DDDHHHHH MMMMMM
 *  10101010 0 00100001 01000110 00001000 00100100 00011100 11111111 01011000 [0xAA] Event: 255 | 2021.01.16 08:09 | Zone alarm: 20
 *  10101010 0 00100000 01100110 10001101 00111000 10011001 00001000 10010110 [0xAA] Event: 008 | 2020.09.20 13:14 | Armed by user code 1
```

Current 0xEC decoding, byte 4 bit 6-7 may be partition but also needs to be checked:
```
 *  0xEC: Event buffer - partitions 1-8
 *  CRC: yes
 *  Structure decoding: *incomplete
 *  Content decoding: *incomplete
 *
 *  Byte 2: Partition
 *  Byte 3 bit 0-3: Year digit 2
 *  Byte 3 bit 4-7: Year digit 1
 *  Byte 4 bit 0-1: Day digit part 1
 *  Byte 4 bit 2-5: Month
 *  Byte 4 bit 6-7: Unknown (partition?)
 *  Byte 5 bit 0-4: Hour
 *  Byte 5 bit 5-7: Day digit part 2
 *  Byte 6 bit 0-1: Unknown
 *  Byte 6 bit 2-7: Minute
 *  Byte 7: Selects set of status commands
 *  Byte 8: Status, printPanelStatus0...printPanelStatus14
 *  Byte 9: Event number
 *  Byte 10: CRC
 *
 *                      YYY1YYY2   MMMMDD DDDHHHHH MMMMMM
 *  11101100 0 00000000 00100000 00101101 01100011 10111100 00000001 10101100 00001011 00010000 [0xEC] Event: 011 | 2020.11.11 03:47 | Exit *8 programming
```

These commands seem to be mostly decoding other than the two mentioned bits - additional logs and feedback on this decoding would be useful to make sure it's being decoded correctly.

taligentx, 2020-11-30:
@BriceDet,
>I don't understand fully the integration done by @taligentx 

The 0xAA command that you decoded turns out to be nearly the same as the 0xA5 panel status command, but it has 1 additional byte for the event index number.  So I was able to use the same decoding functions as 0xA5 (also used by 0xEB) to get the panel status messages.  I only considered that this might be a duplicate of 0xA5 after seeing the status messages that you decoded, they looked familiar - so thank you for digging into this!

@kricon:
>There is some weird symbol characters output into log instead of Byte7 mem.buffer event nr, and also unknown 0xEC cmd send afterwards (log taken from PC5020):
>
>```05:00:56.299 ->  8614.72: 11111111 1 10000111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] >Partition 1 Key: Right arrow
>05:00:56.368 ->  8614.79: 10101010 0 00100000 00101101 01100011 10111101 10101100 00001011 11001110 [0xAA] Event buffer: 0� | 2020.11.11 03:47 | Exit installer programming
>05:00:56.470 ->  8614.89: 11101100 0 00000000 00100000 00101101 01100011 10111100 00000001 10101100 00001011 00010000 [0xEC] Unknown data
>```

Thanks, this should be fixed in `develop` and the log of the 0xEC command was also handy - this is another event buffer message for partitions 3-8 and similar to 0xEB, added this decoding as well for testing.

>When we talk about event buffer, and when its mostly decoded aswell as < > buttons for getting it, it would be nice to implement viewing event buffer from Blynk sketch.

With the decoding above this should be possible, added to the TODO list for the Blynk sketch.

BriceDet, 2020-11-30:
I'm working on an .ino to show how to automatically get the full event buffer.

@kricon :It still have an error in the implementation of the event number (byte 7). Let me check the difference between my initial code and the final merge.

kricon, 2020-12-01:
> Current 0xAA decoding - byte 3 bit 6-7 may be partition number but needs confirmation:

Confirmed:
```
 2224.73: 11111111 1 10000010 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Partition 1 Key: Enter in buffer memory
 2224.98: 10101010 0 00100000 01110000 00100001 01110010 11100110 00000000 10110011 [0xAA] Event: 000 | 2020.12.01 01:28 | Enter *6 programming
 2225.08: 11101100 0 00000001 00100000 00110000 00100001 01110000 00000010 11100110 00000000 10110110 [0xEC] Event: 000 | 2020.12.01 01:28 |  | Partition 1 | Enter *6 programming

 2237.64: 11111111 1 11111111 10000010 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Partition 2 Key: Enter in buffer memory
 2237.72: 10101010 0 00100000 10110000 00100001 01110010 11100110 00000000 11110011 [0xAA] Event: 000 | 2020.12.01 01:28 | Enter *6 programming
 2237.84: 11101100 0 00000010 00100000 00110000 00100001 01110000 00000010 11100110 00000000 10110111 [0xEC] Event: 000 | 2020.12.01 01:28 |  | Partition 2 | Enter *6 programming
```

> Current 0xEC decoding, byte 4 bit 6-7 may be partition but also needs to be checked:

Byte4 bits6-7 remain low all the time no matter of the partition when I checked it thru partitions.

>(does) It still have an error in the implementation of the event number (byte 7).

Its fixed in latest develop commit, as author said in message above.

Some messages from event buffer log which are reported as Unknown data (or their decoded names differ from event buffer on keypads) that I took from panels:
```
  131.34: 10101010 0 00001011 00000100 00100000 00000011 11111111 00000111 11100010 [0xAA] Event: 007 | 20011.01.01 00:00 | Unknown data //No entry (new panel)
  131.43: 11101100 0 00000000 00001011 01000100 00100000 00000000 11111111 11111111 00000111 01100000 [0xEC] Event: 007 | 20011.01.01 00:00 | //No entry (new panel)
  182.29: 10101010 0 00010010 00101000 10001100 11100011 11111111 10010101 11100111 [0xAA] Event: 149 | 2012.10.04 12:56 | Unknown data //EVENT 1BF1 (I dont know what it is, but its in log like that)
  182.38: 11101100 0 00000000 00010010 00101000 10001100 11100000 00011011 11110001 10010101 00110011 [0xEC] Event: 149 | 2012.10.04 12:56 | //EVENT 1BF1 (I dont know what it is, but its in log like that)
 5350.42: 10101010 0 00010010 01101000 10001101 00001001 11010001 10000111 00010010 [0xAA] Event: 135 | 2012.10.04 13:02 | Unknown data //Exit fault pre-alarm
 5350.50: 11101100 0 00000001 00010010 00101000 10001101 00001000 00000001 11010001 10000111 00010101 [0xEC] Event: 135 | 2012.10.04 13:02 |  | Partition 1 | Unknown data //Exit fault pre-alarm
 5253.47: 10101010 0 00010010 00101000 10001011 01110000 11111010 01001100 00100101 [0xAA] Event: 076 | 2012.10.04 11:28 | Unknown data //Downloading Lead-in
 5253.56: 11101100 0 00000000 00010010 00101000 10001011 01110000 00000000 11111010 01001100 01100111 [0xEC] Event: 076 | 2012.10.04 11:28 | Unknown data //Downloading Lead-in
 5281.02: 10101010 0 00010010 00101000 10001011 01110100 11111011 01001011 00101001 [0xAA] Event: 075 | 2012.10.04 11:29 | Unknown data //Downloading Lead-out
 5281.11: 11101100 0 00000000 00010010 00101000 10001011 01110100 00000000 11111011 01001011 01101011 [0xEC] Event: 075 | 2012.10.04 11:29 | Unknown data //Downloading Lead-out
  317.59: 10101010 0 00010010 00101000 10001101 00001011 11111111 01111011 11110110 [0xAA] Event: 123 | 2012.10.04 13:02 | Unknown data //Exit fault
  317.67: 11101100 0 00000001 00010010 00101000 10001101 00001000 00000100 10000111 01111011 11000010 [0xEC] Event: 123 | 2012.10.04 13:02 |  | Partition 1 | Unknown data //Exit fault
  342.21: 10101010 0 00010010 01101000 10001101 00001110 11000000 01110001 11110000 [0xAA] Event: 113 | 2012.10.04 13:03 | Unknown data //*1 access by user 40
  342.30: 11101100 0 00000001 00010010 00101000 10001101 00001100 00000010 11000000 01110001 11110011 [0xEC] Event: 113 | 2012.10.04 13:03 |  | Partition 1 | Unknown data //*1 access by user 40
  797.01: 10101010 0 00011000 00011100 01010000 01000000 11110111 10110011 00011000 [0xAA] Event: 179 | 2018.07.02 16:16 | Unknown data //Phone number 1 FTC
  797.10: 11101100 0 00000000 00011000 00011100 01010000 01000000 00000000 11110111 10110011 01011010 [0xEC] Event: 179 | 2018.07.02 16:16 | Unknown data //Phone number 1 FTC
  797.51: 10101010 0 00011000 00011100 01010000 01011000 11111000 10110010 00110000 [0xAA] Event: 178 | 2018.07.02 16:22 | Unknown data //Phone number 2 FTC
  797.60: 11101100 0 00000000 00011000 00011100 01010000 01011000 00000000 11111000 10110010 01110010 [0xEC] Event: 178 | 2018.07.02 16:22 | Unknown data //Phone number 2 FTC
  798.03: 10101010 0 00011000 00011100 01100001 01011011 11111111 10110001 01001010 [0xAA] Event: 177 | 2018.07.03 01:22 | Unknown data //Periodic test with trouble
  798.13: 11101100 0 00000000 00011000 00011100 01100001 01011000 00000100 10000110 10110001 00010100 [0xEC] Event: 177 | 2018.07.03 01:22 | Unknown data //Periodic test with trouble
 1704.28: 10101010 0 00100000 00010111 10010110 00100100 11111110 11011101 01110110 [0xAA] Event: 221 | 2020.05.28 22:09 | Unknown data //Periodic test transmission
 1704.36: 11101100 0 00000000 00100000 01010111 10010110 00100100 00000000 11111110 11011101 11111000 [0xEC] Event: 221 | 2020.05.28 22:09 | Unknown data //Periodic test transmission
[... cut ...]
  297.10: 11101100 0 00000001 00010010 00101000 10001101 00001000 00000000 01001011 01111111 10000110 [0xEC] Event: 127 | 2012.10.04 13:02 |  | Partition 1 | Partition in alarm //Recent closing in log
 1548.95: 10101010 0 00100000 00110000 00100011 11001011 00110101 00100001 00111110 [0xAA] Event: 033 | 2020.12.01 03:50 | Module tamper restored: Slot 9 //Zone expander 1 Tamper restore
 1549.05: 11101100 0 00000000 00100000 00110000 00100011 11001000 00000011 00110101 00100001 10000000 [0xEC] Event: 033 | 2020.12.01 03:50 | Module tamper restored: Slot 9 //Zone expander 1 Tamper restore
```

taligentx, 2020-12-02:
>Some messages from event buffer log which are reported as Unknown data (or their decoded names differ from event buffer on keypads) that I took from panels:

Thanks for the detailed logs!  I've added these status messages - this is a great method to decode status data so I've updated the issue topic and first message to indicate that logs with unknown data would be useful.

Some of the 0xAA messages are odd - the status byte 6 is `0xFF/11111111` with the byte 5 bit 0-1 `11`, this seems to occur for all messages that are beyond the range of the 0xA5 and 0xAA commands and handled by 0xEB/0xEC.  For now I've marked these as "Extended status" and the actual status will be on the 0xEC message, but this decoding concept is open to interpretation.

There are also a few ranges of status messages that will probably get filled in - for example, in `printPanelStatus2()` it's likely that all messages in the `0xF9 - 0xFF` range are for different zone expander slots supervisory status:
```
    // 0xE9 - 0xF0: Keypad restored: Slots 1-8
    // 0xF1 - 0xF8: Keypad trouble: Slots 1-8
    case 0xF9: stream->print(F("Zone expander restored: Slot 9")); return;
    case 0xFF: stream->print(F("Zone expander trouble: Slot 9")); return;
```

kricon, 2020-12-02:
> For now I've marked these as "Extended status" and the actual status will be on the 0xEC message, but this decoding concept is open to interpretation:
There are also a few ranges of status messages that will probably get filled in - for example, in printPanelStatus2() it's likely that all messages in the 0xF9 - 0xFF range are for different zone expander slots supervisory status:

Just ran updated sketch today, Extended status showing nicely with correct data afterwards. Verified event buffer, everything was correct and I wasnt able to notice any more Unknown data from it (except for "No entry", 0xAA says Extended status but 0xEC reported event but blank).
Suggestion, maybe change zone expander naming from "Slot 9" onto "Zone Expander 1"?
If it helps you, here are more zone expander supervisory status - took yesterday (I can took supervision logs for all zone expanders from 1-7 if needed):
```
 1690.33: 10101010 0 00100000 00110000 00100000 00000011 00000011 01001110 01101110 [0xAA] Event: 078 | 2020.12.01 00:00 | Unknown data //Zone expander 5 supervisory trouble
 1690.43: 11101100 0 00000000 00100000 00110000 00100000 00000000 00000011 00000011 01001110 10110000 [0xEC] Event: 078 | 2020.12.01 00:00 | Unknown data //Zone expander 5 supervisory trouble
 1717.22: 10101010 0 00100000 00110000 00100000 00000011 00000010 01010000 01101111 [0xAA] Event: 080 | 2020.12.01 00:00 | Unknown data //Zone expander 4 supervisory trouble
 1717.32: 11101100 0 00000000 00100000 00110000 00100000 00000000 00000011 00000010 01010000 10110001 [0xEC] Event: 080 | 2020.12.01 00:00 | Unknown data //Zone expander 4 supervisory trouble
 1781.54: 10101010 0 00100000 00110000 00100000 00000010 11111101 01100000 01111001 [0xAA] Event: 096 | 2020.12.01 00:00 | Unknown data //Zone expander 5 supervisory restore
 1781.64: 11101100 0 00000000 00100000 00110000 00100000 00000000 00000010 11111101 01100000 10111011 [0xEC] Event: 096 | 2020.12.01 00:00 | Unknown data //Zone expander 5 supervisory restore
 1798.76: 10101010 0 00100000 00110000 00100000 00000010 11111100 01100001 01111001 [0xAA] Event: 097 | 2020.12.01 00:00 | Unknown data //Zone expander 4 supervisory restore
 1798.86: 11101100 0 00000000 00100000 00110000 00100000 00000000 00000010 11111100 01100001 10111011 [0xEC] Event: 097 | 2020.12.01 00:00 | Unknown data //Zone expander 4 supervisory restore
```

I was comparing Event buffer from LCD to DLS V. As per DLS V5, Event 1BF1 seems to be "System reset transmission". I will check how its displayed on PK5500 event buffer, but I think DLS says correctly and it just wasnt implemented into LCD5500 v3.1.

About `Low battery: Zone 35`, the panel where it occured doesnt even support more than 32 zones, and obviously not more than 32 wireless zones. I remember I saw some weird zone number as reporting keyfob or zone rf deliquency on LCD5500 v1.1 (and possibly 1.3). DLS V says *2, which can be true _(viewing troubles *2 --> 7 for Wireless device low battery --> * again for Wireless keyfobs low battery --> * again for keypads low battery --> * again for RF interference on zones PK55xx)_, its obviously not zone 35 but its likely to be wireless related, probably keyfob or rf deliquency. I wont use that panel anymore (so event buffer wont fill out) and I'll check what PK55XX say for that same event. Also, I'll get some drained batteries which will trigger keyfob low battery troubles and report it here.

_Zone fault restored: 40_ on LCD keypad is _*3 By system master_ on DLS V. I can make Zone fault 40 and restore it, took event buffer and keybusreader status if needed. Other messages seems to be correct.

Keypad buffer on LCD keypads and DLS reports *5/*6 access by user code X, but keybusreader says just "*6 programming", without which user code accessed it. But It show for "*1 Master code 40".

In `dscKeybusPrintData.cpp` 0xEB decoding section, you wrote ` *  Byte 4 bit 6-7: Unknown, always observed as: 00`. But that isnt correct. In logs from post above, I see that bits flip sometimes. I didnt discover when. I was changing partitions from 1-8 and they both stay at 00 whole time.

taligentx, 2020-12-06:
>If it helps you, here are more zone expander supervisory status - took yesterday (I can took supervision logs for all zone expanders from 1-7 if needed):

Hi @kricon - thank you!  I've added the supervisory status for zone expanders 1-6.  If you get a chance to setup expander 7 and capture the supervisory trouble/restored data, I can add this as well (zone expander 7's data is normally separate from expanders 1-6).

>Suggestion, maybe change zone expander naming from "Slot 9" onto "Zone Expander 1"?

Makes sense, done.

>I was comparing Event buffer from LCD to DLS. As per DLS, Event 1BF1 seems to be "System reset transmission". I will check how its displayed on PK5500 event buffer, but I think DLS says correctly and it just wasnt implemented into LCD5500.
>
>About Low battery: Zone 35 it is *2 access by master code 40. Sorry for giving wrong information, but LCD5500 was made before access code needed ability added for *2/*3. Zone fault restored: 40 is *3 by system master. Other messages seems to be correct. Some more *1, *2 and *3 access code by user informations:

Okay, these make sense - the low battery zone 35 didn't seem like it belonged in that set of status data.  The status data normally runs in a sequential range, for example based on your access code data with users 1 and 25, I've extrapolated the rest of the user codes including the duress, master, and supervisor codes that are usually tacked onto the end of the user codes 1-32.  Pending confirmation but it makes sense and fits well with the rest of the status data.

For the user codes that I had listed for auto-arm canceled and you're seeing used for *71, I've simplified the labels to just user code as it seems like this access code data is being used for multiple functions.

>Keypad buffer on LCD keypads and DLS reports *5/*6 access by user code X, but keybusreader says just "*6 programming", without which user code accessed it. But it is showing for "*1 Master code 40".

Can you get logs of this?  I'm curious how the access code is being encoded, it may be in a command before or after the 0xAA line showing *6 programming.

>I've partly decoded new cmd 0x57, which is for wireless keyfobs, atleast part of it. No recorded event in memory buffer.

Great!  I've added 0x57 panel and module response decoding.

>In dscKeybusPrintData.cpp 0xEB decoding section, you wrote * Byte 4 bit 6-7: Unknown, always observed as: 00. But that isnt correct. 

Gotcha, I've updated the documentation to just `Unknown`.