taligentx/dscKeybusInterface #184: Decoded cmd 0x6E (HEX data for LCD pgm) and Module/0xE6 cmd key input bug & 0x1B conflict

kricon, 2020-12-10:
CMD 0x6E, panel send it after entering programming section which contain saved HEX data, for example *5 user code programming/communications programming etc.
 ```
Byte1: bits 4-7: Digit 1	Byte1: bits 0-4: Digit 2
Byte2: bits 4-7: Digit 3	Byte2: bits 0-3: Digit 4
Byte3: bits 4-7: Digit 5	Byte3: bits 0-3: Digit 6
Byte4: bits 4-7: Digit 7	Byte4: bits 0-3: Digit 8

2156.17: 01101110 0 00010000 00000000 00000000 00000000 01111110 [0x6E] Unknown data //100000
 2256.52: 01101110 0 00000001 00000000 00000000 00000000 01101111 [0x6E] Unknown data //010000
 2318.87: 01101110 0 00000000 00010000 00000000 00000000 01111110 [0x6E] Unknown data //001000
 2395.04: 01101110 0 00000000 00000001 00000000 00000000 01101111 [0x6E] Unknown data //000100
 2426.32: 01101110 0 00000000 00000000 00010000 00000000 01111110 [0x6E] Unknown data //000010
  127.57: 01101110 0 10101010 10101010 00000000 00000000 11000010 [0x6E] Unknown data //AAAA00

 1179.67: 01101110 0 11010000 11111111 11111111 11111111 00111011 [0x6E] Unknown data //D0FFFFFF
 1179.75: 11111111 1 10100101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Partition 1 Key: //auto-send by keypad to view next 8 digits of section 301/First phone number
 1179.80: 01101110 0 11111111 11111111 11111111 11111111 01101010 [0x6E] Unknown data //FFFFFFFF
 1179.89: 11111111 1 10100101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Partition 1 Key: //auto-send by keypad to view next 8 digits of section 301/First phone number
 1179.94: 01101110 0 11111111 11111111 11111111 11111111 01101010 [0x6E] Unknown data //FFFFFFFF
 1180.02: 11111111 1 10100101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Partition 1 Key: //auto-send by keypad to view next 8 digits of section 301/First phone number
 1180.08: 01101110 0 11111111 11111111 11111111 11111111 01101010 [0x6E] Unknown data //FFFFFFFF
 1180.32: 00000101 0 10000001 11101101 10010001 11000111 10010001 11000111 10010001 11000111 [0x05] Partition 1: Ready Backlight - Input 32 digits | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
```

I noticed that sometimes when I input keys, I get "Unknown data" instead of partition and which key was pressed. The actual key data is on 0xE6 byte3, doesnt matter for which partition. Byte2 is 0xFF every time when 0xE6 cmd send keys.
```
 1328.46: 11111111 1 00000101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Partition 1 Key: 1 
 1179.61: 11111111 1 11111111 10100101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0xE6] Unknown data //key 1 on partition 1
 1055.41: 11111111 1 00001111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Partition 1 Key: 3 
 1142.07: 11111111 1 11111111 00001111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0xE6] Unknown data //key 3 on partition 1
 1142.65: 11111111 1 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0A] Partition 1 Key: 0 
 1056.31: 11111111 1 11111111 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0xE6] Unknown data //key 0 on partition 1
 2660.80: 11111111 1 11111111 11111111 11111111 11111111 11111111 11111111 00010110 11111111 [Module/0x05] Partition 3 Key: 5 
 2660.29: 11111111 1 11111111 00010110 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0xE6] Unknown data //key 5 partition 3
```

0x1B cmd conflict with 0x05: I was messing with some options, trying to decode some more and I enabled keypad blanking with access code needed to remove blanking. I didnt noticed anything keypad-slot related. 
```
cmd 0x05 byte7 bit3: Access code required notification
cmd 0x1B byte4 bits0-7: partition 1-8

 5056.11: 11111111 1 11111111 11111111 11111111 11111111 11111111 11110111 11111111 11111111 [Module/0x05] Unknown notification  //Access code required to remove blanking notification
 5056.19: 11111111 1 11111111 11111111 11111110 11111111 11111111 11111111 11111111 11111111 [Module/0x1B] Module tamper notification //on Partition 1
 4823.37: 11111111 1 11111111 11111111 11111111 11111111 11111111 11110111 11111111 11111111 [Module/0x05] Unknown notification  //Access code required to remove blanking notification
 4823.45: 11111111 1 11111111 11111111 11111101 11111111 11111111 11111111 11111111 11111111 [Module/0x1B] Unknown data //on Partition 2
 5155.07: 11111111 1 11111111 11111111 11111111 11111111 11111111 11110111 11111111 11111111 [Module/0x05] Unknown notification  //Access code required to remove blanking notification
 5155.15: 11111111 1 11111111 11111111 11110111 11111111 11111111 11111111 11111111 11111111 [Module/0x1B] Wireless module battery notification //on Partition 4
```

Just for reference (maybe it can somehow help for updating unlocker sketch?), when I enter *8 programming on pc1832 from partition other than 1 I constantly get unknown data stream from cmds 0xE6.21, 0x0F, 0xE6.01, 0xE6.02:
```
11465.81: 11100110 0 00100001 10000010 11100100 00000000 00000000 00000000 00000000 10000000 11101101 [0xE6.21] Unknown data //constant stream when in *8 programming from partition 2
11465.98: 00001111 0 10000010 11100100 00000000 00000000 00000000 00000000 00000000 01110101 [0x0F] Unknown data //constant stream while being in *8 programming from partition 2
11547.26: 11100110 0 00000001 10000010 11100100 00000000 00000000 00000000 00000000 00000000 01001101 [0xE6.01] Unknown data //constant stream while being in *8 programming from partition 3
11547.34: 11100110 0 00000001 10000010 11100100 00000000 00000000 00000000 00000000 10000000 11001101 [0xE6.01] Unknown data //constant stream while being in *8 programming from partition 3
11633.13: 11100110 0 00000010 10000010 11100100 00000000 00000000 00000000 00000000 00000000 01001110 [0xE6.02] Unknown data //constant stream while being in *8 programming from partition 4
11633.21: 11100110 0 00000010 10000010 11100100 00000000 00000000 00000000 00000000 10000000 11001110 [0xE6.02] Unknown data //constant stream while being in *8 programming from partition 4
```

taligentx, 2020-12-23:
>CMD 0x6E, panel send it after entering programming section which contain saved HEX data, for example *5 user code programming/communications programming etc.

Added!

>I noticed that sometimes when I input keys, I get "Unknown data" instead of partition and which key was pressed. The actual key data is on 0xE6 byte3, doesnt matter for which partition. Byte2 is 0xFF every time when 0xE6 cmd send keys
>
>EDIT: It seems that CMDs 0x27, 0x2D and 0x3E also carries key data, it seems to be on Byte2 bits2+:

I've added keypad keys decoding for the 0xE6 commands in programming and also fixed a bug where the panel command that the module data is sent with could be wrong.  Good catch with the keys on 0x27, 0x2D, and 0x3E, I didn't come across this previously but was able to verify that partition 1 and 2 keys can be sent during these commands.

>0x1B cmd conflict: I have found that when keypad goes idle (after 30sec), it send 0x05 byte7 bit3 for "Keypad going idle notification" and in 0x1B cmd afterwards on byte4 it show on which partition keypad goes idle.

Looks good, I've split out the 0x1B decoding for byte 4 to indicate the keypad partition.

>Just for reference, when I enter *8 programming on pc1832 from partition other than 1 I constantly get unknown data stream from cmds 0xE6.21, 0x0F, 0xE6.01, 0xE6.02 (probably something similar as 0x0A cmd is for partion1):

I've added a check for redundant data in partition 2 installer programming, but partitions 3-8 will still flood messages because of the way that the 0xE6.01-0xE6.06 commands alternate the same command for different ranges zone lights.  It would eat up memory for all sketches just to improve the view in KeybusReader so for now I'll leave it as is.  0xE6.01-0xE6.06 are at least getting decoded now.

Thanks for all this!

kricon, 2020-12-26:
When panel is requesting 3 digit code input, decimal data send/received on 0x6E and 0x70 cmds is on Byte2 for all 3 numbers (0-255). That's the only case I saw where data isn't send/received as hex. I've made Pull Request fixing this, so after entering [005] section for system times instead `1E000000` you'll get `030` like it's displayed on LCD screen.

Nice to have keypad keys decoded for other commands too, It was confusing me for some time. Thanks! I'll report if I saw key send on any other cmd.

Weird thing I noticed when making RF Jam on PC1616 panel with RF5132 and PC5108 modules installed. It say "Zone expander 3" which is incorrect, and impossible to have on PC1616. Just to note, that previously I verified that PC1832 panel can have supervision of zone expander modules on slots which can't use (for example, zone expander 5 for zones 41-48 which cannot adress).
```
  860.34: 11111111 1 00001111 11110000 00111111 11110011 11111111 11111111 11111111 [Module/0x11] Keypad slots: 1 2 7 8 | Zone expander: 1 | PC/RF5132 
  862.48: 11111111 1 11111111 11111111 11101111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Zone expander notification: 3 
  862.68: 00111001 0 11111111 11111111 11111111 11111111 11111111 [0x39] Zone expander query: 3
  862.68: 11111111 1 01010110 01010101 01010101 01010101 01010100 [Module/0x39] Zone expander: 3 | Zones changed: 25 tamper //RF Jamming
```

BTW updates on decoding 0x8D: byte3 0x01 until 0xF7, with guesses for 0xF8 - 0xFC. Byte3 is different for same sections on different RF module versions. Byte5-6-7 seems to be hex data. I'm currently working on 0x94 cmd which seems similar, and probably uses same byte3 data as 0x8D, it holds HEX data on bytes 7-10 with weird offset (data starts on byte7 bit6 and end on byte10 bit7). Is it better to dump my notes for that commands here on this issue, or to open new one to make it look cleaner?

closed 2021-01-04 by kricon