taligentx/dscKeybusInterface #186: [Question] Changed behaviour of Home Assistant sketch

sj-louw, 2020-12-23:
Hi @taligentx 

I am experiencing a slightly different behaviour since I have upgraded (from v1.2 of the lib and example HA sketch) to the latest develop branch lib and example Home Assistant sketch.

Previously when I attempted to arm the alarm (via Home Assistant or manually via posting into the MQTT Set topic) **and there were open Zones**, the primary keypad provided audible feedback (prompting the user to close the open Zones) with a few beeps, not allowing the arming of the alarm. This behaviour feels consistent with how the remotes also behave.

Using the latest develop branch lib and example HA sketch, there is no audible feedback happening. I assume it is due to this  IF statement in the mqttCallback function of the example HA sketch: https://github.com/taligentx/dscKeybusInterface/blob/0fb9b35bbe67e05851b4189ff251398a8bd02237/examples/esp8266/HomeAssistant-MQTT/HomeAssistant-MQTT.ino#L414

Is it possible to get the previous behaviour back?

v3rm0n, 2020-12-28:
Not sure but this might cause an issue that I'm having as well: when the alarm is triggered and a zone is open, I can't disarm it using HA, I have to use the physical alarm panel.

taligentx, 2021-01-02:
Hi @sj-louw,
Using the latest develop branch lib and example HA sketch, there is no audible feedback happening. I assume it is due to this IF statement in the mqttCallback function of the example HA sketch?

dscKeybusInterface/examples/esp8266/HomeAssistant-MQTT/HomeAssistant-MQTT.ino
Line 414 in 0fb9b35
 // Resets status if attempting to change the armed mode while armed or not ready 
and that dsc.write('x'); is never sent until all zones are "closed"/ready to arm.

You're correct, this line checks to see if the panel is ready before sending any arming codes.  One option would be to add an exception for the "Zones open status", for example:

```
Before:
  // Resets status if attempting to change the armed mode while armed or not ready
  if (payload[payloadIndex] != 'D' && !dsc.ready[partition]) {

After:
  // Resets status if attempting to change the armed mode while armed or not ready
  if (payload[payloadIndex] != 'D' && !dsc.ready[partition] && dsc.status[partition] != 0x03) {
```

The list of status codes is at the end of the `HomeAssistant-MQTT` sketch - you could also exclude other status codes if you'd like HomeAssistant to try to arm in non-ready states.

>Not sure but this might cause an issue that I'm having as well: when the alarm is triggered and a zone is open, I can't disarm it using HA, I have to use the physical alarm panel.

Hi @v3rm0n - I wasn't able to replicate this, at least with the current `develop` branch, I armed a partition stay, opened a zone, let the entry delay expire so the alarm triggered, and with the zone still open was able to disarm through HA.  One way to help debug would be to get an MQTT client and monitor the `dsc/Set` topic - you can verify if HA is sending the disarm code (`1D` for partition 1, `2D` for partition 2, etc).

taligentx, 2021-01-22:
Feel free to re-open if there are any continuing issues under the new 2.0 release.

closed 2021-01-22 by taligentx