taligentx/dscKeybusInterface #189: Panic/Aux Changed

TrendMend, 2020-12-29:
I was just wondering if this library had a dsc.****Changed and dsc.*****Trouble (or other) functions for the panic and auxiliary alarms? (similar to the fire alarm). I am looking to output via mqtt the status of the alarms, but it can only call the Aux/Panic if statement if the function dsc.keypad(Aux/Panic)Alarm is true, this leaves me with a non-resettable statement. 

This of course could be solved on my end by just toggling the statement, but I wanted to see if the library had something similar and suggest to add it if not.

Happy Holidays!

TrendMend, 2020-12-29:
You know what? I'm dumb. I don't even know if I could toggle it (unless the library has a way to detect the master code being entered... Will have to check on that).

And I understand if it can't be implemented if it's due to a decoding issue, no problem. I was just hoping something like that could be added.

taligentx, 2021-01-01:
Hi @TrendMend - the panel sends out a single notification if a keypad's F/A/P alarm keys are pressed and then immediately restores their original status, so from the panel's perspective, these are one shot notices.  For example on the PC1864:

```
  241.70: 10100101 0 00100001 00000100 00101111 10011100 01010000 10010001 01110110 [0xA5] 2021.01.01 15:39 | Keypad Panic alarm
  241.74: 10000111 0 11111100 11110000 01110011 [0x87] PGM outputs enabled: 5 6 7 8 9 10 11 12 13 14 
  241.83: 11100110 0 00011010 01000000 00000001 00000000 00000001 00000000 00000000 00000000 01000010 [0xE6.1A] Partitions in alarm: 1 
  241.86: 01110101 0 10000000 11110101 [0x75] Partition 1 | Tone: constant tone 
  241.93: 11001110 0 00100000 10011100 01010000 00000000 00000000 11011010 [0xCE] Keypad Panic alarm
  242.02: 00000101 0 10010101 00010001 10010001 11000111 10010001 11000111 10010001 11000111 [0x05] Partition 1: Ready Memory Trouble Backlight - Partition in alarm | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
  242.11: 11101011 0 00000000 00100001 00000100 00101111 10011100 00000000 01010000 10010001 10111100 [0xEB] 2021.01.01 15:39 | Keypad Panic alarm
  242.26: 10100101 0 00100001 00000100 00101111 10011100 01010100 00000000 11101001 [0xA5] 2021.01.01 15:39 | Keypad Panic alarm restored
  242.45: 11101011 0 00000000 00100001 00000100 00101111 10011100 00000000 01010100 00000000 00101111 [0xEB] 2021.01.01 15:39 | Keypad Panic alarm restored
```

The library keeps track if these occurred (as seen in the `Status` sketch), but there's no need for a "...Changed" variable because of their one-shot nature - if any of the keypad alarms are set true (dsc.keypadFireAlarm, etc), it must be a new state.  The sketch should set these to false once they're handled so the sketch is ready for the next time the keys are pressed.

TrendMend, 2021-01-02:
Happy new years!
@taligentx Good to know, thanks. To provide some more insight I was looking to detect when the panic was "restored" as that keybus snippet states. I am assuming this is via means of entering the master code? When I trigger my panic alarm, due to my programming it trips the alarm and this is reset once I enter my master code. Would I then be looking for a restored state by the master code being entered? Is this possible with your library? Thank you very much!

taligentx, 2021-01-02:
Happy new year!  The keypad panic key state resets immediately, but the panic key triggers the partition alarm, which is what you're disarming when you enter the master code.  So you can use the normal  partition alarm status check (alarm triggered or disarmed) - from the `Status` sketch:
```
      // Checks alarm triggered status
      if (dsc.alarmChanged[partition]) {
        dsc.alarmChanged[partition] = false;  // Resets the partition alarm status flag
        if (dsc.alarm[partition]) {
          Serial.print(F("Partition "));
          Serial.print(partition + 1);
          Serial.println(F(": Alarm"));
        }
        else if (!dsc.armedChanged[partition]) {
          Serial.print(F("Partition "));
          Serial.print(partition + 1);
          Serial.println(F(": Disarmed"));
        }
```
If you needed to do something different if the partition is disarmed specifically after a keypad panic alarm, you could move the keypad panic check into the above function - something like this pseudocode:
```
...
else if (!dsc.armedChanged[partition]) {
  if (dsc.keypadPanicAlarm) {
    dsc.keypadPanicAlarm = false;
    // Do something here if disarmed after a keypad panic key
  }
  else {
    // Do something if normally disarmed
  }
```

TrendMend, 2021-01-02:
That's perfect! I'll incorporate that as soon as I can, thanks for all of your help. I'll close this issue for now as everything looks good. Thank you!

closed 2021-01-02 by TrendMend