taligentx/dscKeybusInterface #190: Release candidate feedback

taligentx, 2021-01-05:
Hi folks,

The current `develop` branch is set for release, please post here if you encounter issues with this branch.  For example, VirtualKeypad-Blynk and VirtualKeypad-Web are likely areas for bugs as they have both been updated to support displaying zone lights in alarm memory/programming and displaying the event buffer.  Thanks!

kricon, 2021-01-06:
There is one major issue with Blynk QR-codes even since the beginning which is targeting (probably all) Android users, as stated in #41 - QR codes from examples doesn't want to scan on Android phons (I tested on few phones). Scanning on iOS works like charm.
Sharing project from Android Blynk version and capturing it's shared QR code works fine for both for Android and iOS users. I've attached QR codes for project examples, untouched and unchanged, just scanned on iOS and shared from Android and I'm suggesting adding it instead of current ones which works just for iOS users: [16 zone example](https://user-images.githubusercontent.com/56011330/103716039-5c8e2000-4fc2-11eb-8f19-02e1550ab2d5.png) - [32 zone example](https://user-images.githubusercontent.com/56011330/103716057-644dc480-4fc2-11eb-9e95-368514db5fc0.png) - [8 zone example](https://user-images.githubusercontent.com/56011330/103716065-6adc3c00-4fc2-11eb-9620-cff04b8e0ef8.png)

I've also added "*2 Sensors with RF Delinquency" message in Pull Request #191 together with suggestion for using _V20_ (Event buffer time/date/partition info) display for 0x6E data printout when panel is requesting X-digit input and clearing it when panel is in *8 installer section/subsection menu which mean user have finished inputing digits so there is no point of having them displayed anymore.

Another quirk, if you type too fast, for example pressing *85555, it will skip the "Enter * function code" message and thus skip changing zone led lights behavior to follow program/memory-led lights. So instead getting yellow zone lights for programming/memory leds, it will still be green and follow zone openings even while programming.

Worth noting: Zone leds will be green/open even if the zone is not associated to any partition (disabled from system), but their zone definition is not set to 000.

taligentx, 2021-01-06:
Hi @kricon,
> There is one major issue with Blynk QR-codes even since the beginning which is targeting (probably all) Android users, as stated in #41 - QR codes from examples doesn't want to scan on Android phons (I tested on few phones). 

This has been odd as I've never been able to replicate this with the Android phones I've tested.  I've extracted the text data from the original QR codes and manually generated larger versions, perhaps this will help.  If the Blynk template has too many elements to store the template directly in a QR code, Blynk instead generates a link to get the template from the local app server.  For example your 32 zone QR code decodes to:
```
blynk://token/clone/SuyqXX-xDJSfMF_dELSPK6gVJe6brxW1?server=192.168.0.30&port=9443
```

> I've also added "*2 Sensors with RF Delinquency" message in Pull Request #191 together with suggestion for using V20 (Event buffer time/date/partition info) display for 0x6E data printout when panel [...]

Thanks, merged!  As far as handling the 0x6E messages and hex keys for programming, I've added looking at this to the TODO list for the release after this one, it'll be interesting to see if installer programming is fairly straightforward or if it's a rabbit hole of special cases on a per menu basis.  

>Another quirk, if you type too fast, for example pressing *85555, it will skip the "Enter * function code" message and thus skip changing zone led lights behavior to follow program/memory-led lights

I've seen things like this happen with Blynk in general, there's a high latency between actions and responses - perhaps there's a way to block the app entirely until the Blynk server and microcontroller catch up.

> Worth noting: Zone leds will be green/open even if the zone is not associated to any partition (disabled from system), but their zone definition is not set to 000.

For now the zone states are derived directly from the panel commands and doesn't do any processing to try to figure out which zones are enabled, tied to certain partitions, etc.  Tracking this would increase memory usage so it would be down to which use cases would benefit from this info.

Thanks for the testing!

kricon, 2021-01-06:
> I've extracted the text data from the original QR codes and manually generated larger versions, perhaps this will help.

I act dumb by not checking QR codes before submiting, my bad. I knew it was suspicous seeing that small Blynk QR codes.
However issue is still present even after new manually-generated QR code versions - it doesn't work on my Sony phone but works on old iPad2.

> As far as handling the 0x6E messages and hex keys for programming, I've added looking at this to the TODO list for the release after this one, it'll be interesting to see if installer programming is fairly straightforward or if it's a rabbit hole of special cases on a per menu basis.

0x6E cmd get send only after "Input X digits" status message (even section [900] to get panel version info send "Input: 4 digits" before 0x6E containing version info). After input entry is done (or exited by #), panel send status message "*8 installer programming" or "*8 installer pgm subsection" depending of section you're into. Then clearing of _V20_ should be done, just like resetZones is done from statusMessages.
It should be straightforward and easy so I've tried doing it by myself (atleast, just for  3-digit input which are handled the same way as event number - whole byte2 is used for 0-255 decimal value), but I just can't get 0x6E cmd called, even this code doesn't work (virtualkeypad-blynk.ino) and I dont know what I'm missing:
```
// Processes status data not natively handled within the library
void processStatus() {
   switch (dsc.panelData[0]) {
    case 0x6E: Blynk.virtualWrite(V20, "data output test"); break;
...
```
Sorry about hex key input, I was stupid by forgotting I can input * 1 * for A ... * 6 * for F. Works like charm, I can program HEX data successfuly thru Blynk app. It's just hard to know what you're programmed without being able to get and read data from panel (cmd 0x6E).

For tracking zones and blynk lag, I just give feedback what I've noticed which can maybe be helpful in future. I don't see a point tracking valid active zones, but for Blynk lag it would be nice to have little note in .ino file for others to know as at first I thought yellow programming leds don't work because I quickly entered *85555. Another thing I noticed in Blynk and was able to reproduce few times, entering *8 5555 302 (just an example) and then quickly exiting and entering *8 again (probably before resetZones loop thru all zones) I got several yellow zone lights (5, 9, 10, 11, 15, 16, 21, 25, 26...) which doesnt seem to be related to anything.

taligentx, 2021-01-06:
> However issue is still present even after new manually-generated QR code versions - it doesn't work on my Sony phone but works on old iPad2.

Strange - for example, I tested with an inexpensive ZTE Blade 10 Prime and was able to read the QR codes reliably, it seems like this will be monitor/device dependent.  Added a note in the sketch documentation.

> 0x6E cmd get send only after "Input X digits" status message (even section [900] to get panel version info send "Input: 4 digits" before 0x6E containing version info). After input entry is done (or [...]

The LCD keypads send out a request for data (key 0xA5), then the panel sends 0x6E data, one section at a time - here's a normal keypad vs LCD keypad entering installer programming 301 for phone number 1:
```
LED keypad:
  431.72: 10100101 0 00100001 00000100 10101111 01000101 10101101 11111111 01101010 [0xA5] 2021.01.05 15:17 | Enter installer programming
  432.07: 11101011 0 00000000 00100001 00000100 10101111 01000100 00000001 10101101 11111111 10110000 [0xEB] 2021.01.05 15:17 | Enter installer programming
  435.73: 11111111 1 00001111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Partition 1 Key: 3 
  436.23: 11111111 1 00000000 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Partition 1 Key: 0 
  436.26: 11000011 0 00000000 11111111 11000010 [0xC3] Keypad ready
  436.76: 11111111 1 00000101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Partition 1 Key: 1 
  436.79: 01100100 0 00000110 01101010 [0x64] Partition 1 | Beep: 3 beeps
  436.87: 00001010 0 10000001 11101101 00001111 00000000 00000000 00000000 00000000 10000111 [0x0A] Partition 1: Ready Backlight - Input: 32 digits | Zones 1-32 lights: 1 2 3 4 
  436.96: 11100110 0 00100000 10000001 11101101 00000000 00000000 00000000 00000000 10000000 11110100 [0xE6.20] Partition 1: Ready Backlight - Input: 32 digits | Zones 33-64 lights: none
  437.04: 00000101 0 10000001 11101101 10010101 11000111 00010000 11000111 00010000 11000111 [0x05] Partition 1: Ready Backlight - Input: 32 digits | Partition 2: disabled | Partition 3: disabled | Partition 4: disabled
  438.17: 10110001 0 01111111 11111111 11111111 11111111 10000000 00000000 00000000 00000000 10101101 [0xB1] Enabled zones 1-32 | Partition 1: 1 2 3 4 5 6 7 9 10 11 12 13 14 15 16 17 18 19 20 21 22 23 24 25 26 27 28 29 30 31 32 | Partition 2: 8 

LCD keypad:
  519.87: 10100101 0 00100001 00000100 10101111 01001101 10101101 11111111 01110010 [0xA5] 2021.01.05 15:19 | Enter installer programming
  520.22: 11101011 0 00000000 00100001 00000100 10101111 01001100 00000001 10101101 11111111 10111000 [0xEB] 2021.01.05 15:19 | Enter installer programming
  521.20: 11111111 1 11111111 11110111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Partition 2 Key: Menu navigation 
  521.61: 11111111 1 11111111 11110111 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0xE6.21] Partition 2 Key: Menu navigation 
  522.42: 11111111 1 11111111 11110111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0F] Partition 2 Key: Menu navigation 
  522.50: 11111111 1 11111111 00001111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0F] Partition 2 Key: 3 
  522.59: 11111111 1 11111111 00000000 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0F] Partition 2 Key: 0 
  522.67: 11111111 1 11111111 00000101 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x0F] Partition 2 Key: 1 
  522.70: 01101001 0 00000110 01101111 [0x69] Partition 2 | Beep: 3 beeps
  522.78: 00001111 0 10000001 11101101 00001111 00000000 00000000 00000000 00000000 10001100 [0x0F] Partition 2: Ready Backlight - Input: 32 digits | Zones 1-32 lights: 1 2 3 4 
  522.87: 11100110 0 00100001 10000001 11101101 00000000 00000000 00000000 00000000 10000000 11110101 [0xE6.21] Partition 2: Ready Backlight - Input: 32 digits | Zones 33-64 lights: none
  522.95: 00000101 0 10010001 11000111 10000001 11101101 00010000 11000111 00010000 11000111 [0x05] Partition 1: disabled | Partition 2: Ready Backlight - Input: 32 digits | Partition 3: disabled | Partition 4: disabled
  522.97: 11111111 1 11111111 10100101 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0x05] Partition 2 Key: Receive data 
  523.01: 01101110 0 11011111 11111111 11111111 11111111 01001010 [0x6E] LCD display: DFFFFFFF
  523.17: 11111111 1 11111111 10100101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0xE6.21] Partition 2 Key: Receive data 
  523.23: 01101110 0 11111111 11111111 11111111 11111111 01101010 [0x6E] LCD display: FFFFFFFF
  523.40: 11111111 1 11111111 10100101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0xE6.21] Partition 2 Key: Receive data 
  523.45: 01101110 0 11111111 11111111 11111111 11111111 01101010 [0x6E] LCD display: FFFFFFFF
  523.62: 11111111 1 11111111 10100101 11111111 11111111 11111111 11111111 11111111 11111111 11111111 [Module/0xE6.21] Partition 2 Key: Receive data 
  523.67: 01101110 0 11111111 11111111 11111111 11111111 01101010 [0x6E] LCD display: FFFFFFFF
```

0x6E only sends 8 characters at time, so for sections like 301 with more characters, the keypad has to keep sending the 0xA5 key to keep getting more data until the keypad itself decides that it has received enough data.  I've added the 0xA5 key as the 'L' key to the library, pressing this will trigger 0x6E but o [... cut ...]
So it seems like a sketch would need to keep track of the status to see how many digits are being requested and send the receive key the appropriate number of times - the part to investigate is how the 0x05 status messages specifying the number of input digits maps to the number of 0xA5 keys sent, it's likely 1 0xA5 key per 8 characters requested.

And feedback is always appreciated!  The Blynk latency is one of the quirks that would likely need something like a native iOS/Android app to resolve.

kricon, 2021-01-07:
> Strange - for example, I tested with an inexpensive ZTE Blade 10 Prime and was able to read the QR codes reliably, it seems like this will be monitor/device dependent. Added a note in the sketch documentation.

I can check on few other Android devices if needed (I don't see how it would help but whatever), but on two I've tested back then and now it didn't work. Example is enough documented that making own Blynk project from scratch shouldn't be that hard - doable in 10ish min.

> The LCD keypads send out a request for data (key 0xA5), then the panel sends 0x6E data, one section at a time - here's a normal keypad vs LCD keypad entering installer programming 301 for phone number 1:

Yeah, I know that, but I was missing 0xA5 cmd to trigger 0x6E. I've tried looping 0xA5 key send after panel requesting 32 digit data and sucessfully merging 0x6E output into one string, but there was major conflict with programming from real LCD keypad as both sketch and keypad will send 0xA5 cmd. If 0x6E cmd isn't received (by triggering from real LCD keypad) after entering section requesting 32 digit input in second-or-two, only then the sketch should send 0xA5 cmd. I've verified that panel send all 32 characters again after 4 consecutive 0xA5 key presses. And you're correct about specifying number of input digits, its specified on "Enter X digit input" message send before 0xA5 key and receiving the 0x6E data. I made a pull request #192  with LCD data printout on VirtualPin20 of blynk sketch, except when panel requesting 32 digit input.

Well, it's topic for itself so better to keep this issue clear for real feedback and issues rather than possible future features.
And one of them is major issue I recently noticed - pausedZones and yellow-zone-light-mode doesnt work on partitions other than 1.

> The Blynk latency is one of the quirks that would likely need something like a native iOS/Android app to resolve.

In pull request #192 I put 0xA5 (enter master code for *6) and 0xB7 (enter installer code) to call pauseZones in processStatus, so if you have fast fingers and enter *8 or *6 too quickly, leds are still going to be disabled and sketch will catch up.

closed 2021-01-22 by taligentx