taligentx/dscKeybusInterface #199: Exception running VirtualKeypad-Web 1.3 (esp8266)

borjaburgos, 2021-02-02:
Hi there! First of all thanks for this terrific project. 

I get the following exception while running VirtualKeypad-Web on an ESP-8266. I have experienced the same on both a D1-mini and a NodeMCU. 

For context, I get this exception while running this example without connecting it to the DSC Panel. I haven't tried connecting the ESP to my alarm panel yet and running this example – this may work. Nonetheless, I believe this here is probably undesired behavior even if the board isn't connected to the DSC alarm panel.

**To reproduce:**

- Version: 2.0.0

(1) Load VirtualKeypad-Web example on a ESP-8266 board (tried on NodeMCU and D1-mini).
(2) Load example on your browser (http://dsc.local)
(3) Click two buttons on web UI panel. Second button press will result in this exception below.

```
0x40100d76: millis() at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x40217a8d: __yield() at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 135
0x4020cd60: dscKeybusInterface::loop() at /Users/borjaburgos/Documents/Arduino/libraries/DSC_Keybus_Interface/src/dscKeybusInterface.cpp line 161
0x4020d087: dscKeybusInterface::write(char const*, bool) at /Users/borjaburgos/Documents/Arduino/libraries/DSC_Keybus_Interface/src/dscKeybusInterface.cpp line 464
0x40201f6d: onWsEvent(AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at /Users/borjaburgos/Documents/Arduino/libraries/DSC_Keybus_Interface/examples/esp8266/VirtualKeypad-Web/VirtualKeypad-Web.ino line 1573
0x40100b5c: ets_post(uint8, ETSSignal, ETSParam) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100b5c: ets_post(uint8, ETSSignal, ETSParam) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x4010139b: umm_free_core(void*) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 351
0x4010139b: umm_free_core(void*) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 351
0x4022099e: std::_Function_handler ::_M_invoke(std::_Any_data const&, AsyncWebSocket*, AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x40100d76: millis() at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x402211c5: AsyncWebSocket::_handleEvent(AsyncWebSocketClient*, AwsEventType, void*, unsigned char*, unsigned int) at /Users/borjaburgos/Documents/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp line 877
0x4020f769: AsyncWebSocketClient::_onData(void*, unsigned int) at /Users/borjaburgos/Documents/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp line 704
0x40100d76: millis() at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_wiring.cpp line 188
0x4020f795: std::_Function_handler ::_M_invoke(const std::_Any_data &, void *, AsyncClient *, void *, unsigned int) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/xtensa-lx106-elf/include/c++/4.8.2/functional line 2073
0x4021546e: AsyncClient::_recv(std::shared_ptr &, tcp_pcb*, pbuf*, long) at /Users/borjaburgos/Documents/Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp line 649
0x40215616: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at /Users/borjaburgos/Documents/Arduino/libraries/ESPAsyncTCP/src/ESPAsyncTCP.cpp line 738
0x4022c140: tcp_input at core/tcp_in.c line 501
0x40101658: malloc(size_t) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 552
0x40231d1a: ip4_input at core/ipv4/ip4.c line 1467
0x40101623: free(void*) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 398
0x40228d39: ethernet_input_LWIP2 at netif/ethernet.c line 188
0x40228b58: esp2glue_ethernet_input at glue-lwip/lwip-git.c line 469
0x40249242: ethernet_input at glue-esp/lwip-esp.c line 365
0x40249253: ethernet_input at glue-esp/lwip-esp.c line 373
0x4020afa0: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::addDomainCacheItem(void const*, bool, unsigned short) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Structs.cpp line 2445
0x40101658: malloc(size_t) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 552
0x4020b924: esp8266::MDNSImplementation::MDNSResponder::_udpAppendBuffer(unsigned char const*, unsigned int) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1096
0x4020b924: esp8266::MDNSImplementation::MDNSResponder::_udpAppendBuffer(unsigned char const*, unsigned int) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1096
0x4020bc3d: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSRRDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNS_RRDomain const&, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1342
0x4020bdfc: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSServiceDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService const&, bool, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1440
0x4020b924: esp8266::MDNSImplementation::MDNSResponder::_udpAppendBuffer(unsigned char const*, unsigned int) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1096
0x4020b952: esp8266::MDNSImplementation::MDNSResponder::_udpAppend8(unsigned char) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1112
0x4020ba95: esp8266::MDNSImplementation::MDNSResponder::_write8(unsigned char, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1254
0x4020bdb1: esp8266::MDNSImplementation::MDNSResponder::_writeMDNSServiceDomain(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService const&, bool, bool, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/libraries/ESP8266mDNS/src/LEAmDNS_Transfer.cpp line 1426
0x40100b5c: ets_post(uint8, ETSSignal, ETSParam) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40101658: malloc(size_t) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/umm_malloc/umm_malloc.cpp line 552
0x40100b38: ets_intr_unlock() at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 163
0x40100b5c: ets_post(uint8, ETSSignal, ETSParam) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100b5c: ets_post(uint8, ETSSignal, ETSParam) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100b5c: ets_post(uint8, ETSSignal, ETSParam) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100b5c: ets_post(uint8, ETSSignal, ETSParam) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
[... cut ...] nctional line 2069
0x40100b5c: ets_post(uint8, ETSSignal, ETSParam) at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 177
0x40100b7d: esp_schedule() at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 125
0x40217b60: loop_wrapper() at /Users/borjaburgos/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/cores/esp8266/core_esp8266_main.cpp line 197
```

taligentx, 2021-02-04:
Hi @borjaburgos - thanks for the detailed info and exception decoding!  I've added a Keybus connectivity check to `VirtualKeypad-Web` in the develop branch, this ensures that the sketch doesn't get locked up trying to write to a non-existent interface.

hcoiro, 2021-04-12:
Hi, I have been testing and modifiying the VirtualKeypad-Web version 2.0 on an ESP-8266 code to be able to send notification using pushme.now.sh  web to send notification when alarm is triggered or change status.
As when doing development, my ESP-8266 is not connected to the pannel.  I had the same problem: Click two buttons on web UI panel. Second button press will result in this exception.  I see that the comenting the line dsc.write() that is at the function that catch web comunication , right after decoding the input, you won´t get the Exception, So for some reason, that line after second command sent without attached to the pannel, make the ESP-8266 reboot.

dwarf2k, 2021-04-12:
hcoiro,
I had the same problem but with the ESP-32 version.  I found that adding a check to make sure it is connected before sending data aids in testing while disconnected as well as allowing it to send an error announcement if it happens while it should be connected.

if (dsc.keybusConnected) 
   dsc.write(sep_at + 1);
else
 ....

hcoiro, 2021-04-12:
Thanks dwarf2K,
I will add that condition, good idea.

closed 2022-03-08 by taligentx