taligentx/dscKeybusInterface #200: KeypadInterface: use DSC keypads without a panel for general purpose inputs - Feedback/testing

taligentx, 2021-02-03:
Hi folks,

Post issues/feedback for the `KeypadInterface` sketch in the `develop` branch - this part of the library allows a microcontroller (tested on Arduino Uno, esp8266, esp32, esp32-s2) to directly interface with a DSC keypad without a DSC panel.  Currently supported features:

* Read keypad key button presses, including fire/aux/panic alarms
* Set keypad lights: Ready, Armed, Trouble, Memory, Bypass, Fire, Program, Backlight, Zones 1-8

This uses a different wiring setup with a second NPN transistor on the clock line so the microcontroller can manage both the clock and data lines, the wiring diagram is in the sketch.

This is a basic sketch for testing, there are still features on the keypads that are currently not implemented (buzzer, tones, perhaps using the 16x2 LCD on the LCD keypads, etc).  PowerSeries only for now, though Classic series keypads are simpler and should be possible to support.

taligentx, 2021-02-05:
Added MQTT example sketches and added features:
* Set keypad beeps, 1-128
* Set keypad buzzer in seconds, 1-255
* Set keypad tone pattern with a number of beeps, an optional constant tone, and the interval in seconds between beeps

jimtng, 2021-02-13:
Will this still work when the keypad is connected to the panel? e.g. Can we make the keypad beep? That would be quite handy for chime alerts.

taligentx, 2021-02-14:
> Will this still work when the keypad is connected to the panel? e.g. Can we make the keypad beep? That would be quite handy for chime alerts.

That'd be an interesting feature.  This sketch entirely replaces the DSC panel by generating both the clock and data signals, there would be a conflict if connected to another panel - it would be like having two DSC panels on the same installation.  

This might be possible if there's a way to tell the DSC panel to stop transmitting on the Keybus long enough to send a new command.  Normally the panel (and this sketch) send a constant stream of commands, it'd be interesting if to see if there's a way to stop the panel temporarily.

kricon, 2021-02-15:
I've been messing with LCD5500 keypad messages and modified the code (available in my repo if anyone interested) so I can print panel status messages onto 0x05 cmd byte3 with typing HEX data into serial. I expected it to match `printPanelMessages` from `dscKeybusPrintData.cpp` (and decode the rest data to complete missing gaps in list) but it seems that is not case, atleast not fully. It match fine for most used/important statuses like Ready, Stay zones open, Armed stay/away, entry/exit delay, keypad blanking etc.

You can set LCD keypad clock by sending 0x2A to cmd 0x05/byte3. There will be some delay after you initially send cmd before LCD display will change to "Set date and time (HH:00 MM:00)" message. That is because keypad sends "Receive data" key to the sketch, but sketch doesn't do anything so keypad timeouts then continues with all zeros. Probably it can be made that sketch can answer with local PC date and time. You can enter date/time on keypad as usual, it will be saved after last digit entry and will advance by itself (probably following keypad own oscillator clock or clock line generated from sketch?)

I also discover that you're able to enter *6 user functions menu by sending 0x29 to cmd 0x05/byte3, and from there you can scroll to change keypad brightness/contrast/buzzer level. It works just as if keypad was connected onto real panel. Why sending 0xA9 doesn't work, I don't know - do anyone have any idea on that?

Some other things I discover: sending 0x70-0x74 to cmd 0x05/byte3 brings function key 1-5 menu where you can change key output for function keys. Have in mind that, after entering 2 digits, keypad send submit data key but sketch doesnt do anything so it's still stuck at same LCD message screen. New entered key data is saved to keypad and works just fine.

My initial idea/goal when I started to mess with LCD messages was to decode rest unknown messages. I know that keypad looks for more data than just 0x05 cmd, but I did found some new status messages, for example 0x13 is probably _Walk test active or zones open_ and 0x18 could be: _Cannot silence waterflow alarm_. But after some time (hours) messing with sending all kind of different HEX data onto cmd 0x05/byte3, I was able to get _Cannot silence waterflow alarm_ message also on 0xC1 and 0x41 and I wasn't able to reproduce that. Same time, I was able to get LCD messages when sending 0xB2-0xB6 and they were as follows:
`0xB2: Select output command O/P 1; 0xB3: Output control in progress; 0xB4: Command O/P 2 in progress; 0xB5: Command O/P 3 is enabled; 0xB6: Command O/P 3 is disabled`. I think that answers why PC1565-2p was sending 0xB3, and what that exactly mean #187. But I was also unable to reproduce that, and I'm sure I was able to get it displayed using this sketch and changing byte3 of 0x05cmd. My logs shows that keypad was following `printPanelMessages` for some time but why it isn't reliable each time, I don't know.

As I had two ESP's running, one with `KeypadInterface` and another with `KeybusReader` examples, I get constant stream of `Module/05 tamper notification` if I leave the one with `KeybusReader` running while reconnecting the one with `KeypadInterface` sketch on it. Fixed by first connecting ESP with `KeypadInterface` on it and then ESP with `KeybusReader` sketch. Both ESP's was powered from USB. It doesn't seem to do that every time, just when powerseries keypad reboots itself as it detected keybus fault and then recover.
Another thing that happen to me was when keypad became disconnected - `KeybusInterface` would repeatedly log that key 0 was pressed but keypad was diconnected. I don't remember if it was with classic or powerseries keypad.

Testing with PC1500RK keypad, I found that quick-press on keypad numerical keys (especially top row) will make keypad buzzer constant (until another keypress). However, it will correctly log digit which was pressed just once in serial monitor. Holding the digit will repeat beep and log it again in serial monitor as it should, just really quick-tap on digit seems to glitch the keypad. I don't know if that's issue with all keypads (connected to real panels - currently I don't have panel to test with) or if it's case only with this sketch.

smpolymen, 2022-09-25:
Would it be possible to enable both the dscKeypadInterface and the dscKeybusInterface at the same time (Specifically dscClassicInterface and dscClassicKeypadInterface?) I'm trying to relay the keypad via my Arduino to the panel. That way I can hijack the keypad so it can send keys to either the panel or to other functions in my sketch based on a variable. This doesn't seem to work at the moment due to:
dscKeybusInterface.h:
#if defined dscClassicSeries
#elif defined dscClassicKeypad

and the simultaneous use of Timer1.

Do you have any suggestions on how to get around this? I'm pretty new to Arduino and working with ISRs.