taligentx/dscKeybusInterface #202: esp8266 restart problems!

javierferwolf, 2021-02-17:
The problem of constant restart of at least 3 times a day and sometimes up to many more times! I have tested on various PC858 panels and i always get the same problem esp89266 reboot!
I have fixed the serial port error and got the following Debug Messages
ets Jan 8 2013,rst cause:4, boot mode:(3,6) esp8266
wdt reset
load 0x4010f000, len 1392, room 16
tail 0
chksum 0xd0
csum 0xd0
v3d128e5c
~ld

Module: [Generic ESP8266 Module|Nodemcu]
Development Env: [Arduino IDE 1.8.13]
Core Version: [2.7.4] and tested too 2.6.3

I have tried different power supplies and esp8266 modules nodemcu! and i get the same problem!
please is there any solution or someone has the same problem?

thank you!

kricon, 2021-02-19:
I did a bit of google search on that (watchdog reset), and found few possible causes. I must note that I haven't yet tried Telegram sketch. Have you tried other examples, does same thing happens on them?

It could be power-supply related, implying that wiring is good (soldered, no breadboard/dupoint wires and powersupply close to nodemcu).
Voltage fluctuation, some recommends adding 47mF capacitors on rst/vcc pins to gnd
Have you tried powering NodeMCU just from USB? Does same thing happen when powered from 3v3 or 5V from seperate power supply, what are you using as a voltage supply?

It could also be Code-related, have you done any code modifications, or changing pins?
You can try to Erase all flash content and reupload the sketch - it fixed issue to someone.
Verified that you're using correct library resources and correct example (from esp8266 folder).
Someone stated it could be a bad ESP8266 module, but you said you tried on few different ones...Does ESP8266 get hot?

Hope it helps as thats all I can think of, maybe other can help you more.

javierferwolf, 2021-02-20:
thank you @kricon for your answer! I will take into account the recommendation of the 47mF capacitors on rst/vcc pins to gnd!

I have tried power supplies of all kinds from 5v adapters and from the USB computer!
even the current dividers of the 10 and 15kohms resistors that connect to the green and yellow of the panel, I have replaced it with optocouplers but the problem keeps happening.
The telegram sketch that is in the example of this library I have not changed anything I just copied and pasted but I have not changed any pin.

I have used 5 modules esp8266! 2 nodemcu and 3 only esp12-E but in all the problem happens I have even tried with different PC585 panels.
Esp8266 modules do not heat up!

kricon, 2021-02-20:
I forgot capacitor on EN pin. Value isn't critical, it can be anywhere 47mF-100mF.
Current dividers on ESPs should be **33k and 10k**, 15k/10k are for Arduinos which run on 5V.
PC585 is older series with linear voltage regulation, and it keybus operates at about 13.6V while newer pc1616/1832/1864 operates at slightly a lower voltage. You must drop that voltage to ESP8266-safe voltage of about 3.3V. Using 15k/10k resistors, singal voltages on PC585 can go up to 5.5V high and can burn ESP8266 or it can cause issue like you're having to self-restart.
You can try to run some other example, like `Status` or `KeybusReader` for a day and see if it also causes ESP8266 to restart.
BTW: just curious, which optocoupler did you use? There has been few folks around which tried to use optocouplers and it didn't work for them with this project.

Dilbert66, 2021-02-20:
In my experience with esp8266's one of the main culprits for hardware WDT reboots have usually been ISR routines taking too long to run. The ESP is very sensitive to timing due to it's other background processes with WIFI connections.  I have not seen this with this library but have seen it in other code.  It's possible the telegram example  you mentioned using needs some optimizing to keep the WDT timers fed regularly.

taligentx, 2021-05-24:
Hi @javierferwolf - is this still an issue when testing with the other example sketches, or is it only happening with Telegram?

javierferwolf, 2021-05-25:
Hi @taligentx - yes, indeed the ESP8266 continues with the reboot problem, I don't know if this is related to the instability of the internet connection. So my question, how can I monitor the internet to have a reference and rule out that the problem is the internet?  apparently I have a stable connection but I don't know if it is stable enough for the ESP8266

javierferwolf, 2021-06-20:
I have been absent for several days but finally I have had time and I have seen your message thank you @hcoiro  for your answer!
It is certainly a nuisance since there are days the messages sent by telegram of reboot are too many!
I'm a bit of a newbie to arduino programming so I'm not sure where in the sketch I should modify or add what you mentioned.
Please can you specify in which part of the sketch I should put your recommendations?
thank you!