taligentx/dscKeybusInterface #216: Can two or more DSC Power Series alarm panels be integrated into home assistants?

sensormatic1985, 2021-04-12:
Is it possible to integrate 2 or 3 alarm panels from a single Home Assistant server?
I tried to integrate, but I could not separate the two panels, that is, any of the 2 activated a single automation, I want to work on a security project by blocks, but the idea is to be able to centralize everything in a single home assistants server

Dilbert66, 2021-04-12:
You didnt  indicate if you are using the MQTT example.  If you are, you can just change topics in the sketch to something different for the 2nd panel. Ie  Replace dsc/xxx/x/xxx with dsc2/xxx/xxxx.
```
const char* mqttClientName = "dscKeybusInterface";
const char* mqttPartitionTopic = "dsc/Get/Partition";  // Sends armed and alarm status per partition: dsc/Get/Partition1 ... dsc/Get/Partition8
const char* mqttPartitionMessageSuffix = "/Message";   // Sends partition status messages: dsc/Get/Partition1/Message ... dsc/Get/Partition8/Message
const char* mqttZoneTopic = "dsc/Get/Zone";            // Sends zone status per zone: dsc/Get/Zone1 ... dsc/Get/Zone64
const char* mqttFireTopic = "dsc/Get/Fire";            // Sends fire status per partition: dsc/Get/Fire1 ... dsc/Get/Fire8
const char* mqttPgmTopic = "dsc/Get/PGM";              // Sends PGM status per PGM: dsc/Get/PGM1 ... dsc/Get/PGM14
const char* mqttTroubleTopic = "dsc/Get/Trouble";      // Sends trouble status
const char* mqttStatusTopic = "dsc/Status";            // Sends online/offline status
const char* mqttBirthMessage = "online";
const char* mqttLwtMessage = "offline";
const char* mqttSubscribeTopic = "dsc/Set";            // Receives messages to write to the panel
```

sensormatic1985, 2021-04-12:
good afternoon

Yes sir, I am using the example you sent, then I will make the changes you say and do tests, I appreciate your help.

sensormatic1985, 2021-04-12:
Mr. Dilbert66 good night
I already carried out the steps that you told me, you can integrate the zones of the 2 alarm panel, but the armed and disarmed status does not work for me, do you have to change the name too? Can you help me again? Thank you. I'm using a google translator from Spanish to English, in case you don't understand me, please let me know

// MQTT topics - match to Home Assistant's configuration.yaml
const char* mqttClientName = "dscKeybusInterface";
const char* mqttPartitionTopic = "dsc2/Get/partition";  // Sends armed and alarm status per partition: dsc/Get/Partition1 ... dsc/Get/Partition8
const char* mqttZoneTopic = "dsc2/Get/Zone";            // Sends zone status per zone: dsc/Get/Zone1 ... dsc/Get/Zone64
const char* mqttFireTopic = "dsc2/Get/Fire";            // Sends fire status per partition: dsc/Get/Fire1 ... dsc/Get/Fire8
const char* mqttTroubleTopic = "dsc2/Get/Trouble";      // Sends trouble status
const char* mqttStatusTopic = "dsc2/Status";
const char* mqttBirthMessage = "online";
const char* mqttLwtMessage = "offline";
const char* mqttSubscribeTopic = "dsc2/Set";

Dilbert66, 2021-04-13:
Change the mqtt client name as well perhaps to dsc2KeyBusInterface or whatever you prefer.

sensormatic1985, 2021-04-13:
I already changed the name but it still does not update me the status of the alarm panel, I send you the photos of my configuration .yalm suddenly I am doing something wrong
**alarm # 1**
alarm_control_panel:
  - platform: mqtt
    name: "Security System Partition 1"
    state_topic: "dsc/Get/Partition1"
    availability_topic: "dsc/Status"
    command_topic: "dsc/Set"
    payload_disarm: "1D"
    payload_arm_home: "1S"
    payload_arm_away: "1A"
  - platform: mqtt
    name: "Security System Partition 2"
    state_topic: "dsc/Get/Partition2"
    availability_topic: "dsc/Status"
    command_topic: "dsc/Set"
    payload_disarm: "2D"
    payload_arm_home: "2S"
    payload_arm_away: "2A"

**alarm # 2**
platform: mqtt
    name: "dsc2KeybusInterface Partition 1"
    state_topic: "dsc2/Get/Partition1"
    availability_topic: "ds2c/Status"
    command_topic: "dsc2/Set"
    payload_disarm: "1D"
    payload_arm_home: "1S"
    payload_arm_away: "1A"

Dilbert66, 2021-04-13:
this one is incorrect:

availability_topic: "ds2c/Status"

It should be dsc2/Status

Otherwise, I do not see anything else wrong at this point.

sensormatic1985, 2021-04-13:
Hi friend
Too bad I already corrected the error but I can't get alarm # 2 to show me the states of: armed and disarmed

sensormatic1985, 2021-04-13:
this is the last configuration I just made in home assistant configuration.yaml
**alarm_control_panel:**
  - platform: mqtt
    name: "Security System Partition 1"
    state_topic: "dsc/Get/Partition1"
    availability_topic: "dsc/Status"
    command_topic: "dsc/Set"
    payload_disarm: "1D"
    payload_arm_home: "1S"
    payload_arm_away: "1A"
  - platform: mqtt
    name: "Security System Partition 2"
    state_topic: "dsc/Get/Partition2"
    availability_topic: "dsc/Status"
    command_topic: "dsc/Set"
    payload_disarm: "2D"
    payload_arm_home: "2S"
    payload_arm_away: "2A"
  **- platform: mqtt**
    name: "familiaRodriguez Partition 1"
    state_topic: "dsc2/Get/Partition1"
    availability_topic: "dsc2/Status"
    command_topic: "dsc2/Set"
    payload_disarm: "1D"
    payload_arm_home: "1S"
    payload_arm_away: "1A"

sensormatic1985, 2021-04-13:
**this is the photo of my home assistant**

https://drive.google.com/file/d/1ELFr90vL8-ShMFwgIqxJNRqJ-dloNDKR/view?usp=sharing

Dilbert66, 2021-04-13:
mqtt topics are case sensitive but you have "partition" with a lower case P.  
```
const char* mqttPartitionTopic = "dsc2/Get/partition"
```

You have two choices.  Either change your homeassistant  yaml to 
state_topic: "dsc2/Get/partition1" instead of state_topic: "dsc2/Get/Partition1" 

or change the sketch to make 

const char* mqttPartitionTopic = "dsc2/Get/partition" into
const char* mqttPartitionTopic = "dsc2/Get/Partition"

```
const char* mqttPartitionTopic = "dsc2/Get/partition"; // Sends armed and alarm status per partition: dsc/Get/Partition1 ...
```

sensormatic1985, 2021-04-13:
good afternoon,
I made the changes that you told me but I still do not read the status of alarm # 2
**Arduino configuration**
// MQTT topics - match to Home Assistant's configuration.yaml
const char* mqttClientName = "dsc2KeybusInterface";
const char* mqttPartitionTopic = "dsc2/Get/Partition
const char* mqttZoneTopic = "dsc2/Get/Zone"
const char* mqttFireTopic = "dsc2/Get/Fire"
const char* mqttTroubleTopic = "dsc2/Get/Trouble"
const char* mqttStatusTopic = "dsc2/Status"
const char* mqttBirthMessage = "online"
const char* mqttSubscribeTopic = "dsc2/Set"
unsigned long mqttPreviousTime;

**configuration
configuration home assistan**

alarm_control_panel:
  - platform: mqtt
    name: "Security System Partition 1"
    state_topic: "dsc/Get/Partition1"
    availability_topic: "dsc/Status"
    command_topic: "dsc/Set"
    payload_disarm: "1D"
    payload_arm_home: "1S"
    payload_arm_away: "1A"
  - platform: mqtt
    name: "Security System Partition 2"
    state_topic: "dsc/Get/Partition2"
    availability_topic: "dsc/Status"
    command_topic: "dsc/Set"
    payload_disarm: "2D"
    payload_arm_home: "2S"
    payload_arm_away: "2A"
  - platform: mqtt
    name: "Security2 System Partition 1"
    state_topic: "dsc2/Get/Partition1"
    availability_topic: "dsc2/Status"
    command_topic: "dsc2/Set"
    payload_disarm: "1D"
    payload_arm_home: "1S"
    payload_arm_away: "1A"

Dilbert66, 2021-04-13:
Unfortunately, at this point I have no other suggestions.  I just suggested the obvious but I do not see anything else that stands out.   You will need to debug your mqtt messages on your setup to see what's going on.  You can use this program to view the activity on your mqtt server;

http://mqtt-explorer.com/

sensormatic1985, 2021-04-13:
I really appreciate your help, I'll try the mqtt signal analyzer and I'll tell you anything

sensormatic1985, 2021-04-14:
hello Dilbert66
The last thing I did was restart the alarm # 2 arduino and it started working, thank you very much for your great help

Dilbert66, 2021-04-14:
Glad you got it working!

sensormatic1985, 2021-04-14:
By chance you know how I can enable the arming and disarming from home assistant, I made the connection as the sketch says, connecting the ntp but it does not arm or disarm me

closed 2021-04-16 by sensormatic1985