taligentx/dscKeybusInterface #217: Decoding of set date/time

sbrown4, 2021-04-13:
A couple of samples from setting the panel's date/time from an IT100.
```
           YYY1YYY2     MMMM    DDDDD HHH1HHH2 MMM1MMM2   CRC
11111111 1 00100001 00000100 00001101 00001000 01001001 10000011 [Module/0xD0] Unknown data (Set date/time)
              2   1        4       13        8    4   9    8   3

           YYY1YYY2     MMMM    DDDDD HHH1HHH2 MMM1MMM2   CRC
11111111 1 00100001 00000100 00001101 00010011 01000000 10000101 [Module/0xD0] Unknown data (Set date/time)
              2   1        4       13    1   3    4   0    8   5

```

taligentx, 2021-05-24:
Hi @sbrown4 - thank you!  This will be a cleaner way to set the time from the library, added to the TODO list to see if panels will accept 0xD0 command without the IT-100 attached.

Dilbert66, 2022-05-12:
What was the previous 05/0a request cmd? The it-100 must have triggered a cmd first in order to have a 0xD0 command response?

sbrown4, 2022-05-12:
On Wed, 2022-05-11 at 18:30 -0700, Alain Turbide wrote:
> What was the previous 05/0a request cmd? The it-100 must have triggered
> a cmd first in order to have a 0xD0 command response?
> —
Here is the dump around the 0xD0

  763.82: 11111111 1 11111111 11111111 11111111 11111101 11111111 11111111 11111111 11111111 [Module/0x05] Wireless notification 
  763.89: 11010000 0 11111111 11111111 11111111 11111111 11111111 11111111 [0xD0] [CRC Error]
  763.89: 11111111 1 00100010 00000101 00001100 00000100 00000000 00110111 [Module/0xD0] Unknown data
  763.98: 11100110 0 00011010 01000000 00000000 00000000 00000001 00000001 00000000 00000000 01000010 [0xE6.1A] Partitions in alarm: none

Dilbert66, 2022-05-12:
Ok, I see, the it100 sets bit 1 in byte 5 to request to do an update.   Thank you. That is what I needed.