taligentx/dscKeybusInterface #225: SendsimpleMessage works sendMessageWithInlineKeyboard or sendmessage not working

hcoiro, 2021-04-26:
Hi, I am having the following problem, at some point of my code I wanto to send a message:
I tryed with these three options: where the varibles message, estado or devolver ar string.
The only one that work at that point of the code is the 1)  "sendSimpleMessage", the other not working.
1) telegramBot.sendMessage(telegramUserID, message, "HTML")
2) telegramBot.sendMessageWithInlineKeyboard(telegramUserID, estado , "HTML", keyboardJson,ultimo_message_id);
3) telegramBot.sendSimpleMessage(telegramUserID, devolver, "HTML");
if I use them in other part of the code, they work fine.
Does anyone have an idea of what could be going on?

taligentx, 2022-03-08:
Hi @hcoiro - did you find a solution for this issue?  It would be useful to see your code but is this a bug with the example sketch or a bug when adding an additional message?

hcoiro, 2022-03-09:
I finaly use  the command telegramBot.sendSimpleMessage(telegramUserID, devolver, "HTML"); 
And it was working. There is a time that I have not touch the code, need to review it

taligentx, 2022-03-15:
Sure, if you run into this problem again, feel free to re-open this issue.

closed 2022-03-15 by taligentx