taligentx/dscKeybusInterface #241: Communication Broken with Mosquitto broker 6.0.1

ChrisPotDev, 2021-05-25:
Not sure if there is a bug listed for this as yet.
MQTT communication is broken with Mosquitto broker 6.0.1

From other issues ive seen it relates to using / in the clientname

javierferwolf, 2021-12-16:
@ChrisPotDev  hi , it's been a while since your message, have you been able to find a solution?

kghandi, 2021-12-16:
I for one am using the current version of dsckeybusinterface and it is working fine with Mosquito 6.0.1 in Home Assistant. The only time I had issues was when I had not included the correct username/password in dsckeybusinterface for my mqtt broker.

javierferwolf, 2021-12-16:
@kghandi thank you for your answer  question, what is the data that you filled in the configuration?
Brocker: localhost or IP computer (raspberrypi)?

[image]

kghandi, 2021-12-17:
HomeAssistant running on a VirtualBox VM on a Mac.
Broker: core-mosquitto
Port: 1833

Broker config:
```
logins: []
customize:
  active: false
  folder: mosquitto
certfile: fullchain.pem
keyfile: privkey.pem
require_certificate: false
```

I use dscKeybusInterface example HomeAssistant-MQTT in which:
mqttServer is specified by IP address.
mqttUsername & mqttPassword are the same as those I use to log into home assistant.

taligentx, 2022-03-03:
Shifting this over to Discussions as it appears to be an issue with the HomeAssistant mosquitto component.  One option is to check running the [current release of Mosquitto](https://mosquitto.org/download/) as a separate service (current version: 2.0.14).  I've tested with 2.0.11 and haven't run across this issue.