taligentx/dscKeybusInterface #246: Illegal instruction on dscWrite

Elektrik1, 2021-06-22:
Hey there, I have decided to merge some new features for Web example, and found out that current stable (and dev) branches Web example does not work on ESP8266 (NodeMCU) with latest ESP8266 Core 3.0.0. The restart occur on attempt to perform http query (open in browser), failing at:

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (0):
epc1=0x4021a83c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffe600 end: 3fffffb0 offset: 0190
3fffe790: bdf47752 00007752 00000000 40100db8
3fffe7a0: 00000000 00000001 3fff0488 401004e0
3fffe7b0: 00000000 00000000 00000000 ffffffff
3fffe7c0: 00000005 00000001 00000000 40100f28
3fffe7d0: c0037035 00000020 5efa3ba9 00000000
3fffe7e0: 00000001 00000001 00000031 00000000
3fffe7f0: 00000000 00000000 4024c9b9 00000022
3fffe800: 3fffc200 40100e5c 3fffc258 4000050c
3fffe810: 4000437d 00000030 0000001a ffffffff
3fffe820: 60000200 00000008 ffffffff 80000000
3fffe830: 20000000 3fff234c 80000000 20306080
3fffe840: 00000000 00000005 00000000 3fff2350
3fffe850: 00000080 00306080 00000100 00000030
3fffe860: 3fffea40 3fffea60 00000010 0000006c
3fffe870: 00000240 3fffe970 000000ff 3fffe960
3fffe880: 3fff24ec 00000001 3fff24ec 00000030
3fffe890: 401035d2 3ffeaffc 00000001 00000000
3fffe8a0: 3ffeaffc 4010373b 3ffeaffc 00000001
3fffe8b0: 00007fff 0048d24b 40103c2e 00000100
3fffe8c0: 3ffeaffc 7fffffff 00000000 00000001
3fffe8d0: 00000001 00004a88 4000050c 3fffc278
3fffe8e0: 3ffeaffc 3fffc200 00000022 0048d24b
3fffe8f0: 3ffeae28 2c9f0300 4000050c 3fffc278
3fffe900: 00000000 400042db 00000022 90928724
3fffe910: 40004b31 3fff22d0 00000100 00306000
3fffe920: 401070f8 7bf36ae2 05050202 401004eb
3fffe930: 40100b42 00000001 3fff22d0 00306000
3fffe940: 40100f3c 00000100 00000000 40216048
3fffe950: c0037015 00000020 3fff22d0 402160d5
3fffe960: 63736403 74685f05 5f047074 05706374
3fffe970: 3fffc200 40100b88 00000020 40100bd0
3fffe980: 3fffc200 40100e5c 3fffc230 4000050c
3fffe990: 4021bdf4 00000030 0000001b ffffffff
3fffe9a0: 4021c1ff 00002000 00000100 00000001
3fffe9b0: 0000001b 3fff2f84 00000000 4021bdcc
3fffe9c0: 00000130 00000005 00000000 3fff2f84
3fffe9d0: 0000001b 00000100 3fff13ec 00000030
3fffe9e0: 00000000 00000005 00000000 3fff13ec
3fffe9f0: 00000000 3fff1f6c 00000100 00000030
3fffea00: 00000000 00000000 00000000 00000000
3fffea10: 00000000 00000000 00000000 00000000
3fffea20: 00000000 00000000 00000000 00000000
3fffea30: 00000000 00000000 00000000 00000000
3fffea40: 00000000 00000000 00000000 00000000
3fffea50: 00000000 00000000 00000000 00000000
3fffea60: 00000100 76726573 73656369 6e645f07
3fffea70: 64732d73 64755f04 6f6c0570 00000100
3fffea80: 00000100 3fff22d0 00306000 4021a8c5
3fffea90: 3fff1f6c 00000000 3fff13ec 40218eb8
3fffeaa0: 00000000 00000000 00000000 4021e84c
3fffeab0: 3fff22d0 00000100 3fff22bc 00306000
3fffeac0: 3fff1d6c 00000002 00000000 ffffffff
3fffead0: 40100f3c 00000001 00000000 00006000
3fffeae0: 00000100 3fff13ec 00000003 4021c14d
3fffeaf0: 00000000 00000000 00000000 00000080
3fffeb00: 3fff1daa 00000000 00000ec6 00000000
3fffeb10: 00000ec5 00000000 4021bdcc 3fff1d6c
3fffeb20: 00000000 00000100 00000018 ffffffff
[... 484 lines cut ...]
0x4020d4e4: dscKeybusInterface::loop() at C:\Dropbox\Arduino\Sketches\libraries\dscKeybusInterface\src\dscKeybusInterface.cpp line 135
0x40207d25: loop() at C:\Dropbox\Arduino\Sketches\Projects\DSC1864\VirtualKeypad-Web/VirtualKeypad-Web.ino line 174
0x4021e862: spiffs_phys_rd(spiffs*, u8_t, spiffs_file, u32_t, u32_t, u8_t*) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\spiffs\spiffs_cache.cpp line 172
0x4021a84a: _stopPWM(uint8_t) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_waveform_pwm.cpp line 264
0x4021c19d: spiffs_obj_lu_find_entry_visitor(spiffs*, spiffs_block_ix, int, u8_t, spiffs_obj_id, spiffs_visitor_f, void const*, void*, spiffs_block_ix*, int*) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\spiffs\spiffs_nucleus.cpp line 172
0x40100dc0: __digitalWrite(uint8_t, uint8_t) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_wiring_digital.cpp line 87
0x40100adc: ets_post(uint8, ETSSignal, ETSParam) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 181
0x40100adc: ets_post(uint8, ETSSignal, ETSParam) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 181
0x40100e89: interrupt_handler(void*, void*) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_wiring_digital.cpp line 141
0x40100adc: ets_post(uint8, ETSSignal, ETSParam) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 181
0x4021a84a: _stopPWM(uint8_t) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_waveform_pwm.cpp line 264
0x40100dc0: __digitalWrite(uint8_t, uint8_t) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_wiring_digital.cpp line 87
0x40218138: loop_wrapper() at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 196
0x40218138: loop_wrapper() at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 196
0x40216474: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/HardwareSerial.h line 193
0x40216964: Print::write(char const*) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Print.h line 59
0x40216b10: Print::println() at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Print.h line 57
0x40216b34: Print::println(__FlashStringHelper const*) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\Print.cpp line 238
0x402015dc: setup() at C:\Dropbox\Arduino\Sketches\Projects\DSC1864\VirtualKeypad-Web/VirtualKeypad-Web.ino line 159
0x402217dc: std::_Function_handler ::_M_invoke(std::_Any_data const&, AsyncWebSocket*&&, AsyncWebSocketClient*&&, AwsEventType&&, void*&&, unsigned char*&&, unsigned int&&) at c:\arduino\application\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.0-newlib4.0.0-gnu23-48f7b08\xtensa-lx106-elf\include\c++\10.2.0\bits/std_function.h line 289
0x40100adc: ets_post(uint8, ETSSignal, ETSParam) at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 181
0x4021814c: loop_wrapper() at C:\Arduino\Application\portable\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266\core_esp8266_main.cpp line 201
```

I am not sure what is the problem here, is it ESP thing (maybe digitalWrite is not supported in ISR routine anymore?)...

Dilbert66, 2021-06-22:
Try doing debug prints before the digitalwrite to see what it is that you are attempting to write and to what port.  I suspect it's the input data that is at fault.

Elektrik1, 2021-06-22:
The digitalWrite is internal in dscKeybusInterface::dscClockInterrupt() function: https://github.com/taligentx/dscKeybusInterface/blob/3cf3653de7f5f346bd99c05af5e9c5f56b43b339/src/dscKeybusInterface.cpp#L690

Dilbert66, 2021-06-22:
Yes, but wanted to see if the dscWritePin value is correct.  LOW is a constant which is fine.

Dilbert66, 2021-06-22:
ok, I suppose it could be a bug in the esp library. I've seen that a few times on updates.  You will need to check for other bug reports from other users/projects

Elektrik1, 2021-06-22:
Ok, so my theory is correct - downgraded to 2.7.4 - everything works fine. So it is particularly 3.0.0 update that causes this, perhaps some logic has changed, changelog need to be checked.

Dilbert66, 2021-06-22:
Odd, I just tried 3.0.0 with my own similar library that works on Vista20's and I also use calls to digitalwrite from ISR routine and see no issues. I do see the warnings about the deprecated ICACHE_RAM_ATTR  to use instead IRAM_ATTR which is fine. I will add checks later to use the right attribute for ISR functions.

Elektrik1, 2021-06-22:
I do not use digitalWrite in ISR on my own - the dsc library does it, and exception points to that line. Try this example (https://github.com/taligentx/dscKeybusInterface/blob/3cf3653de7f5f346bd99c05af5e9c5f56b43b339/examples/esp8266/VirtualKeypad-Web/VirtualKeypad-Web.ino), make sure DSC pins are harware connected, ant try to open http interface. I am not sure if this is NodeMCU specific issue, do not have spare Wemos atm, but since they are both 12E ESP's, I assume the issue will persist.

Dilbert66, 2021-06-22:
Not sure when I will get a chance to try it.  Also, a major revision change always introduces many breaking changes so I won't rush to update my code to it just yet until at least a minor version comes out with outstanding bugs fixed.

Dilbert66, 2021-06-22:
Ok, I set some time aside and tried your example as well as the Homeassistant-mqtt example.  The web keypad crashes as soon as you connect (as you indicated), except mine crashes at the clockinterrupt.  When I tried the homeassistant-mqtt example, it worked perfectly with no resets.  This tells me that the dsc library is not the problem in this case since the same library was used with both.  I would suspect either the SPIFF library or another custom component used for the web keypad.   I lean more towards the SPIFF handling since it was deprecated with 3.0.0 to something called LittleFS.   I don't know, just guessing here.
The way it's crashing is as if the heap gets corrupted and affecting return address stack values

hcoiro, 2021-06-22:
Just one comment. Some time ago I was trying to make some modification on the web interface like mixing with  telegram notification. The problem  I had was with  <ESPAsyncWebServer.h>  and  <ESPAsyncTCP.h>. The problem I have is when trying to connect to the web keyboard, the 12E ESP reset. I was told by someone here that it is some capacity problem of the 12E ESP to manage async comunication with other  stuff doing by the ESP,  May be this is related also to the capabilty of handle async comunication of 12E with the new core.

Elektrik1, 2021-06-22:
Thank you guys, for narrowing the issue to Web only. Since I am the creator of original Web example, I will port it to LittleFS and check whether it fixes the problem, if not, will continue to investigate what of used components causes this. I am closing the issue, and will pull request once port will be done and checked.

closed 2021-06-22 by Elektrik1

Dilbert66, 2021-06-22:
The interesting part is that the spiffs do work fine when the dsc is not connected. It is possible like hcoiro noted that this is a resource issue as well since the new 3.0.0 might use more ram/iram. I've seen this too with other versions.

Dilbert66, 2021-07-04:
Looks like it might be an issue with the 3.0.0 library specifically  the call to calls _stopPWM(). See here:
https://github.com/esp8266/Arduino/issues/8043