taligentx/dscKeybusInterface #255: Checks the access code used to arm or disarm

SebastianJoan, 2021-11-16:
when i try to Checks the access code used to arm or disarm it only show access code used to arm the alarm but it didn´t show the user code used to disarm the alarm

UltimatPronin, 2022-02-10:
I confirm the problem

taligentx, 2022-03-03:
Hi @SebastianJoan @UltimatPronin - the library keeps track of access codes for each partition both for arming and disarming, at any time you can get the access code used on the partition from `dsc.accessCode[partition]`.

However, `dsc.accessCodeChanged[partition]` will only be true if the access code used to arm/disarm the partition is different than the last arm/disarm code.  

If you would like to get the access code used to disarm a partition even if it's the same code used to arm the partition, you can check `dsc.accessCode[partition]` in your sketch, there's no need to check `dsc.accessCodeChanged[partition]`.

Example for the `Status` sketch:
```
Before:
...

        else {
          Serial.print(F("Partition "));
          Serial.print(partition + 1);
          Serial.println(F(": Disarmed"));
        }

After:
...
        else {
          Serial.print(F("Partition "));
          Serial.print(partition + 1);
          Serial.println(F(": Disarmed"));
          Serial.print(F("Partition "));
          Serial.print(partition + 1);
          switch (dsc.accessCode[partition]) {
            case 40: Serial.print(F(": Master")); break;
            default: Serial.print(F(": Access")); break;
          }
          Serial.print(F(" code "));
          Serial.println(dsc.accessCode[partition]);
        }
```
This should be repeated for each section of the sketch that shows when the partition is disarmed.