taligentx/dscKeybusInterface #258: Resistor Selection

AKMedi9, 2021-12-17:
The documentation suggests using a 33k resistor for the clock and read pins for esp8266/esp32 and a 15k resistor for arduino boards. The PCB schematic lists 15k resistor for the NodeMCU board. Is this resistor value critical? Im building a board with an esp32 NodeMCU and I was going to use a 33k resistor here but I just wanted to double check.

[image]
[image]

kghandi, 2021-12-17:
DSC operates at 12.6V, Arduino at 5V, and ESP at 3.3V. 
The resistors act as a voltage divider. 
Vout = Vin R2 / (R1+R2).  
So with 10k and 15k we get:
Vout = 12.6 * 10 / 25 = 5.06 good for Arduino
And with 10k and 33k:
Vout = 12.6 * 10 / 43 = 2.93 good for ESP

AKMedi9, 2021-12-17:
Thanks that answers my question.