taligentx/dscKeybusInterface #259: Pushbullet connection failed

aritonigr, 2021-12-22:
Hello,
I tried to use pushbullet example on an ESP8266 d1 mini unfortunately with no luck.  What I have noted is that in the code there is this line:
 // HTTPS root certificate for api.pushbullet.com: GlobalSign Root CA - R2, expires 2021.12.15
Is there an update on this certificate? Can this cause the no connection issue?

I use the email example without any issue.

Dilbert66, 2021-12-22:
it's possible. Here's an easy way to know for sure. 
Replace line:
```
ipClient.setTrustAnchors(&pushbulletCert);
```
with this line instead:
```
ipClient.setInsecure();
```
This disables the certificate check.

taligentx, 2021-12-23:
Hi @aritonigr - thanks for the heads-up, I've updated the Pushbullet examples in the `develop` branch with the current root certificate for api.pushbullet.com.  Let me know if this resolves the issue for you.

aritonigr, 2022-01-08:
Hi @taligentx, yes now it works without any issue for me! Thank you!