taligentx/dscKeybusInterface #263: [Question] Make device supervised by panel

Alviunta, 2021-12-29:
Great job!
Does anyone know if it is possible to get the panel to monitor the arduino connection?
As far as I understand, the modules are supervised by the panels through the code 0x11 (module supervision query) to which each module places a series of bits at a specific time to indicate presence and connection, so if a module is disconnected it is disconnected you can see this directly on the keypad.
As the panel can support up to 8 keypads, perhaps a slot could be used to monitor the aruino connection and ensure that it is working at all times. I tried to use panelData [0] == 0x11 and a timer to write 00 in the KeyPad 2 slot (slot 8 was already in use) but it didn't work out as expected and the panel showed a module error, I don't know if any other is necessary type of interaction with the panel.

I have a series of modules that can be of help to try to decipher more information and achieve what I am proposing, as soon as I have more time I will get to work on it:
- [Escort5580](https://www.dsc.com/index.php?n=products&o=view&id=18) telephone interface
- I[T100 bi-directional RS232](https://www.dsc.com/alarm-security-products/IT-100%20-%20PowerSeries%20Integration%20Module/22) module (used to make integrations with other systems)
- IP communicator (I don't know the model)

On the other hand, I have seen third-party IP communicators that connect to the KeyBus and wrap themselves in the [printer](https://www.dsc.com/alarm-security-products/PC5400%20-%20Printer%20Interface%20Module/44) slot (I suppose to explicitly obtain all the information or something like that). Does anyone have information on this? Perhaps this could make things easier for the future. If I get a module that works like this, I will gladly share what I find.

Is there any documentation on how the KeyBus protocol works? I looked in many places but I could find little, if someone can give me help I would be grateful.

I apologize for my bad English (Google Translate did most of the work) and I hope someone can help me!

taligentx, 2021-12-30:
>Does anyone know if it is possible to get the panel to monitor the arduino connection?

Hi @Alviunta - this should be possible, I've considered it previously but it's been lower priority as I haven't seen requests for this feature (until yours) and it would use additional resources on the microcontrollers (mostly an issue for Arduino).  As you mentioned it would be a matter of deciding which type of module to emulate and then decode the Keybus signaling for it.  Emulating a keypad should be straightforward since the 0x11 command is decoded.

>I have a series of modules that can be of help to try to decipher more information and achieve what I am proposing, as soon as I have more time I will get to work on it

This would be great!  Getting more of the Keybus decoded is always a goal for this library.

>On the other hand, I have seen third-party IP communicators that connect to the KeyBus and wrap themselves in the printer slot

I haven't seen data for the PC5400 printer module, logs from the `KeybusReader` sketch would be handy if anyone has the module.  Ideally, a log with the module disconnected, then another log with the module connected.

>Is there any documentation on how the KeyBus protocol works?

[This thread on AVRFreaks](https://www.avrfreaks.net/forum/dsc-keybus-protocol?page=all) is where I first came across folks decoding the Keybus protocol.  The [dscKeybusPrintData.cpp](https://github.com/taligentx/dscKeybusInterface/blob/develop/src/dscKeybusPrintData.cpp) file documents the parts of the protocol that are currently decoded.

>I apologize for my bad English (Google Translate did most of the work) and I hope someone can help me!

No apologies needed, your post was very clear!

Thanks,
Nikhil

Dilbert66, 2021-12-30:
Supervision is definitively doable but in my opinion is not really a useful addition since when you think about it, the main idea of this library is to monitor and control the alarm system and not the reverse so having the alarm system monitor the keypad emulation seems unecessary. Of course, I could be missing a scenario.  

Anyhow, I did implement supervision with my expander board emulation version of this library.  The firmware fully allowed the emulated expander boards to be supervised. I did not supervise the keypad since i found it unecessary but it would simply have needed the addition of setting a predefined unused slot as you noted.   You need to be able to respond to the cmd 11 directly in the ISR routine to send back the slot info.   Here's a link to the branch where you can view the implementation.   In my own version , I don't bother activating the supervision as I found it to be just very annoying since during testing/development needed the restart of the chip and the panel would always scream that a device is missing at that time!  

Nikhil is correct in noting that it will impact resource use on lower ram devices so would need to be a #define to limit this function to more powerful devices.   I am rewriting the expander version to be more modular and fit better within the new structure of Nikhil's library.  

Here's the link:
https://github.com/taligentx/dscKeybusInterface/blob/expander/src/dscKeybusInterface.cpp