taligentx/dscKeybusInterface #264: Home Assistant - MQTT

aritonigr, 2022-01-08:
I admit that I am not familiar with MQTT and Home Assistant and I need some help:
In the following code:
...
const char* mqttServer = "192.168.1.200";    _**The domain or IP where the MQTT is installed (namely the IP of Home Assistant)?**_
const int   mqttPort = 1883;    // MQTT server port
const char* mqttUsername = "user";  // Optional, leave blank if not required **_Username of HomeAssistant or MQTT?_**
const char* mqttPassword = "pass";  // Optional, leave blank if not required _**Password of Home Assistant or MQTT?**_
...

Is there something I have to change on yaml code (except of course the configuration of zones, partitions etc.)?

aritonigr, 2022-01-11:
Finally I managed to make it work simply by putting the... correct ssid... Now, after around 20hours functioning correctly this message appears on Arduino Monitor:
MQTT....connection error: 192.168.1.200
MQTT disconnected, failed to reconnect.
What is wrong here?

brucehatherell, 2022-01-26:
I had an issue where my router was not giving out a new DHCP lease due to an arp issue - I found a lot more stability with static IP configured - may help

daexpert, 2022-02-04:
I want to set up a static ip for arduino uno, not dhcp, can some one help me with the synthax for the ino file?

lubster85, 2022-02-04:
I'm having a similar issue around this. I added something force the wifi to reconnect and then try the MQTT connection again. This works reliably, but I am noticing MQTT is still losing connection every few minutes.

Is there anything special that should be set regarding MQTT?

daexpert, 2022-02-05:
I think it's rather a network communication issue between the arduino/esp and the router/mqtt server, from a pc/laptop open a command prompt and execute a ping command with the -t parameter into arduino/esp and see if there is packet loss.

daexpert, 2022-02-05:
I've managed to modify the sketch for arduino uno to setup static ip instead of dhcp:

-added these lines:
byte ip[] = { 192, 168, 1, 7 };
byte subnet[] = { 255, 255, 255, 0 };
byte gateway[] = { 192, 168, 1, 1 };

- modified these ones:
EthernetClient client;
PubSubClient mqtt(mqttServer, mqttPort, client);

Ethernet.begin(mac, ip, subnet, gateway);
Serial.print(F("connected ip: "));
Serial.println(Ethernet.localIP());
Serial.print(F("subnet mask: "));
Serial.println(Ethernet.subnetMask());
Serial.print(F("gateway: "));
Serial.println(Ethernet.gatewayIP());

lubster85, 2022-02-07:
> I think it's rather a network communication issue between the arduino/esp and the router/mqtt server, from a pc/laptop open a command prompt and execute a ping command with the -t parameter into arduino/esp and see if there is packet loss.

So I'm seeing times when I ping and it's OK
--- 192.168.XX.XX ping statistics ---
19 packets transmitted, 19 received, 0% packet loss, time 18284ms
rtt min/avg/max/mdev = 1.544/30.925/227.866/64.967 ms

Other times I'm seeing issues, like this:
--- 192.168.XX.XX ping statistics ---
34 packets transmitted, 14 received, 58.8235% packet loss, time 34866ms
rtt min/avg/max/mdev = 63.388/964.241/3487.802/926.397 ms, pipe 4

I also looked at the router + Mosquitto logs. Again I see peroids of nothing happening.. then I see this repeat over and over.
Mon Feb  7 14:25:18 2022 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED xx:xx:xx:xx:xx:xx
Mon Feb  7 14:25:18 2022 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: disassociated
Mon Feb  7 14:25:19 2022 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Mon Feb  7 14:25:46 2022 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: authenticated
Mon Feb  7 14:25:47 2022 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: associated (aid 5)
Mon Feb  7 14:25:47 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED xx:xx:xx:xx:xx:xx
Mon Feb  7 14:25:47 2022 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Mon Feb  7 14:25:47 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED xx:xx:xx:xx:xx:xx
Mon Feb  7 14:25:47 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) xx:xx:xx:xx:xx:xx
Mon Feb  7 14:25:47 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.0.8 xx:xx:xx:xx:xx:xx
Mon Feb  7 14:25:47 2022 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.0.8 xx:xx:xx:xx:xx:xx
Mon Feb  7 14:25:47 2022 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.0.8 xx:xx:xx:xx:xx:xx ESP-1D9D75
Mon Feb  7 14:25:57 2022 daemon.notice mosquitto[1768]: 1644236757: Client dscKeybusInterface disconnected due to malformed packet.
Mon Feb  7 14:25:57 2022 daemon.notice mosquitto[1768]: 1644236757: New connection from 192.168.xx.xx:57572 on port 1883.
Mon Feb  7 14:25:57 2022 daemon.notice mosquitto[1768]: 1644236757: New client connected from 192.168.xx.xx:57572 as dscKeybusInterface (p2, c1, k90, u'DSC').

daexpert, 2022-02-07:
It's clear a comunication issue between the device and router. You have packet loss and the device is reauthenticating (DHCPOFFER, DHCPREQUEST, DHCPACK) If the device is an esp try put it more closer to the router and if it's an arduino uno with ethernet shield replace the network cable between the router and the arduino (my arduino is not working if the cable length to the router is more than 10 meters)

taligentx, 2022-02-07:
Hi @lubster85,

>Mon Feb 7 14:25:19 2022 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

Are you running OpenWRT on the router?  Looks like there's an OpenWRT setting that disconnects devices that don't sent ACK packets frequently (which would apply to this use case):

https://openwrt.org/faq/deauthenticated_due_to_inactivity

>Issue: Wifi problems with Android / iOS devices:
>
>constant disconnects
>logfile shows deauthenticated due to inactivity
>
>Solution: [/etc/config/wireless](https://openwrt.org/docs/guide-user/network/wifi/basic) → @ config wifi-iface → add option disassoc_low_ack '0'

You could also add something in the sketch that pings the router at a certain interval to keep the connection alive, but this looks like a general networking issue.

aritonigr, 2022-02-07:
> It's clear a comunication issue between the device and router. You have packet loss and the device is reauthenticating (DHCPOFFER, DHCPREQUEST, DHCPACK) If the device is an esp try put it more [...]

Hi, in my case the router is next to esp8266 but still communication issues after some time.

lubster85, 2022-02-07:
> Hi @lubster85,
> 
> > Mon Feb 7 14:25:19 2022 daemon.info hostapd: wlan1: STA xx:xx:xx:xx:xx:xx IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
> 
> Are you running OpenWRT on the router? Looks like there's an OpenWRT setting that disconnects devices that don't send ACK packets frequently (which would apply to this use case):
> 
> https://openwrt.org/faq/deauthenticated_due_to_inactivity
> 
> > Issue: Wifi problems with Android / iOS devices:
> > constant disconnects
> > logfile shows deauthenticated due to inactivity
> > Solution: [/etc/config/wireless](https://openwrt.org/docs/guide-user/network/wifi/basic) → @ config wifi-iface → add option disassoc_low_ack '0'
> 
> You could also add something in the sketch that pings the router at a certain interval to keep the connection alive, but this looks like a general networking issue.

Thank you for the suggestion. Yes, I am using OpenWRT. The ESP device is a NodeMCU device and it is a fair bit away from the router (but it should not be out of range). I have that setting in the config already on the 2.4GHZ network, but II'll try pinging the router ever-so-often and see if that helps.

Dilbert66, 2022-02-07:
I know this has been mentioned before but power issues will also cause problems with wifi.  Are yo powering via usb? Try a different adapter just as an experiment as well.

lubster85, 2022-02-07:
> I know this has been mentioned before but power issues will also cause problems with wifi. Are you powering via usb? Try a different adapter just as an experiment as well. Normally you would see this as reboots but you didnt mention those so it could be something else. I have also seen a bad esp that would disconnect and never reconnect until a reboot. A new esp solved the issue.

Yes, I am at the moment. I'm using a 5V 1amp (iPhone charger) power supply. I'll try a 2amp supply and see how that goes.

lubster85, 2022-02-09:
- [x] The power supply does not seem to be the issue.

Yesterday I tried adding the following to my sketch. The idea was to pink the router every 5 seconds.
I simply placed it into my loop section. It appeared to be working flawlessly for many minutes but later on I saw the same behaviour.

I'm going to reflash it tonight and make sure no other changes are in the sketch (I'm using the HomeAssistant-MQTT example as my base) and will retest. To ping, I've added the "Pinger" library and I've placed it directly in the loop section just above mqttHandle();

```
void loop() {
  unsigned long pingCurrentTime = millis();

  if (pingCurrentTime - pingPreviousTime > 5000) {

    pingResult = pinger.Ping(routerIP);
    if (pingResult >= 0) {
      Serial.print("SUCCESS! RTT = ");
      Serial.print(pingResult);
      Serial.println(" ms");
    } else {
      Serial.print("FAILED! Error code: ");
      Serial.println(pingResult);
    }
    pingPreviousTime = pingCurrentTime;
  }

  mqttHandle();

  dsc.loop();.......
```

lubster85, 2022-02-09:
@daexpert I'll give that a shot to test it, although unfortunately it has to be positioned where it is if i am to use it. I've had it in place for ages without issue and the router is the same device too so I'm really lost for a good explanation.

For completeness sake, here are some logs from the ESP device and router after flashing the updated sketch.
I've shortened the log a little, but assume that where you see dots, there were successful pings. Perhaps someone spots something useful.

```
19:23:53.659 -> WiFi.........................................connected: 192.168.X.XX
19:24:14.883 -> MQTT....connected: 192.168.X.XY
19:24:20.292 -> DSC Keybus Interface is online.
19:24:20.292 -> SUCCESS! RTT = 1 ms
19:24:25.319 -> SUCCESS! RTT = 1 ms
19:24:30.295 -> SUCCESS! RTT = 1 ms
19:24:35.308 -> SUCCESS! RTT = 1 ms
19:24:40.298 -> SUCCESS! RTT = 1 ms
19:24:45.318 -> SUCCESS! RTT = 1 ms
19:24:50.324 -> SUCCESS! RTT = 1 ms
19:24:55.307 -> SUCCESS! RTT = 1 ms
19:24:59.815 -> MQTT....connection error: 192.168.X.XY
19:25:04.968 -> MQTT disconnected, failed to reconnect.
19:25:05.016 -> SUCCESS! RTT = 1 ms
19:25:05.016 -> MQTT....connection error: 192.168.X.XY
19:25:10.148 -> MQTT disconnected, failed to reconnect.
19:25:10.148 -> SUCCESS! RTT = 1 ms
19:25:10.148 -> MQTT....connection error: 192.168.X.XY
19:25:15.313 -> MQTT disconnected, failed to reconnect.
19:25:15.313 -> SUCCESS! RTT = 1 ms
19:25:15.313 -> MQTT....connection error: 192.168.X.XY
19:25:20.471 -> MQTT disconnected, failed to reconnect.
19:25:20.471 -> SUCCESS! RTT = 1 ms
19:25:20.471 -> MQTT....connected: 192.168.X.XY
19:25:23.453 -> MQTT disconnected, successfully reconnected.
19:25:25.467 -> SUCCESS! RTT = 1 ms
.
.
.
19:27:17.266 -> SUCCESS! RTT = 1 ms
19:27:17.266 -> MQTT....connected: 192.168.X.XY
19:27:17.266 -> MQTT disconnected, successfully reconnected.
.
.
.
19:27:47.273 -> SUCCESS! RTT = 1 ms
19:27:52.265 -> SUCCESS! RTT = 1 ms
19:27:57.274 -> SUCCESS! RTT = 1 ms
```

```
Wed Feb  9 19:24:13 2022 daemon.info hostapd: wlan1: STA aa:bb:aa:bb:aa:bb IEEE 802.11: authenticated
Wed Feb  9 19:24:13 2022 daemon.info hostapd: wlan1: STA aa:bb:aa:bb:aa:bb IEEE 802.11: associated (aid 8)
Wed Feb  9 19:24:14 2022 daemon.notice hostapd: wlan1: AP-STA-CONNECTED aa:bb:aa:bb:aa:bb
Wed Feb  9 19:24:14 2022 daemon.info hostapd: wlan1: STA aa:bb:aa:bb:aa:bb WPA: pairwise key handshake completed (RSN)
Wed Feb  9 19:24:14 2022 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED aa:bb:aa:bb:aa:bb
Wed Feb  9 19:24:14 2022 daemon.info dnsmasq-dhcp[1]: DHCPDISCOVER(br-lan) aa:bb:aa:bb:aa:bb
Wed Feb  9 19:24:14 2022 daemon.info dnsmasq-dhcp[1]: DHCPOFFER(br-lan) 192.168.x.xx aa:bb:aa:bb:aa:bb
Wed Feb  9 19:24:14 2022 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.x.xx aa:bb:aa:bb:aa:bb
Wed Feb  9 19:24:14 2022 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.x.xx aa:bb:aa:bb:aa:bb ESP-2A7E07
Wed Feb  9 19:24:18 2022 daemon.notice mosquitto[7229]: 1644427458: New connection from 192.168.x.xx:49669 on port 1883.
Wed Feb  9 19:24:18 2022 daemon.notice mosquitto[7229]: 1644427458: New client connected from 192.168.x.xx:49669 as dscKeybusInterface (p2, c1, k25, u'DSC').
Wed Feb  9 19:24:59 2022 daemon.notice mosquitto[7229]: 1644427499: Client dscKeybusInterface has exceeded timeout, disconnecting.
Wed Feb  9 19:25:23 2022 daemon.notice mosquitto[7229]: 1644427523: New connection from 192.168.x.xx:60970 on port 1883.
Wed Feb  9 19:25:23 2022 daemon.notice mosquitto[7229]: 1644427523: New client connected from 192.168.x.xx:60970 as dscKeybusInterface (p2, c1, k25, u'DSC').
Wed Feb  9 19:27:17 2022 daemon.notice mosquitto[7229]: 1644427637: Client dscKeybusInterface has exceeded timeout, disconnecting.
Wed Feb  9 19:27:17 2022 daemon.notice mosquitto[7229]: 1644427637: New connection from 192.168.x.xx:61775 on port 1883.
Wed Feb  9 19:27:17 2022 daemon.notice mosquitto[7229]: 1644427637: New client connected from 192.168.x.xx:61775 as dscKeybusInterface (p2, c1, k25, u'DSC').
Wed Feb  9 19:28:11 2022 daemon.notice mosquitto[7229]: 1644427691: Client dscKeybusInterface has exceeded timeout, disconnecting.
```

aritonigr, 2022-02-26:
I have checked the arduino monitor when the service became unavailable. Here is what I get:

MQTT Disconnected, failed to reconnect
MQTT.... connection error: 192.168.1.200

When I have the reset button of esp8266 pressed the service becomes available again.

Any idea?

sidlauskas, 2022-12-30:
I have the same issue that it becomes unavailable and have to reset the ESP8266. 

I also tried to ping the ESP8266 and found that requests keep timing out, even when the WiFi signal is strong (-50dBm)

sidlauskas, 2023-01-09:
It seems I managed to fix this issue just by re-flashing the firmware and selecting the correct board in the Arduino IDE. I was using NodeMCU 1.0 (ESP-12E) instead NodeMCU 0.9 (ESP-12)

adie86, 2024-08-23:
> I admit that I am not familiar with MQTT and Home Assistant and I need some help: In the following code: ... const char* mqttServer = "192.168.1.200"; _**The domain or IP where the MQTT is installed [...]

How is it possible for you to assist with the yaml config.