taligentx/dscKeybusInterface #272: Night mode of multiple partitions

realthk, 2022-02-05:
I'm using this great project for about 2 years now with Home Assistant (MQTT), thanks!

Recently I re-configured my DSC 1864 to have 2 partitions: the first for the house with many zones, and the second only for the gate zone, as to be able to arm these separately.

For the night I use no-entry mode (1N and 2N commands) which works fine, with a strange restriction: partition 1 cannot be armed in night mode while partition 2 is still in pending state of the exit delay! Partition 1 message is stuck at "Enter access code", as if it is waiting for the code... (which is specified in HomeAssistant-MQTT.ino I used, but also given as `code:` parameter of `alarm_control_panel.alarm_arm_night` service calls of Home Assistant)

Oddly enough no problem if
- I arm P1 first, then I can arm P2 in night mode even while the P1 is still in exit delay, pending state
- I arm P2 first, then arm P1 in plain stay or away modes even while P2 is still pending
 
It's only the no-entry night mode of P1 that cannot be armed while P2 is still pending.

Obviously not a huge problem, and I'm not even sure if this is an issue with dscKeybusinterface at all, or just a peculiarity of DSC systems, but it would be nice to know the reason.

taligentx, 2022-02-10:
Hi @realthk - Thanks for the detailed info!  I was able to replicate this issue and confirmed it's an issue in the library, fixed in the `develop` branch.  Let me know if this resolves the issue for you, thanks!

-Nikhil

realthk, 2022-02-10:
Thanks Nikhil, I'll try at the weekend!

realthk, 2022-02-13:
Thanks, it works now fine, no restrictions on arming in "no entry" modes!

closed 2022-02-13 by realthk

taligentx, 2022-02-13:
Great to hear, thanks for the update!