taligentx/dscKeybusInterface #285: HomeAssistant - LWT offline after 24 - 48 hrs

brucehatherell, 2022-03-18:
I have a strange issue with Home Assitant (migrated from HomeKit) where my ESP-8266 goes offline after about 24-48 hrs … it’s not connected to serial so suspect it could be buffer related as I am still able to connect via OTA and re-flash to bring back online - have implemented buffer logs to MQTT this morning but anyone had similar issues?

brucehatherell, 2022-04-12:
Still running into this issue on Home Assistant - removed OTA as I thought that was the cause - anyone else having their ESP8266 drop after about 24 hrs?

brucehatherell, 2022-04-17:
Issue was mine, was not erasing esp8266 when flashing - all working great again.

closed 2022-04-17 by brucehatherell

taligentx, 2022-04-18:
Hi @brucehatherell - great to hear you were able to resolve this, noting this as a troubleshooting step for future intermittent issues on the esp8266.