taligentx/dscKeybusInterface #288: When arming Away the status of the alarm continuously changes between armed and disarmed

armandvanderwalt, 2022-04-22:
I have an interesting problem, when I arm the alarm using the stay command the alarm functions 100% as expected.  But when arming with away the alarm status continues to switch between armed and disarmed. From what I have tested so far the alarm is in fact armed, but the status gets swapped between armed and disarmed for as long as the alarm is armed.

I am using the Home Assistant sketch.

Originally I had arm night also defined, but removed it today to test if that might have been the issue.

Here is my configuration.yaml

``` yaml

alarm_control_panel:
  - platform: mqtt
    name: "Home Alarm"
    state_topic: "dsc/Get/Partition1"
    availability_topic: "dsc/Status"
    command_topic: "dsc/Set"
    payload_disarm: "1D"
    payload_arm_home: "1S"
    payload_arm_away: "1A"

sensor:
  - platform: mqtt
    name: "Home Alarm"
    state_topic: "dsc/Get/Partition1/Message"
    availability_topic: "dsc/Status"
    icon: "mdi:shield"

binary_sensor:
  - platform: mqtt
    name: "Security Trouble"
    state_topic: "dsc/Get/Trouble"
    device_class: "problem"
    payload_on: "1"
    payload_off: "0"
  - platform: mqtt
    name: "Front Door"
    state_topic: "dsc/Get/Zone1"
    device_class: "door"
    payload_on: "1"
    payload_off: "0"
  - platform: mqtt
    name: "Sliding Doors"
    state_topic: "dsc/Get/Zone2"
    device_class: "door"
    payload_on: "1"
    payload_off: "0"
  - platform: mqtt
    name: "Back Door"
    state_topic: "dsc/Get/Zone3"
    device_class: "door"
    payload_on: "1"
    payload_off: "0"
  - platform: mqtt
    name: "Living Room"
    state_topic: "dsc/Get/Zone4"
    device_class: "motion"
    payload_on: "1"
    payload_off: "0"
  - platform: mqtt
    name: "Hallway"
    state_topic: "dsc/Get/Zone5"
    device_class: "motion"
    payload_on: "1"
    payload_off: "0"
  - platform: mqtt
    name: "Master Bedroom"
    state_topic: "dsc/Get/Zone6"
    device_class: "motion"
    payload_on: "1"
    payload_off: "0"

```

I have the PC1832

I only have one partition setup so I excluded the partitions I didn't need

I used nodered to get the mqtt messages posted on the Partition1 topic since connecting directly to the nodemcu with PC is tricky when connected to the panel, but if I need to do that let me know and I can make a plan.

This is the messages sent with their topics

dsc/Get/Partition1/Message
"Exit delay in progress"

dsc/Get/Partition1/Message
"Exit delay in progress"

dsc/Get/Partition1/Message
"Armed: Away"

dsc/Get/Partition1
"armed_night"

dsc/Get/Partition1/Message
"Armed: Stay"

dsc/Get/Partition1

ideen1, 2022-07-05:
I am having this same issue using Homespan

armandvanderwalt, 2022-08-11:
I just rebuilt mine using this schematic and components and now it works perfectly
https://github.com/taligentx/dscKeybusInterface/tree/master/extras/PCB%20Layouts/NodeMCU

closed 2022-08-11 by armandvanderwalt